chipfoundry/openlane2 #17: XOR fails if `RUN_KLAYOUT_STREAMOUT` is set to 0

mo-hosni, 2023-03-13:
### Issue: 
XOR fails if `RUN_KLAYOUT_STREAMOUT` is set to 0 and the flow fails.

### Suggestion:
XOR should not run if `RUN_KLAYOUT_STREAMOUT` is set to 0.

donn, 2023-03-13:
No. This is a misconfigured flow on the part of the designer using a step that requires a KLayout GDS input without generating a KLayout GDS output earlier in the step.

closed 2023-03-13 by donn

mo-hosni, 2023-03-13:
I understand; but this wasn't the case in OpenLane 1. XOR was skipped in OpenLane 1 when Klayout GDS is not generated.

donn, 2023-03-13:
I'm very comfortable breaking compatibility here. Adding a billion exceptions to solve a minor problem is not why I started this rewrite. :slightly_smiling_face: