chipfoundry/openlane2 #61: Expressions don't work in python dict config

smunaut, 2023-05-20:
When using `init_with_config` using a python dict and not a json file, then `_load_dict` ends up being called without `resolve_json=True` argument and the user has no way to set it, which makes expressions don't work.

donn, 2023-05-21:
This is by design- the intent is, if you're writing your dict in Python, there's no need for the expression engine. I'm open to discussing this though.

smunaut, 2023-05-21:
So my use case for it here is when building the top level. Some of the variable are provided from the default project as "Don't touch theses" (so that the result is "caravel compatible") and so I load those from a `.json` file and then add some other custom/dynamic config (like macro and such) programatically.

But those "don't touch" config do include expressions ...

donn, 2023-05-21:
That does make sense.

I'll look into it.

donn, 2023-07-16:
They should work now- but I'll still go ahead and advise against it in scenarios other than precisely this.

closed 2023-07-16 by donn