chipfoundry/openlane2 #90: PDN_MACRO_CONNECTIONS doesn't work. Value gets mangled.

smunaut, 2023-06-02:
The value that gets to the actual TCL script is mangled and not the actual string specified ...
This is what I put in my dict :
```python
              "PDN_MACRO_CONNECTIONS": [
                       "top_I.*tt_um_I vccd1 vssd1 vccd1 vssd1",
               ],
```
But then because there is only one line (I guess) or because of spaces, it get split into a lits of words and then for some reason the first one end up with single quotes around it in the TCL which completely screws it  and doesn't match ...

donn, 2023-06-21:
It's joined with `shlex` to ensure that spaces and such are properly escaped- that said, the splitting is a bug.

However, as far as I can see, testing shows that it does match...

smunaut, 2023-07-17:
That still doesn't work.

To debug, in `set_global_connections.tcl`, I added `puts "|$pdn_hook|$instance_name|"` in the `for` loop.
And this is what gets printed in the logs :

```
|'top_I.*tt_um_I vccd1 vssd1 vccd1 vssd1'|'top_I.*tt_um_I|
```

And you can notice there are extraneous `'` at beginning and end. And the one at the beginning ends up in the `instance_name` and the final one ends up in `ground_pin` ....

donn, 2023-07-18:
Should be fixed now- sorry for the long delay.

Have I mentioned I hate Tcl?

closed 2023-07-18 by donn

smunaut, 2023-07-18:
:tada:   I confirm this fixed the issue for me.

PR #115 (2023-07-13): Stricter Config Validation
* JSON configuration files with `meta.version: 2` and dictionary configurations
  now both subject to stricter validation
    * Strings no longer automatically converted to lists, dicts, numbers, booleans, et cetera
    * Numbers no longer automatically converted to
    booleans
    * Unrecognized keys throw an error instead of a warning
* Steps now only keep a copy of configuration variables that are either common
  or explicitly declared
    * Explicitly declare global routing variables for resizer steps
    * Explicitly declare MagicStep variables for DRC step
* ` [... cut]
PR #121 (2023-07-18): Fix Tcl String Escaping
* Reworked Tcl unsafe string escaping to use home-cooked functions instead of
  "shlex"

---
Resolves #90