chipfoundry/openlane2 #93: KeyError: 'STD_CELL_LIBRARY' on default designs

daquintero, 2023-06-17:
Hi Donn! Hope you are well. Just logging this for posterity, whatever it is worth. 

My environment is the default [ICC-OSIC-TOOLS](https://github.com/iic-jku/iic-osic-tools) which is a mantained version of the Efabless FOSS-ASIC-TOOLS. I clone and `pip install -e .` this repository, and go into the `designs/spm` directory. 

After running the `python harden.py` this error appears:
```
Traceback (most recent call last):
  File "/foss/designs/openlane2/designs/spm/harden.py", line 5, in <module>
    flow = Classic(
  File "/foss/designs/openlane2/openlane/flows/sequential.py", line 77, in __init__
    super().__init__(*args, **kwargs)
  File "/foss/designs/openlane2/openlane/flows/flow.py", line 152, in __init__
    config, design_dir = Config.load(
  File "/foss/designs/openlane2/openlane/config/config.py", line 327, in load
    loaded = loader(
  File "/foss/designs/openlane2/openlane/config/config.py", line 390, in _load_dict
    config_in, pdkpath, scl = Self._get_pdk_config(
  File "/foss/designs/openlane2/openlane/config/config.py", line 565, in _get_pdk_config
    scl = pdk_env["STD_CELL_LIBRARY"]
KeyError: 'STD_CELL_LIBRARY'
```

I think there are a few reasons for this. 

If I were to add this into the configuration dictionary of the `harden.py`default design, say:
```
import os
from openlane import Flow

Classic = Flow.get("Classic")

flow = Classic(
    {
        "PDK": "sky130A",
        "STD_CELL_LIBRARY": os.environ["STD_CELL_LIBRARY"], # I know it is 'sky130_fd_sc_hd'
        "DESIGN_NAME": "spm",
        "VERILOG_FILES": ["./src/spm.v"],
        "CLOCK_PORT": "clk",
        "CLOCK_PERIOD": 10,
    },
    design_dir=".",
)

flow.start()
```
I still get the same error. Looking into the source code,

 I can see `STD_CELL_LIBRARY` is set in the config.py `interactive` method and is passed as a parameter into:

config.py
```
  @classmethod
  def interactive(
      Self,
      DESIGN_NAME: str,
      PDK: str,
      STD_CELL_LIBRARY: Optional[str] = None,
      PDK_ROOT: Optional[str] = None,
      **kwargs,
  )
  ....
config_in, _, _ = Self._get_pdk_config(
      PDK,
      STD_CELL_LIBRARY,
      PDK_ROOT,
  )
```

What I have been able to notice through print statements, is that the `Config.load() config_in ` parameter is properly set based on the dictionary input but the `scl` parameter is just propagated along
```
{'PDK': 'sky130A', 'STD_CELL_LIBRARY': 'sky130_fd_sc_hd', 'DESIGN_NAME': 'spm', 'VERILOG_FILES': ['./src/spm.v'], 'CLOCK_PORT': 'clk', 'CLOCK_PERIOD': 10}
``` 

So if I simply change the `harden.py` script to:
```
import os
from openlane import Flow

Classic = Flow.get("Classic")

flow = Classic(
    {
        "PDK": "sky130A",
        "STD_CELL_LIBRARY": os.environ["STD_CELL_LIBRARY"], # I know it is 'sky130_fd_sc_hd'
        "DESIGN_NAME": "spm",
        "VERILOG_FILES": ["./src/spm.v"],
        "CLOCK_PORT": "clk",
        "CLOCK_PERIOD": 10,
    },
    scl=os.environ["STD_CELL_LIBRARY"], # I know it is 'sky130_fd_sc_hd'
    design_dir=".",
)

flow.start()
```

It all works fine now! Do you indend the config dictionary "  "STD_CELL_LIBRARY": os.environ["STD_CELL_LIBRARY"], # I know it is 'sky130_fd_sc_hd'" to still be valid or you want people to input it via the `scl` parameter?

I am going to propose a fix to the default examples in `designs` so that this does not happen to people migrating from `OpenLanes1` to this version. I decided I will port my whole project into using this tools and I am happy to help develop relevant functionality too!

Cheers for making such great software!

donn, 2023-06-18:
Hello Dario- when the SCL is not specified it's *supposed* to just pick out the default. For some reason it didn't. I'll look into it.

closed 2023-06-18 by donn

daquintero, 2023-06-18:
Cheers Mohamed!!!

PR #96 (2023-06-18): Fix Tcl load bug + incorrect filename
* Fixed a bug with extracting variables from Tcl config files when the variable
  is already set in the environment
* Fixed a bug with saving lib and SDF files
* Fixed `check_antennas.tcl` being mis-named

---

Resolves #91
Resolves #93
Supersedes #94