chipfoundry/openlane2 #98: Error: dpl_cell_pad.tcl, 22 DPL-0039

daquintero, 2023-06-19:
Hi Donn!

Cheers for the help previously! Apologies, I might be annoying again. I am happy to help debug this if you'd like and if you are busy, but since you know how best to fix things I'd rather ask.

I am still trying just to run the `harden.py` example in the latest branch, and now I am getting an error related to the OpenLane flow directly. I have not changed anything as am just exploring the base example and how it manages data and completes.

```
 [INFO RSZ-0039] Resized 242 instances.
           [WARNING DPL-0028] 'sky130_fd_sc_hd__tap*' did not match any masters.
           [WARNING DPL-0028] 'sky130_fd_sc_hd__decap*' did not match any masters.
           [WARNING DPL-0028] 'sky130_ef_sc_hd__decap*' did not match any masters.
           [WARNING DPL-0028] 'sky130_fd_sc_hd__fill*' did not match any masters.
           [ERROR DPL-0039] "'sky130_fd_sc_hd__tap*' 'sky130_fd_sc_hd__decap*' 'sky130_ef_sc_hd__decap*'
           'sky130_fd_sc_hd__fill*'" did not match any masters.
           This could be due to a change from using regex to glob to search for cell masters.
           https://github.com/The-OpenROAD-Project/OpenROAD/pull/3210
           Error: dpl_cell_pad.tcl, 22 DPL-0039
```

I have created a reproducible but I am unsure how to best share it. I've uploaded it to our university sharepoint but let me know what you prefer. 
https://uob-my.sharepoint.com/:u:/g/personal/dq16394_bristol_ac_uk/EU-eOfoPZaVFjFHJgsmUIz4B3cM5h6LDYrVRz2nV34vSsA?e=EcLQt9

donn, 2023-06-21:
Dario, I do not believe you're using a compatible version of OpenROAD, given that I can't load the odb file...

daquintero, 2023-06-21:
Hi Mohamed,

It's a bit weird, so I've got the latest docker environment at [IIC-OSIC-TOOLS](https://github.com/iic-jku/iic-osic-tools). So I am just sending this to explain the context whilst I investigate the environment installation situation you suggest in more detail, but it's strange as I think this is basically the default IIC-OSIC-TOOLS environment with all the latest versions of the tools.

```
daquintero@IT080646:~$ cd iic-osic-tools/
daquintero@IT080646:~/iic-osic-tools$ git log
commit 91bec934ad30cda161706619a22dcf20bd43be25 (HEAD -> main, origin/main, origin/HEAD)
Author: Harald Pretl <harald.pretl@jku.at>
Date:   Mon Jun 5 20:39:12 2023 +0200

    Fix wrong path (due to renaming of IIC-OSIC-TOOLS)
```

And I am on the latest commit of `openlane2`
```
/foss/designs/openlane2 > git log
commit 48cb283b0ff822d261769d0777bf586d4e459752 (HEAD -> main, origin/main, origin/HEAD)
Author: Mohamed Gaber <me@donn.website>
Date:   Mon Jun 19 12:10:30 2023 +0300

    Update Magic (#97)

/foss/designs/openlane2/designs > cd spm/
/foss/designs/openlane2/designs/spm > ls
config.json  harden.py  interactive.ipynb  pin_order.cfg  runs  src
/foss/designs/openlane2/designs/spm > python
Python 3.10.6 (main, Mar 10 2023, 10:55:28) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import openlane
>>> openlane.__file__
'/foss/designs/openlane2/openlane/__init__.py'
>>> quit()
/foss/designs/openlane2/designs/spm > ls
config.json  harden.py  interactive.ipynb  pin_order.cfg  runs  src
/foss/designs/openlane2/designs/spm > git status
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean
/foss/designs/openlane2/designs/spm > python harden.py
[10:41:57] Starting a new run of the 'Classic' flow with the tag 'RUN_2023-06-21_10-41-57'.
```

It is by running this that I get the error? 

I also know that Openlane v1 works because I have been using it for some design hardening too:

```
[STEP 42]
[INFO]: Running Circuit Validity Checker ERC (log: designs/spm/runs/RUN_2023.06.20_14.41.57/logs/signoff/42-erc_screen.log)...
[INFO]: Saving current set of views in 'designs/spm/runs/RUN_2023.06.20_14.41.57/results/final'...
[INFO]: Saving runtime environment...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/spm/runs/RUN_2023.06.20_14.41.57/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/spm/runs/RUN_2023.06.20_14.41.57/reports/metrics.csv'.
[INFO]: There are no max slew, max fanout or max capacitance violations in the design at the typical corner.
[INFO]: There are no hold violations in the design at the typical corner.
[INFO]: There are no setup violations in the design at the typical corner.
[SUCCESS]: Flow complete.
[INFO]: Note that the following warnings have been generated:
[WARNING]: 1 warnings found by Verilator
[WARNING]: Current core area is too small for the power grid settings chosen. The power grid will be scaled down.
[WARNING]: Module sky130_fd_sc_hd__tapvpwrvgnd_1 blackboxed during sta
[WARNING]: Module sky130_ef_sc_hd__decap_12 blackboxed during sta
[WARNING]: Module sky130_fd_sc_hd__fill_1 blackboxed during sta
[WARNING]: Module sky130_fd_sc_hd__fill_2 blackboxed during sta

/foss/tools/openlane/2023.05 > cd $OPENLANE_ROOT
/foss/tools/openlane > cd 2023.05/
/foss/tools/openlane/2023.05 > ls
AUTHORS.md       default.cvcrc  docker  flow.tcl     klayoutrc  README.md             requirements_lint.txt  runs
configuration    dependencies   docs    gui.py       LICENSE    regression_results    requirements.txt       scripts
CONTRIBUTING.md  designs        env.py  Jenkinsfile  Makefile   requirements_dev.txt  run_designs.py         tests
```

I'll investigate your lead and get back to you! If I find a solution ofc I'll share for future users in a likely similar situation

donn, 2023-06-21:
The only version of OpenROAD confirmed to work is the one listed here:

https://github.com/efabless/openlane2/blob/48cb283b0ff822d261769d0777bf586d4e459752/nix/openroad.nix#LL30C10-L30C50

The one in iic-osic-tools is considerably newer:

https://github.com/iic-jku/iic-osic-tools/blob/91bec934ad30cda161706619a22dcf20bd43be25/tool_metadata.yml#L68

I probably need to update OpenROAD in OL2 though because the version I'm on is ancient, but in the future please do mention the mechanism by which you ran OL2 when filing an issue. It saves me a lot of time.

daquintero, 2023-06-21:
Sorry about that! I should have referenced the first issue https://github.com/efabless/openlane2/issues/93 where I did mention the environment.

If I can offer you a hand at any point during the upgrade, very happy to help as directed to save you time! Thanks again for all the help so far!

donn, 2023-09-11:
*Should* be resolved by #128…?

daquintero, 2023-09-12:
Hi donn, thanks so much and really enjoying openlane2! I hear Harald is about to do a September build in IIC-OSIC-TOOLS and will test when he releases it. Thanks again

donn, 2023-12-11:
We're going to lock this as it's been long enough- feel free to file a new issue if you find that the problem persists.

closed 2023-12-11 by donn