chipfoundry/openlane2 #139: Final spef views are in folders max_, min_, nom_. Why?

mo-hosni, 2023-08-17:
The final spef views are saved in the folders `max_`, `min_`, `nom_`. 
The spef files already have the post-fix `<interconnect_corner>.spef` so I guess the folders are not necessary.

donn, 2023-08-17:
I think there was a reason for it but I cannot recall at the moment

donn, 2023-08-24:
So, the reason is they're named after the keys for the SPEF file list for the dictionary, which are all wildcards for fully qualified IPVT corners: `min_*` would match `min_ff_...` etc. I remove the asterisk because exactly 0 utilities handle asterisks in filenames without incredible hassle.

What I do want to do here is remove them from the filenames, actually.

donn, 2023-08-24:
Merging into #164

closed 2023-08-24 by donn