chipfoundry/openlane2 #164: Final directory file names

kareefardi, 2023-08-24:
A list of proposals for filenames in `final` directory:
- `gds/<design>.<tool>.gds` -> `gds/<design>.gds`
- `(lib|sdf)/<corner>/<design><corner>.*` -> `*/<corner>/<design>.*`
- `nl/<design>.nl.v` -> `nl/<design>.v`
- `pnl/<design>.pnl.v` -> `pnl/<design>.v`

donn, 2023-08-24:
So, after discussing it offline, Kareem and I decided that this will be kept as is, except for the file in the `gds` folder.

We're currently proceeding with the principle if that if flattened, none of the files should collide. Duplication of information in folder names isn't new (see: the entire sky130 PDK and corpus of previous Efabless IP)

donn, 2023-09-11:
Resolved in #165