chipfoundry/openlane2 #167: There is no PDN configuration variables documentation in the floorplan section

mo-hosni, 2023-08-27:
There is no PDN configuration variables documentation in the floorplan section

donn, 2023-08-28:
Yes, that would be because it's in the PDN section??????

mo-hosni, 2023-08-28:
I cannot see a PDN section here: https://openlane2.readthedocs.io/en/latest/usage/hardening_macros.html
The PDN variables exist only in the PDK variables list here: https://openlane2.readthedocs.io/en/latest/reference/common_pdk_vars.html
which is a bit annoying because those should be modified for every macro. I know the default value should be different from one PDK to another. But I think those are very common variables that should be added to the "hardening macros" guide. Also, VPITCH and HPITCH are not in the PDK variables list and those are the 2 most important variables.

PR #173 (2023-08-28): Better PDN Documentation + DPL Bugfixes
* Added diode padding to `dpl_cell_pad.tcl`
* Added legalization after `repair_antennas`
* Added `FULL_LIBS` to `YosysStep` without any redactions
* Moved all PDN variables to the the `OpenROAD.GeneratePDN` step as step-specific
  PDK variables
* Updated PDN documentation to make sense and added an illustrated diagram of
  values
* Slightly reordered Classic Flow so detailed placement happens right after
  `Odb.HeuristicDiodeInsertion`
* Fixed translation of `FP_PDN_MACRO_HOOKS` when a string is provided (only
  splitting if a `,` is found)
* Removed ext [... cut]
closed 2023-08-28 by donn