chipfoundry/openlane2 #201: Checklist of Step Unit Tests (excl. OpenROAD)

donn, 2023-10-02:
- [ ] checker.py
- [ ] ~~cvc_rv.py~~
- [ ] klayout.py
- [ ] magic.py
- [ ] netgen.py
- [ ] odb.py
- [ ] verilator.py
- [ ] yosys.py

kareefardi, 2023-10-03:
- [ ] Checker.DisconnectedPins
- [ ] Checker.IllegalOverlap
- [ ] Checker.LVS
- [ ] Checker.LintErrors
- [ ] Checker.LintTimingConstructs
- [ ] Checker.LintWarnings
- [ ] Checker.MagicDRC
- [ ] Checker.TrDRC
- [ ] Checker.WireLength
- [ ] Checker.XOR
- [ ] Checker.YosysChecks
- [ ] Checker.YosysUnmappedCells
- [ ] KLayout.OpenGUI
- [ ] KLayout.Render
- [ ] KLayout.StreamOut
- [ ] KLayout.XOR
- [ ] Magic.DRC
- [ ] Magic.SpiceExtraction
- [ ] Magic.StreamOut
- [ ] Magic.WriteLEF
- [ ] Misc.LoadBaseSDC
- [ ] Netgen.LVS
- [ ] Odb.ApplyDEFTemplate
- [ ] Odb.CustomIOPlacement
- [ ] Odb.DiodesOnPorts
- [ ] Odb.HeuristicDiodeInsertion
- [ ] Odb.ManualMacroPlacement
- [ ] Odb.ReportDisconnectedPins
- [ ] Odb.ReportWireLength
- [ ] OpenROAD.BasicMacroPlacement
- [ ] OpenROAD.CTS
- [ ] OpenROAD.CheckAntennas
- [ ] OpenROAD.CheckSDCFiles
- [ ] OpenROAD.DetailedPlacement
- [ ] OpenROAD.DetailedRouting
- [ ] OpenROAD.FillInsertion
- [ ] OpenROAD.Floorplan
- [ ] OpenROAD.GeneratePDN
- [ ] OpenROAD.GlobalPlacement
- [ ] OpenROAD.GlobalPlacementSkipIO
- [ ] OpenROAD.GlobalRouting
- [ ] OpenROAD.IOPlacement
- [ ] OpenROAD.IRDropReport
- [ ] OpenROAD.LayoutSTA
- [ ] OpenROAD.OpenGUI
- [ ] OpenROAD.RCX
- [ ] OpenROAD.RepairDesign
- [ ] OpenROAD.RepairDesignPostGPL
- [ ] OpenROAD.RepairDesignPostGRT
- [ ] OpenROAD.ResizerTimingPostCTS
- [ ] OpenROAD.ResizerTimingPostGRT
- [ ] OpenROAD.STAMidPNR
- [ ] OpenROAD.STAPostPNR
- [ ] OpenROAD.STAPrePNR
- [ ] OpenROAD.TapEndcapInsertion
- [ ] Verilator.Lint
- [ ] Yosys.EQY
- [ ] Yosys.JsonHeader
- [ ] Yosys.Synthesis

kareefardi, 2023-10-17:
## Unit tests per step:
- General case:
    - [ ] success
    - [ ] failure:
        - [ ] bad input
        - [ ] bad configuration. e.g. floorplan impossible util...
    - [ ] reproducibles that are fixed
- [x] Checker.DisconnectedPins
- [x] Checker.IllegalOverlap
- [x] Checker.LVS
- [x] Checker.LintErrors
- [ ] Checker.LintTimingConstructs
    - [ ] fail
    - [ ] success
    - [ ] not found
- [x] Checker.LintWarnings
- [x] Checker.MagicDRC
- [x] Checker.TrDRC
- [ ] Checker.WireLength
    - [ ] fail
    - [ ] success
    - [ ] not found
- [x] Checker.XOR
- [x] Checker.YosysChecks
- [x] Checker.YosysUnmappedCells
- [ ] KLayout.OpenGUI
- [ ] KLayout.Render
    - [ ] success comparison with a known png
    - [ ] fail - bad DEF.
- [ ] KLayout.StreamOut
    - [ ] success no macros
    - [ ] success macros only
    - [ ] success hybrid (macros + std cells)
    - [ ] fail bad DEF.
    - [ ] fail bad LEF.
- [ ] KLayout.XOR
    - [ ] success
    - [ ] fail xor
    - [ ] fail bad DEF
    - [ ] fail bad LEF.
    - [ ] multithread option - check speed ??
    - [ ] ignore layers option. True XOR in an ignored layer reporting clean XOR.
- [ ] Magic.DRC
    - [ ] true positive - there are DRC errors with GDS
    - [ ] true positive - there are DRC errors with DEF-LEF
    - [ ] true negative - no DRC errors with GDS
    - [ ] true negative - no DRC errors with DEF-LEF
    - [ ] fail bad DEF
    - [ ] fail bad LEF
    - [ ] fail bad GDS
- [ ] Magic.SpiceExtraction
    - NOTE: MAGIC_NO_EXT_UNIQUE - JC (user) test case add to integartion tests - a design
    - [ ] success using GDS
    - [ ] success using DEF
    - [ ] success with IllegalOverlap
- [ ] Magic.StreamOut
    - Pending StreamOut sequence rework
- [ ] Magic.WriteLEF
    - success abstract DEF
    - success full DEF
    - success abstract GDS
    - success full GDS
    - bad input DEF
    - bad input LEF
    - bad input GDS
- [ ] Misc.LoadBaseSDC
- [ ] Netgen.LVS
    - pass
    - pass blackbox
    - pass full
    - fail on each of the following:
        - [ ] design__lvs_device_difference__count
        - [ ] design__lvs_net_differences__count
        - [ ] design__lvs_property_fails__count
        - [ ] design__lvs_errors__count
        - [ ] design__lvs_unmatched_devices__count
        - [ ] design__lvs_unmatched_nets__count
        - [ ] design__lvs_unmatched_pins__count
    - fail bad spice
    - fail bad verilog
- [ ] Odb.ApplyDEFTemplate
    - [ ] success
    - [ ] bad input DEF
    - [ ] mismatching input DEF
- [ ] Odb.CustomIOPlacement
    - [ ] success
    - [ ] fail due to umatched pins
    - [ ] fail due to bad input odb
    - [ ] fail due to bad configuration file. There are multiple ways in which the configuration file can be bad.
    TBD point of failures in the configuration file.
- [ ] Odb.DiodesOnPorts
    - [ ] success on in
    - [ ] success on out
    - [ ] success on both
    - [ ] skip on none
    - [ ] fail on bad DIODE_CELL config - multiple points of failure
    - [ ] fail on bad odb 
- [ ] Odb.HeuristicDiodeInsertion
    - [ ] success on different threshold values
    - [ ] fail on bad odb
    - [ ] fail on bad DIODE_CELL config - multiple points of failure
- [ ] Odb.ManualMacroPlacement
    - [ ] success
    - [ ] fail bad odb
    - [ ] fail misconfigured MACRO object - potential enhancement  ??
    - no need to test cfg file as it is under deprecation.
- [ ] Odb.ReportDisconnectedPins
    - [ ] positive disconnected power and signal pins
    - [ ] fail bad odb
- [ ] Odb.ReportWireLength
    - [ ] success
    - [ ] success no wires in odb
    - [ ] fail bad odb
- [ ] OpenROAD.BasicMacroPlacement
    - Not implemented ??
- [ ] OpenROAD.CTS
    - CTS is sensitive to SDC
    - There seems to be alot of configuration paramters for CTS. I am not entierly sure yet what would be a strategy to test all of them
    - [ ] success multiple clocks
    - [ ] success multiple generated clocks
    - [ ] success single clock
    - [ ] success no clock
    - [ ] fail bad odb
- [ ] OpenROAD.CheckAntennas
    - [ ] success no violations
    - [ ] success violations
    - [ ] fail no routes
    - [ ] fail bad odb
- [ ] OpenROAD.CheckSDCFiles
- [ ] OpenROAD.DetailedPlacement
    - fail no space -> over exagerate padding
    - fail no space -> resonable padding util > 100%
    - fail max displacement x
    - fail max displacement y
    - success max displacement x
    - success max displacement y
    - success
    - fail bad odb
- [ ] OpenROAD.DetailedRouting
    - [ ] fail too little iterations
    - [ ] fail no access using min and max layerss
    - [ ] fail no guides - this might not fail and trigger FlexGR
    - [ ] fail no access obs
    - [ ] success
    - [ ] fail bad odb
- [ ] OpenROAD.FillInsertion
    - [ ] fail bad filler cell
    - [ ] fail/success no space
    - [ ] success
    - [ ] fail bad odb
- [ ] OpenROAD.Floorplan
    - [ ] success relative
    - [ ] success aboslute
    - [ ] test 0 margin
    - [ ] test -ve margin
    - [ ] test margin > area
    - [ ] test impossible dimensions
    - [ ] fail bad netlist
- [ ] OpenROAD.GeneratePDN
- [ ] OpenROAD.GlobalPlacement
- [ ] OpenROAD.GlobalPlacementSkipIO
- [ ] OpenROAD.GlobalRouting
- [ ] OpenROAD.IOPlacement
- [ ] OpenROAD.IRDropReport
- [ ] OpenROAD.LayoutSTA
- [ ] OpenROAD.OpenGUI
- [ ] OpenROAD.RCX
- [ ] OpenROAD.RepairDesign
- [ ] OpenROAD.RepairDesignPostGPL
- [ ] OpenROAD.RepairDesignPostGRT
- [ ] OpenROAD.ResizerTimingPostCTS
- [ ] OpenROAD.ResizerTimingPostGRT
- [ ] OpenROAD.STAMidPNR
- [ ] OpenROAD.STAPostPNR
- [ ] OpenROAD.STAPrePNR
- [ ] OpenROAD.TapEndcapInsertion
- [ ] Verilator.Lint
- [ ] Yosys.EQY
- [ ] Yosys.JsonHeader
- [ ] Yosys.Synthesis