chipfoundry/openlane2 #226: UnicodeDecodeError in Stage 10 - STA (Pre-PnR)

7FM, 2023-10-18:
### Description

My project currently failes with a python stacktrace due to a decoding error in stage 10.
I tried the latest beta `2.0.0b15` via a nix flake:
```
        my_openlane2 = let
          src = pkgs.fetchFromGitHub {
            owner = "efabless";
            repo = "openlane2";
            rev = "2.0.0b15";
            sha256 = "sha256-jQDICuw5H3Z86jl2dlwuQi49FYcUPYlEcZJ8MnwjeHs=";
          };
        in import "${src}/default.nix" { pkgs = openlane2_pkgs; };
```
with `openlane2_pkgs` being pinned to the same version as specified here https://github.com/efabless/openlane2/blob/d712f051a921068d9506ec97fcd6365c5785baac/nix/pkgs.nix#L14

### Expected Behavior

Do not produce a python stacktrace.

### Environment report

```sh
openlane.env_info || /nix/store/z8r9wkxv0m3jix2pm5f17h23w69f2b5f-python3.10-openlane/lib/python3.10/site-packages/openlane/env_info.py
kernel: Linux
kernel_version: 6.1.54
supported: True
distro: nixos
distro_version: 23.11
python_version: 3.10.9
python_path:
  - /nix/store/z8r9wkxv0m3jix2pm5f17h23w69f2b5f-python3.10-openlane/bin
  - /nix/store/96165i71r0l69b1n3pn8z4qksby5g30c-openroad/lib/python3.10/site-packages
  - /nix/store/yizhphjr22q1rr77n1067vyykq520gdd-klayout-6a36bfa7c04f55bd732f8e0f91b553c8f9cebed7/lib/python3.10/site-packages
  - /nix/store/65cp4izx3bllnwqn7c7dhrq9h9gmjkal-python3-3.10.9/lib/python3.10/site-packages
  - /nix/store/y3ymwmqpmarqqqxp537jsgiqf1mp5rqz-netgen/lib/python3.10/site-packages
  - /nix/store/l4aj7qwjpia6ympk913jsvv26q22bb01-yosys-0.23/lib/python3.10/site-packages
  - /nix/store/1xfp6kmcah4pmic46v2s2nbgmm08ncv0-magic-vlsi/lib/python3.10/site-packages
  - /nix/store/z41wwdc6zlmxz0mgcjmg33gmdcxwsww4-iverilog-11.0/lib/python3.10/site-packages
  - /nix/store/z9sqf8a4vy55416cbxz40wz6mdhjw946-verilator/lib/python3.10/site-packages
  - /nix/store/cf2fghb8vxxwmniv8cl6cvsyzabjrvl3-ruby-2.7.6/lib/python3.10/site-packages
  - /nix/store/378rxqdx6lh4swhw49ifbilc6hbk7h2r-tcl-8.6.11/lib/python3.10/site-packages
  - /nix/store/ns7p6q9gw667v41ysn750103v7dgxgwz-python3.10-click-8.1.3/lib/python3.10/site-packages
  - /nix/store/n0kq2qalwygww8h3pr7v6av7h7jxgd25-python3.10-cloup-1.0.1/lib/python3.10/site-packages
  - /nix/store/l8y1hkbki6pjbr1lm0k0iv51ii9864wz-python3.10-pyyaml-6.0/lib/python3.10/site-packages
  - /nix/store/gkphk53397gfy6mr5nhlqyg5cmgk304h-python3.10-rich-12.5.1/lib/python3.10/site-packages
  - /nix/store/i09vhwzivdhr017688ywhzjwz14xdhn9-python3.10-requests-2.28.1/lib/python3.10/site-packages
  - /nix/store/yrp4f59aa0a1kvrgzdm58v56q8h90604-python3.10-pcpp-1.21/lib/python3.10/site-packages
  - /nix/store/25dlz6av123fbc6qclh6w2agq62dlkmh-python3.10-volare/lib/python3.10/site-packages
  - /nix/store/qvb8isbg9kgmjzkmm9i20r05hns807cj-python3.10-tkinter-3.10.9/lib/python3.10/site-packages
  - /nix/store/xkvl0krzk8l9ch8ksxkb9xmg2jjiszq7-python3.10-lxml-4.9.1/lib/python3.10/site-packages
  - /nix/store/cdgsnj5kmpawp52iramjc7craindq2hr-python3.10-deprecated-1.2.13/lib/python3.10/site-packages
  - /nix/store/aip2pp28ld9wcbqgwsgy3xybqhc21025-python3.10-immutabledict-2.2.2/lib/python3.10/site-packages
  - /nix/store/qll21g8p4421c0bw98qs8rg9gr64256g-python3.10-libparse/lib/python3.10/site-packages
  - /nix/store/7yiqh8h334alb7pf340p3pjindqh5f5n-python3-3.10.12-env/lib/python3.10/site-packages
  - /nix/store/z8r9wkxv0m3jix2pm5f17h23w69f2b5f-python3.10-openlane/lib/python3.10/site-packages
  - /nix/store/49pg4ljx6qandwdr6mbc8w8k8fh0dd2j-python3.10-setuptools-65.3.0/lib/python3.10/site-packages
  - /nix/store/k5b7jpzi9dvg3x7h2a90df14aq8wh6bl-python3.10-commonmark-0.9.1/lib/python3.10/site-packages
  - /nix/store/8zzhma74gr86jzakxd8f1pzlrdf8snzx-python3.10-future-0.18.2/lib/python3.10/site-packages
  - /nix/store/cpv31grqdbxvsq7jx4kdgbxg2rsxkkfy-python3.10-pygments-2.13.0/lib/python3.10/site-packages
  - /nix/store/9ljcxsc2gqnf5gysl45993v4v88iwzz7-python3.10-docutils-0.19/lib/python3.10/site-packages
  - /nix/store/6bkqcc8d265j605rzrb92lr2pn82442h-python3.10-brotlicffi-1.0.9.2/lib/python3.10/site-packages
  - /nix/store/pjjdz5sjrf2xk8vzf7461wbhkky00k4r-python3.10-cffi-1.15.1/lib/python3.10/site-packages
  - /nix/store/6i9gf8hz2h9as3lxsraysv7qjh102g67-python3.10-pycparser-2.21/lib/python3.10/site-packages
  - /nix/store/8k41q7l3ff3i4lz0drn34w3phk8fw49z-python3.10-certifi-2022.12.07/lib/python3.10/site-packages
  - /nix/store/2s5sa0zibgjshn0s3hhjna4cql3flsnn-python3.10-charset-normalizer-2.1.0/lib/python3.10/site-packages
  - /nix/store/wl8rwlabb29g5gqpy0dsqf9g4xagnc2j-python3.10-idna-3.4/lib/python3.10/site-packages
  - /nix/store/3whxw66aiiixk326lj6h63cxbv0cpilg-python3.10-urllib3-1.26.12/lib/python3.10/site-packages
  - /nix/store/vwrdsq0v60p5gv5v6vlydmy045ihf29q-python3.10-brotli-1.0.9/lib/python3.10/site-packages
  - /nix/store/ah2w40lqazbkjd4cga7binijihm8dvb3-python3.10-pysocks-1.7.1/lib/python3.10/site-packages
[... 63 lines cut ...]
start
    final_state, step_objects = self.run(
  File "/nix/store/z8r9wkxv0m3jix2pm5f17h23w69f2b5f-python3.10-openlane/lib/python3.10/site-packages/openlane/flows/sequential.py", line 276,
in run
    current_state = step.start(
  File "/nix/store/z8r9wkxv0m3jix2pm5f17h23w69f2b5f-python3.10-openlane/lib/python3.10/site-packages/openlane/steps/step.py", line 704, in
start
    views_updates, metrics_updates = self.run(state_in_result, **kwargs)
  File "/nix/store/z8r9wkxv0m3jix2pm5f17h23w69f2b5f-python3.10-openlane/lib/python3.10/site-packages/openlane/steps/openroad.py", line 379, in
run
    views_updates, metrics_updates = super().run(state_in, env=env, **kwargs)
  File "/nix/store/z8r9wkxv0m3jix2pm5f17h23w69f2b5f-python3.10-openlane/lib/python3.10/site-packages/openlane/steps/openroad.py", line 263, in
run
    views_updates, metrics_updates = super().run(state_in, env=env, **kwargs)
  File "/nix/store/z8r9wkxv0m3jix2pm5f17h23w69f2b5f-python3.10-openlane/lib/python3.10/site-packages/openlane/steps/tclstep.py", line 193, in
run
    generated_metrics = self.run_subprocess(
  File "/nix/store/z8r9wkxv0m3jix2pm5f17h23w69f2b5f-python3.10-openlane/lib/python3.10/site-packages/openlane/steps/step.py", line 851, in
run_subprocess
    while line := process_stdout.readline():
  File "/nix/store/65cp4izx3bllnwqn7c7dhrq9h9gmjkal-python3-3.10.9/lib/python3.10/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xf0 in position 7828: invalid continuation byte
Classic - Stage 10 - STA (Pre-PnR) ━━━━━━╸━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━  9/55 0:07:23%
```

7FM, 2023-10-18:
I was able to run the `sta` command manually and write the log to a file:
[sta_log.zip](https://github.com/efabless/openlane2/files/13029639/sta_log.zip)
Starting in line 181382, `sta` seems to print binary gibberish at the end of the lines, this is likely why python failes reading the lines.
Combined with the fact that removing `"SYNTH_AUTONAME": 1,` from the config.json seems to mitigate this.
I would assume that openroad has somewhere a hard limit for module names and my design exceeds this and might cause some sort of buffer overflow or other weird undefined behavior.

donn, 2023-10-18:
Thank you for the incredibly detailed report and debugging. It's a breath of fresh air.

There are two action items for us here:

1. Forward this test case to OpenROAD/OpenSTA
2. OpenLane should capture and report a more meaningful error for gibberish

7FM, 2023-10-18:
Thanks for your fast response!

>     1. Forward this test case to OpenROAD/OpenSTA

Done :) https://github.com/The-OpenROAD-Project/OpenSTA/issues/204

The second action point would still be helpful and might pin down such issues in the future faster, took me quite some time to assemble the command (especially with all env vars) to obtain a full log file with the gibberish.

PR #385 (2024-02-13): Handle non-UTF-8 outputs by subprocesses
* `openlane.steps.Step`
  * Fixed `run_subprocess` crashing upon a tool outputting gibberish - a `StepException` is raised now

---

Fixes #226
closed 2024-02-13 by kareefardi