chipfoundry/openlane2 #253: OpenROAD.IOPlacement options

kareefardi, 2023-11-05:
### Description

io_layer_variables https://github.com/efabless/openlane2/blob/12c6d69ce9d08f0c524fdc6aef179f7632b9fa30/openlane/steps/common_variables.py#L19 used in `OpenROAD.IOPlacement` have no validation. Layout would be generated with geometry related issues such as overlapping pins and pins that violate spacing rules. Back in the day, these variables were added to ensure maximum routability for macro integration. In a recent (will find the link) discussion with OpenROAD it was questioned whether these options are really needed and should be removed. We didn't get back to OpenROAD if we value these options. 

### Proposal

Study the feasibility and value of these options and decice to either remove them or keep them and have some sort of validation for these variables

donn, 2024-04-03:
Do you have a link to the issue where they asked for us to re-investigate our need for those options?

kareefardi, 2024-07-07:
> Do you have a link to the issue where they asked for us to re-investigate our need for those options?

I will look for it

kareefardi, 2024-07-07:
https://github.com/The-OpenROAD-Project/OpenROAD/issues/3372

donn, 2024-07-07:
@mo-hosni, weigh in?