chipfoundry/openlane2 #257: env(synthesis_results)no such variable

Vinayakamk, 2023-11-08:
### Description

my design is very huge (above 200k+ cells), so i used synth_no_flat:1 in config.json.and as i dont want **lint** check to happen **i commented first 4 steps in classic.py**.
as i set it run,during synthesis (after abc exeuting pass).i got the error pasted below.
how can i fix it.

[image]

### Expected Behavior

expected behavior is as usual when we run smoke test or any other design ,without error

### Environment report

```sh
as shown in image.
there are no other warning.
```

### Reproduction material

-

### Relevant log output

```sh
-
```

donn, 2023-11-08:
For future reference; we would appreciate it if you would fill out the issue template accurately. Had this error not been extremely simple, I would have had to ask you to re-file it.

---

This is a matter of `SYNTH_NO_FLAT` using an environment variable that was available in OpenLane 1, but not 2, which is a mistake on my part. It should be fixed by #258 which will be part of beta 16.

donn, 2023-11-08:
Also- you don't need to comment anything out. You can just add `--from yosys.jsonheader` while invoking `openlane`

closed 2023-11-08 by donn

Vinayakamk, 2023-11-10:
thanks for the help and fix .but issue still persits in this way mentioned below.(it moved on forward but error occured during flattening)
[image]
[yosys-synthesis.log](https://github.com/efabless/openlane2/files/13316701/yosys-synthesis.log)

curently errors.log and warning.log is empty.

can you kindly help!!

PR #258 (2023-11-08): Fix `SYNTH_NO_FLAT` using an OpenLane 1-only environment variable
Resolves #257