chipfoundry/openlane2 #272: Flow can't run where there's a large number of macros

urish, 2023-11-18:
### Description

The flow uses environment variables to pass configuration options to the commands executed by the steps. 

When running the flow with a large number of macros, the flow runs out of space for the environment variables, and fails with the following error:

```
OSError: [Errno 7] Argument list too long: 'yosys'
```

As a matter of fact, the argument list isn't the problem (the command it is trying to run is just `yosys -c /nix/store/cbyby0fylkjnbv36hzrbsmxc8srh4d41-python3-3.10.9-env/lib/python3.10/site-packages/openlane/scripts/yosys/json_header.tcl`). 

Rather, it fails because command line args and environment variables [share the same pool](https://stackoverflow.com/a/28865503/830623).

Looking at the environment variables that OpenLane passes to downstream commands, `MACROS`, which contains a JSON object, is huge - seems like the size is [above the 128K limit](https://askubuntu.com/a/1385554/261952)

A workaround I found is simply to avoid passing "MACROS" as an environment variable. I'm not sure if any of the steps actually need it, but I was able to get through must of flow just by adding `        del env["MACROS"]` after https://github.com/efabless/openlane2/blob/4c7deb421e35285432410719a326a2e8801d7340/openlane/steps/tclstep.py#L192-L193.

### Expected Behavior

The flow would run to completion successfully

### Environment report

```sh
kernel: Linux
kernel_version: 5.15.133.1-microsoft-standard-WSL2
supported: True
distro: ubuntu
distro_version: 20.04
python_version: 3.10.9
python_path:
  - /nix/store/jplv2wmvbkjzd7phx0713sin0vry1i3s-python3.10-openlane/bin
  - /nix/store/ns7p6q9gw667v41ysn750103v7dgxgwz-python3.10-click-8.1.3/lib/python3.10/site-packages
  - /nix/store/n0kq2qalwygww8h3pr7v6av7h7jxgd25-python3.10-cloup-1.0.1/lib/python3.10/site-packages
  - /nix/store/l8y1hkbki6pjbr1lm0k0iv51ii9864wz-python3.10-pyyaml-6.0/lib/python3.10/site-packages
  - /nix/store/gkphk53397gfy6mr5nhlqyg5cmgk304h-python3.10-rich-12.5.1/lib/python3.10/site-packages
  - /nix/store/i09vhwzivdhr017688ywhzjwz14xdhn9-python3.10-requests-2.28.1/lib/python3.10/site-packages
  - /nix/store/yrp4f59aa0a1kvrgzdm58v56q8h90604-python3.10-pcpp-1.21/lib/python3.10/site-packages
  - /nix/store/25dlz6av123fbc6qclh6w2agq62dlkmh-python3.10-volare/lib/python3.10/site-packages
  - /nix/store/qvb8isbg9kgmjzkmm9i20r05hns807cj-python3.10-tkinter-3.10.9/lib/python3.10/site-packages
  - /nix/store/xkvl0krzk8l9ch8ksxkb9xmg2jjiszq7-python3.10-lxml-4.9.1/lib/python3.10/site-packages
  - /nix/store/cdgsnj5kmpawp52iramjc7craindq2hr-python3.10-deprecated-1.2.13/lib/python3.10/site-packages
  - /nix/store/aip2pp28ld9wcbqgwsgy3xybqhc21025-python3.10-immutabledict-2.2.2/lib/python3.10/site-packages
  - /nix/store/qll21g8p4421c0bw98qs8rg9gr64256g-python3.10-libparse/lib/python3.10/site-packages
  - /nix/store/l4aj7qwjpia6ympk913jsvv26q22bb01-yosys-0.23/lib/python3.10/site-packages
  - /nix/store/96165i71r0l69b1n3pn8z4qksby5g30c-openroad/lib/python3.10/site-packages
  - /nix/store/yizhphjr22q1rr77n1067vyykq520gdd-klayout-6a36bfa7c04f55bd732f8e0f91b553c8f9cebed7/lib/python3.10/site-packages
  - /nix/store/y3ymwmqpmarqqqxp537jsgiqf1mp5rqz-netgen/lib/python3.10/site-packages
  - /nix/store/1xfp6kmcah4pmic46v2s2nbgmm08ncv0-magic-vlsi/lib/python3.10/site-packages
  - /nix/store/z41wwdc6zlmxz0mgcjmg33gmdcxwsww4-iverilog-11.0/lib/python3.10/site-packages
  - /nix/store/z9sqf8a4vy55416cbxz40wz6mdhjw946-verilator/lib/python3.10/site-packages
  - /nix/store/378rxqdx6lh4swhw49ifbilc6hbk7h2r-tcl-8.6.11/lib/python3.10/site-packages
  - /nix/store/pjjdz5sjrf2xk8vzf7461wbhkky00k4r-python3.10-cffi-1.15.1/lib/python3.10/site-packages
  - /nix/store/6i9gf8hz2h9as3lxsraysv7qjh102g67-python3.10-pycparser-2.21/lib/python3.10/site-packages
  - /nix/store/65cp4izx3bllnwqn7c7dhrq9h9gmjkal-python3-3.10.9/lib/python3.10/site-packages
  - /nix/store/3gb94vhj1ycgvn135h19qh5zlr39mr2f-python3.10-argon2-cffi-bindings-21.2.0/lib/python3.10/site-packages
  - /nix/store/cbyby0fylkjnbv36hzrbsmxc8srh4d41-python3-3.10.9-env/lib/python3.10/site-packages
  - /nix/store/22pfzrq241ipn89330825ybdildl23wx-python3.10-notebook-6.4.12/lib/python3.10/site-packages
  - /nix/store/19n7yd6drwv783w65yf7h7mcrfaagpb7-python3.10-Jinja2-3.1.2/lib/python3.10/site-packages
  - /nix/store/j0icmszh4azcmikmrzib4vicnwns6fn9-python3.10-babel-2.11.0/lib/python3.10/site-packages
  - /nix/store/yfxxhqmckk7kljaqydz07xjfhv3wnx2d-python3.10-pytz-2022.2.1/lib/python3.10/site-packages
  - /nix/store/yfis2i220a8pgy53kx0g5wqwdl7hjn42-python3.10-markupsafe-2.1.1/lib/python3.10/site-packages
  - /nix/store/5jphlf51x69jjlj4did10nd9f68xnml0-python3.10-tornado-6.2/lib/python3.10/site-packages
  - /nix/store/gmac070bibirfdf5wysk4ch5r2hvfnng-python3.10-ipython_genutils-0.2.0/lib/python3.10/site-packages
  - /nix/store/g2yv0ykq3baqrfza664rivffd96sicrw-python3.10-traitlets-5.4.0/lib/python3.10/site-packages
  - /nix/store/v1l9pwi92fbnjvsd0mpck2xd1cjigvsc-python3.10-decorator-5.1.1/lib/python3.10/site-packages
  - /nix/store/kx5anc33hbjj9cd9h3wll51p9b6lasb4-python3.10-six-1.16.0/lib/python3.10/site-packages
  - /nix/store/bfh8fgpcpskjiss1diq92yxvfhw0mvhg-python3.10-jupyter_core-4.11.2/lib/python3.10/site-packages
  - /nix/store/20ckyqfx5hf6riilgzv0sr26jva9ws9i-python3.10-Send2Trash-1.8.1b0/lib/python3.10/site-packages
[... 91 lines cut ...]
line 553, in start
    final_state, step_objects = self.run(
  File 
"/nix/store/cbyby0fylkjnbv36hzrbsmxc8srh4d41-python3-3.10.9-env/lib/python3.10/site-packages/openlane/flows/sequential.py", 
line 276, in run
    current_state = step.start(
  File "/nix/store/cbyby0fylkjnbv36hzrbsmxc8srh4d41-python3-3.10.9-env/lib/python3.10/site-packages/openlane/steps/step.py", 
line 768, in start
    views_updates, metrics_updates = self.run(state_in_result, **kwargs)
  File "/nix/store/cbyby0fylkjnbv36hzrbsmxc8srh4d41-python3-3.10.9-env/lib/python3.10/site-packages/openlane/steps/yosys.py", 
line 194, in run
    return super().run(state_in, env=env, **kwargs)
  File 
"/nix/store/cbyby0fylkjnbv36hzrbsmxc8srh4d41-python3-3.10.9-env/lib/python3.10/site-packages/openlane/steps/tclstep.py", line 
193, in run
    generated_metrics = self.run_subprocess(
  File "/nix/store/cbyby0fylkjnbv36hzrbsmxc8srh4d41-python3-3.10.9-env/lib/python3.10/site-packages/openlane/steps/step.py", 
line 906, in run_subprocess
    process = subprocess.Popen(
  File "/nix/store/65cp4izx3bllnwqn7c7dhrq9h9gmjkal-python3-3.10.9/lib/python3.10/subprocess.py", line 971, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/nix/store/65cp4izx3bllnwqn7c7dhrq9h9gmjkal-python3-3.10.9/lib/python3.10/subprocess.py", line 1847, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 7] Argument list too long: 'yosys'
```

urish, 2023-11-18:
Update: I was able to get the flow to complete successfully using the following workaround (instead of the one mentioned in the original comment):

```diff
diff --git a/openlane/steps/tclstep.py b/openlane/steps/tclstep.py
index 86d6069..68f046e 100644
--- a/openlane/steps/tclstep.py
+++ b/openlane/steps/tclstep.py
@@ -157,6 +157,8 @@ class TclStep(Step):
             filename = f"{self.config['DESIGN_NAME']}.{output.value.extension}"
             env[f"SAVE_{output.name}"] = os.path.join(self.step_dir, filename)
 
+        if "MACROS" in env: del env["MACROS"] # Workaround for https://github.com/efabless/openlane2/issues/272
+
         return env
 
     @protected
```

urish, 2023-11-21:
@donn would a minimal reproduction be useful here?

donn, 2023-11-21:
No, I have what I need- just thinking of a non-jank solution to this

`TclStep.prepare_env` arbitrarily excluding dataclasses would be weird

PR #394 (2024-03-04): Subprocess Enhancements
## Misc
* Various internal environment variables changed from `_lower_snake_case` to `_UPPER_SNAKE_CASE` for (relative) consistency
* Various handcrafted joins now use `TclStep.value_to_tcl` or `TclUtils.join`
* `openlane.common.TclUtils`
  * Empty strings now escaped as `""`
* `openlane.steps.Step`
  * Printing of last 10 lines of a file now uses a ring buffer instead of concatenating then splitting then joining
* `openlane.steps.TclStep` 
  * **API Break**: `value_to_tcl` no longer converts dataclasses to JSON, rather, they're converted to Tcl dicts
  * **API Break**: [... cut]
closed 2024-03-04 by donn