chipfoundry/openlane2 #289: Assert non-empty CLOCK_PORT exists

kareefardi, 2023-11-23:
### Description

Flow will continue normally. There will be warnings along the flow inside steps logs but I don't think that is sufficient. 

### Proposal

A bad CLOCK_PORT definition should force the flow to quit.

donn, 2023-11-23:
What about combinational IP? These do exist. Should we perhaps add a "COMBINATIONAL" variable to toggle this behavior?

kareefardi, 2023-11-27:
For combinational IPs the designer can define an empty clock port variable or even chose not to define it at all. However, if one chooses to define a clock port (non-empty value) we should assert it's existence as a definition of an undefined port will cause issues in the flow and will not be considered a critical fail.

donn, 2024-07-07:
I think an empty clock port is used in some corner cases, so I think this issue should be part of a wider discussion on clock and power domains in OpenLane.

kareefardi, 2024-07-07:
The issue has nothing to do with empty clock port definitions

donn, 2024-10-07:
With OpenLane 3, the SDC will be the single source of truth for clock info. So, we'll create a new issue for that.