chipfoundry/openlane2 #293: [STA-0376] cell not found

Andrea-Spitale, 2023-12-03:
### Description

Hello,
I've recently updated openlane2 to the latest available version as of 2nd december 2023.
I can run the classic flow on any design, but I keep getting the warnings below

[WARNING STA-0376] cell 'sky130_fd_sc_hd__xnor3_4' not found.

The same warning applies for some other cells as well, even though I think I see the warning only when using sky130 pdk. These warnings are seen in the stages of the logs I am reporting below

13-openroad-cutrows/openroad-cutrows.log
14-openroad-tapendcapinsertion/openroad-tapendcapinsertion.log
15-openroad-globalplacementskipio/openroad-globalplacementskipio.log
16-openroad-ioplacement/openroad-ioplacement.log
19-openroad-globalplacement/openroad-globalplacement.log
21-openroad-generatepdn/openroad-generatepdn.log
23-openroad-stamidpnr/openroad-stamidpnr.log
25-openroad-detailedplacement/openroad-detailedplacement.log
27-openroad-stamidpnr-1/openroad-stamidpnr-1.log
29-openroad-stamidpnr-2/openroad-stamidpnr-2.log
30-openroad-globalrouting/openroad-globalrouting.log
32-openroad-repairantennas/openroad-repairantennas.log
34-openroad-stamidpnr-3/openroad-stamidpnr-3.log
35-openroad-detailedrouting/openroad-detailedrouting.log
36-openroad-checkantennas/openroad-checkantennas.log
41-openroad-fillinsertion/openroad-fillinsertion.log
44-openroad-irdropreport/openroad-irdropreport.log

I am pretty sure I didn't update anything but openlane2. Anyone else is seeing the same? I've tried to trace where the warning comes from, and I also checked https://openroad.readthedocs.io/en/latest/main/README.html#understanding-warning-and-error-messages, as I believe the warning comes from openSTA, without luck.

### Expected Behavior

I guess those warnings should not be present. Could it be a pdk issue? I don't recall those warnings using my previous setup

### Environment report

```sh
Failed to get Docker info: Command '['docker', 'info', '--format', '{{json .}}']' returned non-zero exit status 1.
kernel: Linux
kernel_version: 5.15.133.1-microsoft-standard-WSL2
supported: True
distro: ubuntu
distro_version: 22.04
python_version: 3.10.9
python_path:
  - /nix/store/plag2nl7ql88c1xs6q4k5d6v88306frd-python3.10-openlane/bin
  - /nix/store/ns7p6q9gw667v41ysn750103v7dgxgwz-python3.10-click-8.1.3/lib/python3.10/site-packages
  - /nix/store/n0kq2qalwygww8h3pr7v6av7h7jxgd25-python3.10-cloup-1.0.1/lib/python3.10/site-packages
  - /nix/store/l8y1hkbki6pjbr1lm0k0iv51ii9864wz-python3.10-pyyaml-6.0/lib/python3.10/site-packages
  - /nix/store/gkphk53397gfy6mr5nhlqyg5cmgk304h-python3.10-rich-12.5.1/lib/python3.10/site-packages
  - /nix/store/i09vhwzivdhr017688ywhzjwz14xdhn9-python3.10-requests-2.28.1/lib/python3.10/site-packages
  - /nix/store/yrp4f59aa0a1kvrgzdm58v56q8h90604-python3.10-pcpp-1.21/lib/python3.10/site-packages
  - /nix/store/8iqa76maxlj9n6233fg6byl5rxvjinv7-python3.10-volare/lib/python3.10/site-packages
  - /nix/store/qvb8isbg9kgmjzkmm9i20r05hns807cj-python3.10-tkinter-3.10.9/lib/python3.10/site-packages
  - /nix/store/xkvl0krzk8l9ch8ksxkb9xmg2jjiszq7-python3.10-lxml-4.9.1/lib/python3.10/site-packages
  - /nix/store/cdgsnj5kmpawp52iramjc7craindq2hr-python3.10-deprecated-1.2.13/lib/python3.10/site-packages
  - /nix/store/aip2pp28ld9wcbqgwsgy3xybqhc21025-python3.10-immutabledict-2.2.2/lib/python3.10/site-packages
  - /nix/store/qll21g8p4421c0bw98qs8rg9gr64256g-python3.10-libparse/lib/python3.10/site-packages
  - /nix/store/c2ahwjcjnhi2r9yw42fa7l66gnvqsbp7-python3.10-psutil-5.9.3/lib/python3.10/site-packages
  - /nix/store/vp7ydsz6l7y724zhb8fcskjwc5ndchl2-yosys-0.23/lib/python3.10/site-packages
  - /nix/store/bgjw47kd190mg4v6rpq3sipsw74lfl4d-opensta/lib/python3.10/site-packages
  - /nix/store/xw09m82d7k08wpx26rn7h71lwv315hlb-openroad/lib/python3.10/site-packages
  - /nix/store/sl923qzcxy33s8wyg7l14qzlrlm6ngjk-klayout/lib/python3.10/site-packages
  - /nix/store/y3ymwmqpmarqqqxp537jsgiqf1mp5rqz-netgen/lib/python3.10/site-packages
  - /nix/store/1xfp6kmcah4pmic46v2s2nbgmm08ncv0-magic-vlsi/lib/python3.10/site-packages
  - /nix/store/z41wwdc6zlmxz0mgcjmg33gmdcxwsww4-iverilog-11.0/lib/python3.10/site-packages
  - /nix/store/6xvch03v46dh2h4m8kmilrhlssygxlwa-verilator/lib/python3.10/site-packages
  - /nix/store/378rxqdx6lh4swhw49ifbilc6hbk7h2r-tcl-8.6.11/lib/python3.10/site-packages
  - /nix/store/kcqiwya701lhcwhl1gci4cmjk7c5bcav-surelog-1.76/lib/python3.10/site-packages
  - /nix/store/pjjdz5sjrf2xk8vzf7461wbhkky00k4r-python3.10-cffi-1.15.1/lib/python3.10/site-packages
[... 133 lines cut ...]
[WARNING STA-0376] cell 'sky130_fd_sc_hd__lpflow_isobufsrckapwr_16' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_1' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_2' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_4' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_4' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_1' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_2' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_4' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__mux4_4' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__o21ai_0' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__o311ai_0' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__or2_0' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__probe_p_8' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__probec_p_8' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__xor3_1' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__xor3_2' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__xor3_4' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__xnor3_1' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__xnor3_2' not found.
[WARNING STA-0376] cell 'sky130_fd_sc_hd__xnor3_4' not found.
[INFO TAP-0004] Inserted 20 endcaps.
[INFO TAP-0005] Inserted 12 tapcells.
Setting global connections for newly added cells…
[INFO] Setting global connections...
```

kareefardi, 2023-12-04:
These are the cells found in the file `PNR_EXCLUSION_CELL_LIST`. This warning specifically happens during this method: https://github.com/efabless/openlane2/blob/08da726f70c8efcd2e71d3eca7595472c0e48bbd/openlane/scripts/openroad/common/io.tcl#L206 The cells are removed from the liberty that is read by OpenROAD steps but we also attempt to exclude them using the method referenced above (using `set_dont_use`). Since they are missing from the liberty, OpenSTA warns about it. So these warning are harmless. However, we can look into weather we need the double exclusion or not

Andrea-Spitale, 2023-12-04:
@kareefardi hi Kareem, thanks a lot! Now it makes sense, as they are harmless.

PR #339 (2024-01-11): Remove redundant warnings about missing cells in `OpenROAD` steps
Don't trim liberty files for OpenROAD steps. 

---

Fix #293
closed 2024-01-11 by donn