chipfoundry/openlane2 #295: LVS errors depending on SYNTH_AUTONAME value

7FM, 2023-12-05:
### Description

I am using the latest commit on master:
```
        my_openlane2 = let
          src = pkgs.fetchFromGitHub {
            owner = "efabless";
            repo = "openlane2";
            rev = "08da726f70c8efcd2e71d3eca7595472c0e48bbd";
            sha256 = "sha256-pByrF+SPC+CSDaj4K/6LomQ6X6bCNBnaLG2qnUUXkbM=";
          };
        in import "${src}/default.nix" { pkgs = openlane2_pkgs; system = pkgs.system; };
```
For the attached project I am getting LVS errors when using `"SYNTH_AUTONAME": 1`.
With `"SYNTH_AUTONAME": 0` I do not get such an error.

### Expected Behavior

LVS result should be not depend on the value of `SYNTH_AUTONAME`

### Environment report

```sh
kernel: Linux
kernel_version: 6.1.57
supported: True
distro: nixos
distro_version: 23.11
python_version: 3.10.12
python_path:
  - /nix/store/374qzawi30mmngj7hqkiji5ia23zv935-python3.10-openlane/lib/python3.10/site-packages/openlane
  - /nix/store/7yiqh8h334alb7pf340p3pjindqh5f5n-python3-3.10.12-env/lib/python3.10/site-packages
  - /nix/store/374qzawi30mmngj7hqkiji5ia23zv935-python3.10-openlane/lib/python3.10/site-packages
  - /nix/store/ns7p6q9gw667v41ysn750103v7dgxgwz-python3.10-click-8.1.3/lib/python3.10/site-packages
  - /nix/store/65cp4izx3bllnwqn7c7dhrq9h9gmjkal-python3-3.10.9/lib/python3.10/site-packages
  - /nix/store/n0kq2qalwygww8h3pr7v6av7h7jxgd25-python3.10-cloup-1.0.1/lib/python3.10/site-packages
  - /nix/store/l8y1hkbki6pjbr1lm0k0iv51ii9864wz-python3.10-pyyaml-6.0/lib/python3.10/site-packages
  - /nix/store/gkphk53397gfy6mr5nhlqyg5cmgk304h-python3.10-rich-12.5.1/lib/python3.10/site-packages
  - /nix/store/k5b7jpzi9dvg3x7h2a90df14aq8wh6bl-python3.10-commonmark-0.9.1/lib/python3.10/site-packages
  - /nix/store/8zzhma74gr86jzakxd8f1pzlrdf8snzx-python3.10-future-0.18.2/lib/python3.10/site-packages
  - /nix/store/cpv31grqdbxvsq7jx4kdgbxg2rsxkkfy-python3.10-pygments-2.13.0/lib/python3.10/site-packages
  - /nix/store/9ljcxsc2gqnf5gysl45993v4v88iwzz7-python3.10-docutils-0.19/lib/python3.10/site-packages
  - /nix/store/i09vhwzivdhr017688ywhzjwz14xdhn9-python3.10-requests-2.28.1/lib/python3.10/site-packages
  - /nix/store/6bkqcc8d265j605rzrb92lr2pn82442h-python3.10-brotlicffi-1.0.9.2/lib/python3.10/site-packages
  - /nix/store/pjjdz5sjrf2xk8vzf7461wbhkky00k4r-python3.10-cffi-1.15.1/lib/python3.10/site-packages
  - /nix/store/6i9gf8hz2h9as3lxsraysv7qjh102g67-python3.10-pycparser-2.21/lib/python3.10/site-packages
  - /nix/store/8k41q7l3ff3i4lz0drn34w3phk8fw49z-python3.10-certifi-2022.12.07/lib/python3.10/site-packages
  - /nix/store/2s5sa0zibgjshn0s3hhjna4cql3flsnn-python3.10-charset-normalizer-2.1.0/lib/python3.10/site-packages
  - /nix/store/wl8rwlabb29g5gqpy0dsqf9g4xagnc2j-python3.10-idna-3.4/lib/python3.10/site-packages
  - /nix/store/3whxw66aiiixk326lj6h63cxbv0cpilg-python3.10-urllib3-1.26.12/lib/python3.10/site-packages
  - /nix/store/vwrdsq0v60p5gv5v6vlydmy045ihf29q-python3.10-brotli-1.0.9/lib/python3.10/site-packages
  - /nix/store/ah2w40lqazbkjd4cga7binijihm8dvb3-python3.10-pysocks-1.7.1/lib/python3.10/site-packages
  - /nix/store/yrp4f59aa0a1kvrgzdm58v56q8h90604-python3.10-pcpp-1.21/lib/python3.10/site-packages
  - /nix/store/8iqa76maxlj9n6233fg6byl5rxvjinv7-python3.10-volare/lib/python3.10/site-packages
  - /nix/store/filc9bjiifd21l66dicrwr4y3rbza01x-python3.10-zstandard-0.19.0/lib/python3.10/site-packages
  - /nix/store/qvb8isbg9kgmjzkmm9i20r05hns807cj-python3.10-tkinter-3.10.9/lib/python3.10/site-packages
  - /nix/store/xkvl0krzk8l9ch8ksxkb9xmg2jjiszq7-python3.10-lxml-4.9.1/lib/python3.10/site-packages
  - /nix/store/cdgsnj5kmpawp52iramjc7craindq2hr-python3.10-deprecated-1.2.13/lib/python3.10/site-packages
  - /nix/store/hjhzx3ic251y5gjigcxiyp2chdzd8s9k-python3.10-wrapt-1.13.3/lib/python3.10/site-packages
  - /nix/store/aip2pp28ld9wcbqgwsgy3xybqhc21025-python3.10-immutabledict-2.2.2/lib/python3.10/site-packages
  - /nix/store/qll21g8p4421c0bw98qs8rg9gr64256g-python3.10-libparse/lib/python3.10/site-packages
  - /nix/store/c2ahwjcjnhi2r9yw42fa7l66gnvqsbp7-python3.10-psutil-5.9.3/lib/python3.10/site-packages
  - /nix/store/vp7ydsz6l7y724zhb8fcskjwc5ndchl2-yosys-0.23/lib/python3.10/site-packages
  - /nix/store/49pg4ljx6qandwdr6mbc8w8k8fh0dd2j-python3.10-setuptools-65.3.0/lib/python3.10/site-packages
  - /nix/store/1r6n7v2wam7gkr18gxccpg7p5ywgw551-python3-3.10.12/lib/python310.zip
  - /nix/store/1r6n7v2wam7gkr18gxccpg7p5ywgw551-python3-3.10.12/lib/python3.10
  - /nix/store/1r6n7v2wam7gkr18gxccpg7p5ywgw551-python3-3.10.12/lib/python3.10/lib-dynload
  - /nix/store/1r6n7v2wam7gkr18gxccpg7p5ywgw551-python3-3.10.12/lib/python3.10/site-packages
tkinter: True
container_info:
  engine: docker
  version: 20.10.26
  conmon: False
  rootless: True
nix_info:
  version: nix-env (Nix) 2.17.0
  multi_user: True
  sandbox: True
  channels: None
  nixpkgs: /nix/store/4smdp3v5a9bcg1hpdhl2hs70rq385ykq-source
```

### Reproduction material

`openlane.steps create-reproducible -d .` output:
[reproducible_part1.zip](https://github.com/efabless/openlane2/files/13573557/reproducible_part1.zip)
[reproducible_part2.zip](https://github.com/efabless/openlane2/files/13573558/reproducible_part2.zip)
The whole project:
[project.zip](https://github.com/efabless/openlane2/files/13573563/project.zip)

### Relevant log output

```sh
Final result:
Netlists do not match.
Logging to file "runs/RUN_2023-12-05_21-45-17/51-netgen-lvs/lvs.rpt" disabled
LVS Done.
──────────────────────────────────────────────────────────────────────────────────────── Layout vs. Schematic Error Checker ────────────────────────────────────────────────────────────────────────────────────────
[22:55:53] ❌  5946 LVS errors found. - deferred                                                                                                                                                       checker.py:69
           Gating variable for step 'Yosys.EQY' set to 'False'- the step will be skipped.                                                                                                          sequential.py:250
           Skipping step 'Equivalence Check'…                                                                                                                                                      sequential.py:258
           ❌  The following deferred step errors have been encountered:                                                                                                                           sequential.py:287
           ❌  5946 LVS errors found.                                                                                                                                                              sequential.py:289
           ❌  OpenLane will now quit.                                                                                                                                                               __main__.py:140
```

donn, 2023-12-06:
We'll investigate.

donn, 2023-12-11:
So the longest-named wire is `isax_xcorevsimd.inst_CV_DOTSP_SC_H.mockup_2_0[31]_sky130_fd_sc_hd__dfxtp_2_Q_D_sky130_fd_sc_hd__o22a_2_X_B1_sky130_fd_sc_hd__nor2_2_Y_B_sky130_fd_sc_hd__xnor2_2_Y_B_sky130_fd_sc_hd__xnor2_2_Y_A_sky130_fd_sc_hd__o21a_2_X_A2_sky130_fd_sc_hd__a21oi_2_Y_A1_sky130_fd_sc_hd__a21o_2_X_A2_sky130_fd_sc_hd__xnor2_2_Y_A_sky130_fd_sc_hd__o21bai_2_Y_B1_N_sky130_fd_sc_hd__o21a_2_X_A2_sky130_fd_sc_hd__and2_2_X_B_sky130_fd_sc_hd__mux2_2_X_A1_sky130_fd_sc_hd__mux2_2_X_A0_sky130_fd_sc_hd__nand2_2_Y_A_sky130_fd_sc_hd__inv_2_A_Y` and I'm honestly surprised more things didnt break

Still investigating