chipfoundry/openlane2 #303: OpenROAD crashes in repair_design

mo-hosni, 2023-12-13:
### Description

OpenROAD crashes in repair design after or during `Setting RC values…`

### Expected Behavior

repair_design finishing successfully or reporting an error.

### Environment report

```sh
ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
this path will be fetched (0.00 MiB download, 0.00 MiB unpacked):
  /nix/store/fyvbj5mvrvdnvhbx6hc977wsgl33pk3i-nix-info
copying path '/nix/store/fyvbj5mvrvdnvhbx6hc977wsgl33pk3i-nix-info' from 'https://cache.nixos.org'...
ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
kernel: Linux
kernel_version: 4.19.0-23-cloud-amd64
supported: True
distro: debian
distro_version: 10
python_version: 3.12.0
python_path:
  - /home/hosni/Cheetah_V3/openlane2/openlane
  - /home/hosni/.pyenv/versions/3.12.0/lib/python312.zip
  - /home/hosni/.pyenv/versions/3.12.0/lib/python3.12
  - /home/hosni/.pyenv/versions/3.12.0/lib/python3.12/lib-dynload
  - /home/hosni/.pyenv/versions/3.12.0/lib/python3.12/site-packages
tkinter: True
container_info:
  engine: docker
  version: 20.10.8
  conmon: False
  rootless: False
nix_info:
  version: nix-env (Nix) 2.18.1
  multi_user: True
  sandbox: True
  channels:
    root:
      - nixpkgs
  nixpkgs: /nix/var/nix/profiles/per-user/root/channels/nixpkgs
```

### Reproduction material

The ejected script for OpenROAD is there. OpenROAD version used is `6f9b2bb8b808b1bb5831d4525d868212ae50517a`.
[reproducible.tar.partaa.gz](https://github.com/efabless/openlane2/files/13658804/reproducible.tar.partaa.gz)
[reproducible.tar.partab.gz](https://github.com/efabless/openlane2/files/13658805/reproducible.tar.partab.gz)

### Relevant log output

```sh
[INFO] Setting RC values…
Signal 11 received
Stack trace:
 0# 0x0000000000CA0EDE in /nix/store/xw09m82d7k08wpx26rn7h71lwv315hlb-openroad/bin/openroad
 1# 0x00007F7FBBCB7BF0 in /nix/store/9xfad3b5z4y00mzmk2wnn4900q0qmxns-glibc-2.35-224/lib/libc.so.6
 2# odb::dbObject::getObjectType() const in /nix/store/xw09m82d7k08wpx26rn7h71lwv315hlb-openroad/bin/openroad
 3# sta::dbNetwork::staToDb(sta::Instance const*) const in /nix/store/xw09m82d7k08wpx26rn7h71lwv315hlb-openroad/bin/openroad
 4# rsz::Resizer::bufferInput(sta::Pin const*, sta::LibertyCell*) in /nix/store/xw09m82d7k08wpx26rn7h71lwv315hlb-openroad/bin/openroad
 5# rsz::Resizer::bufferInputs() in /nix/store/xw09m82d7k08wpx26rn7h71lwv315hlb-openroad/bin/openroad
 6# 0x0000000000DF64BE in /nix/store/xw09m82d7k08wpx26rn7h71lwv315hlb-openroad/bin/openroad
 7# TclNRRunCallbacks in /nix/store/378rxqdx6lh4swhw49ifbilc6hbk7h2r-tcl-8.6.11/lib/libtcl.so
 8# 0x00007F7FBEC89BC6 in /nix/store/378rxqdx6lh4swhw49ifbilc6hbk7h2r-tcl-8.6.11/lib/libtcl.so
 9# Tcl_EvalEx in /nix/store/378rxqdx6lh4swhw49ifbilc6hbk7h2r-tcl-8.6.11/lib/libtcl.so
10# Tcl_Eval in /nix/store/378rxqdx6lh4swhw49ifbilc6hbk7h2r-tcl-8.6.11/lib/libtcl.so
11# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /nix/store/xw09m82d7k08wpx26rn7h71lwv315hlb-openroad/bin/openroad
12# ord::tclAppInit(Tcl_Interp*) in /nix/store/xw09m82d7k08wpx26rn7h71lwv315hlb-openroad/bin/openroad
13# Tcl_MainEx in /nix/store/378rxqdx6lh4swhw49ifbilc6hbk7h2r-tcl-8.6.11/lib/libtcl.so
14# main in /nix/store/xw09m82d7k08wpx26rn7h71lwv315hlb-openroad/bin/openroad
15# 0x00007F7FBBCA324E in /nix/store/9xfad3b5z4y00mzmk2wnn4900q0qmxns-glibc-2.35-224/lib/libc.so.6
16# __libc_start_main in /nix/store/9xfad3b5z4y00mzmk2wnn4900q0qmxns-glibc-2.35-224/lib/libc.so.6
17# _start in /nix/store/xw09m82d7k08wpx26rn7h71lwv315hlb-openroad/bin/openroad
./run.sh: line 108:  1145 Segmentation fault      openroad -exit -no_splash -metrics ./or_metrics_out.json ./scripts/openroad/repair_design.tcl
```

kareefardi, 2023-12-13:
Created an issue in OpenROAD https://github.com/The-OpenROAD-Project/OpenROAD/issues/4374

kareefardi, 2023-12-13:
Fixed in OpenROAD upstream.

PR #310 (2023-12-26): Update OpenROAD
* Updated OpenROAD to `75f2f32`
  * Added some conveniences for manual compilation to the Nix derivation

## CI
* Added unit test for #303

---
Resolves #303 
Depends on https://github.com/efabless/openlane2-ci-designs/pull/15
Depends on https://github.com/efabless/openlane2-step-unit-tests/pull/18
closed 2023-12-26 by donn