chipfoundry/openlane2 #327: Antenna repaire crash in OpenROAD

mo-hosni, 2024-01-02:
### Description

OpenROAD crashes while fixing antenna violations

### Expected Behavior

OpenROAD finishing successfully or reporting an error

### Environment report

```sh
warning: Nix search path entry '/nix/var/nix/profiles/per-user/hosni/channels/nixpkgs' does not exist, ignoring
warning: Nix search path entry '/nix/var/nix/profiles/per-user/hosni/channels' does not exist, ignoring
error:
       … <borked>

         at «none»:0: (source not available)

       … while calling the 'import' builtin

         at «string»:1:18:

            1| {...}@args: with import <nixpkgs> args; (pkgs.runCommandCC or pkgs.runCommand) "shell" { buildInputs = [ (nix-info) ]; } ""
             |                  ^

       (stack trace truncated; use '--show-trace' to show the full trace)

       error: file 'nixpkgs' was not found in the Nix search path (add it using $NIX_PATH or -I)

       at «none»:0: (source not available)
Failed to get Docker info: Command '['nix-shell', '-p', 'nix-info', '--run', 'nix-info -m']' returned non-zero exit status 1.
kernel: Linux
kernel_version: 6.2.0-39-generic
supported: True
distro: ubuntu
distro_version: 23.04
python_version: 3.11.4
python_path:
  - /home/hosni/caravel-3/openlane2/openlane
  - /usr/lib/python311.zip
  - /usr/lib/python3.11
  - /usr/lib/python3.11/lib-dynload
  - /home/hosni/.local/lib/python3.11/site-packages
  - /usr/local/lib/python3.11/dist-packages
  - /usr/lib/python3/dist-packages
  - /usr/lib/python3.11/dist-packages
tkinter: False
container_info:
  engine: docker
  version: 24.0.7
  conmon: False
  rootless: False
nix_info: None
```

### Reproduction material

[groute-post-antenna-repair.partaa.gz](https://github.com/efabless/openlane2/files/13810107/groute-post-antenna-repair.partaa.gz)
[groute-post-antenna-repair.partab.gz](https://github.com/efabless/openlane2/files/13810108/groute-post-antenna-repair.partab.gz)
[groute-post-antenna-repair.partac.gz](https://github.com/efabless/openlane2/files/13810109/groute-post-antenna-repair.partac.gz)
[groute-post-antenna-repair.partad.gz](https://github.com/efabless/openlane2/files/13810110/groute-post-antenna-repair.partad.gz)

### Relevant log output

```sh
INFO GRT-0012] Found 286 antenna violations.
[INFO GRT-0015] Inserted 2078 diodes.
[INFO GRT-0054] Using detailed placer to place 82 diodes.
[INFO GRT-0009] rerouting 30474 nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 51
[INFO GRT-0006] Repairing antennas, iteration 2.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0012] Found 64 antenna violations.
[INFO GRT-0015] Inserted 353 diodes.
[INFO GRT-0054] Using detailed placer to place 1 diodes.
[INFO GRT-0009] rerouting 942 nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 43
[INFO GRT-0006] Repairing antennas, iteration 3.
[INFO GRT-0043] No OR_DEFAULT vias defined.
Signal 11 received
Stack trace:
 0# 0x0000563A5E03FBFF in openroad
 1# 0x00007FE331E55A70 in /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
 2# odb::dbBox::getBox() in openroad
 3# odb::dbWirePathItr::getNextShape(odb::dbWirePathShape&) in openroad
 4# odb::tmg_conn::loadWire(odb::dbWire*) in openroad
 5# odb::tmg_conn::analyzeNet(odb::dbNet*) in openroad
 6# odb::orderWires(utl::Logger*, odb::dbNet*) in openroad
 7# grt::RepairAntennas::makeNetWire(odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >&, std::map<int, odb::dbTechVia*, std::less<int>, std::allocator<std::pair<int const, odb::dbTechVia*> > >&) in openroad
 8# grt::RepairAntennas::makeNetWires(std::map<odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >, grt::cmpById, std::allocator<std::pair<odb::dbNet* const, std::vector<grt::GSegment, std::allocator<grt::GSegment> > > > >&, int) in openroad
 9# grt::RepairAntennas::checkAntennaViolations(std::map<odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >, grt::cmpById, std::allocator<std::pair<odb::dbNet* const, std::vector<grt::GSegment, std::allocator<grt::GSegment> > > > >&, int, odb::dbMTerm*, float) in openroad
10# grt::GlobalRouter::repairAntennas(odb::dbMTerm*, int, float) in openroad
11# 0x0000563A5E62C1D0 in openroad
12# TclNRRunCallbacks in /nix/store/2wmwl9g0204202qpj7p53w56rmi76m92-tcl-8.6.13/lib/libtcl.so
13# 0x00007FE335E0E228 in /nix/store/2wmwl9g0204202qpj7p53w56rmi76m92-tcl-8.6.13/lib/libtcl.so
14# Tcl_EvalEx in /nix/store/2wmwl9g0204202qpj7p53w56rmi76m92-tcl-8.6.13/lib/libtcl.so
15# Tcl_Eval in /nix/store/2wmwl9g0204202qpj7p53w56rmi76m92-tcl-8.6.13/lib/libtcl.so
16# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
17# ord::tclAppInit(Tcl_Interp*) in openroad
18# Tcl_MainEx in /nix/store/2wmwl9g0204202qpj7p53w56rmi76m92-tcl-8.6.13/lib/libtcl.so
19# main in openroad
20# 0x00007FE331E3FFCE in /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
21# __libc_start_main in /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
22# _start in openroad
```

kareefardi, 2024-01-02:
Reproduced uasing latest OpenROAD. https://github.com/The-OpenROAD-Project/OpenROAD/issues/4471

kareefardi, 2024-02-29:
Fixed by updating OpenROAD in https://github.com/efabless/openlane2/pull/391

closed 2024-02-29 by kareefardi