chipfoundry/openlane2 #358: Deep instances

kareefardi, 2024-01-24:
### Description

In a config, a user should be able to specify an instance of MACRO, deeper in the hierarchy, inside another instance of a macro. At the moment, any instance must have a location entry, while deeper instances are already placed. 

### Proposal

Remove hard requirement on location for instances.

donn, 2024-01-25:
I'm not sure what this achieves exactly. If a macro is placed within another macro, it should be abstracted by said Macro. This seems like a workaround for macros that were hardened improperly.

kareefardi, 2024-01-28:
> I'm not sure what this achieves exactly. If a macro is placed within another macro, it should be abstracted by said Macro. This seems like a workaround for macros that were hardened improperly.

This achieves more comprehensive `STA`.

kareefardi, 2024-02-29:
Fixed by https://github.com/efabless/openlane2/pull/359

closed 2024-02-29 by kareefardi