chipfoundry/openlane2 #360: OpenGUI: missing required input 'ODB  on Nix

hakan-demirli, 2024-01-29:
### Description

In Openlane I used to use `flow.tcl --interactive`command to visualize heatmap if the congestion is too high on global routing.

I want to do the same on openlane2. I run the following command since gui is defined in OpenROAD class.
```python
python3 -m openlane ./openlane/examples/spm/config.json -f OpenInOpenROAD
```
https://github.com/efabless/openlane2/blob/499367593546b27c7583f6aaaf9bdfa4313ef56f/openlane/steps/openroad.py#L1979

### Expected Behavior

Open up a GUI so I can visualize the heatmap.

### Environment report

```sh
Failed to get Docker info: [Errno 2] No such file or directory: 'docker'
kernel: Linux
kernel_version: 6.7.1
supported: True
distro: nixos
distro_version: 24.05
python_version: 3.11.6
python_path:
  - /nix/store/xc12z7izn7psa0af9xqm6drzda7x5xq1-python3.11-openlane/bin
  - /nix/store/xanckpmyylk3d0ygpy11d223r1q77f96-python3-3.11.6-env/lib/python3.11/site-packages
  - /nix/store/qp5zys77biz7imbk6yy85q5pdv7qk84j-python3-3.11.6/lib/python311.zip
  - /nix/store/qp5zys77biz7imbk6yy85q5pdv7qk84j-python3-3.11.6/lib/python3.11
  - /nix/store/qp5zys77biz7imbk6yy85q5pdv7qk84j-python3-3.11.6/lib/python3.11/lib-dynload
  - /nix/store/qp5zys77biz7imbk6yy85q5pdv7qk84j-python3-3.11.6/lib/python3.11/site-packages
  - /nix/store/xc12z7izn7psa0af9xqm6drzda7x5xq1-python3.11-openlane/lib/python3.11/site-packages
  - /nix/store/2z0jalw27lp6zm642aqvnlydlqblhcdf-python3.11-click-8.1.7/lib/python3.11/site-packages
  - /nix/store/m1a3bk12cjgk6wq445pi7xch3v2gx8ql-python3.11-cloup-3.0.2/lib/python3.11/site-packages
  - /nix/store/a8z71acldic6hqdbd18sd81sqcqgygd3-python3.11-pyyaml-6.0.1/lib/python3.11/site-packages
  - /nix/store/jlb8fjb88bx2adivdz8pq8cx5aj5615h-python3.11-rich-13.5.2/lib/python3.11/site-packages
  - /nix/store/8x0qa1j4z2907a9pvvf9lwzs6n58i5m5-python3.11-markdown-it-py-3.0.0/lib/python3.11/site-packages
  - /nix/store/69l2fjl76rad6fz5n5v38px2kiwcz5x8-python3.11-mdurl-0.1.2/lib/python3.11/site-packages
  - /nix/store/ljd22jdzx6kj0hj7wsnhr71gqa4iqj88-python3.11-pygments-2.16.1/lib/python3.11/site-packages
  - /nix/store/y8yq448xka665bqyd5gvdhajqdbsnyys-python3.11-setuptools-68.2.2/lib/python3.11/site-packages
  - /nix/store/71js2p8zx17fqgq2m1b5f7q92ig0v7nn-python3.11-requests-2.31.0/lib/python3.11/site-packages
  - /nix/store/6jj18jqyik8k06plad2hm9fwcrnhpxf3-python3.11-brotlicffi-1.1.0.0/lib/python3.11/site-packages
  - /nix/store/qxsnv4322wqi9w629b572k898viw756h-python3.11-cffi-1.16.0/lib/python3.11/site-packages
  - /nix/store/r0mzj4k7gzx3bp4mbnr69x6if6lxxsb1-python3.11-pycparser-2.21/lib/python3.11/site-packages
  - /nix/store/hk5xz5kyhr1vaq2p65990c2fxw7mxlmp-python3.11-certifi-2023.07.22/lib/python3.11/site-packages
  - /nix/store/g3kzbrnirxky2w4g3n67p01745pa8xcc-python3.11-charset-normalizer-3.2.0/lib/python3.11/site-packages
  - /nix/store/g3w192qs0avwjaw776yiyaq5l0dcbwdp-python3.11-idna-3.4/lib/python3.11/site-packages
  - /nix/store/71236y06cag3yb637q5jj358ci6f20ah-python3.11-urllib3-2.0.7/lib/python3.11/site-packages
  - /nix/store/bnadfkpcdkmrzymnpyj8x2bdxq1f6kdw-python3.11-pcpp-1.30/lib/python3.11/site-packages
  - /nix/store/i41k9kc0s446fw5vl4k05vfq3prw06hv-python3.11-volare/lib/python3.11/site-packages
  - /nix/store/i46q5s4xj3700m2srcx4rd2bv2x6w51h-python3.11-httpx-0.25.0/lib/python3.11/site-packages
  - /nix/store/6ckbxrfj5q372da9svdh7hnh67vbwyhc-python3.11-httpcore-0.18.0/lib/python3.11/site-packages
  - /nix/store/43sv2g5lsl58md1lzidlx1g7xam2mpqp-python3.11-anyio-4.0.0/lib/python3.11/site-packages
  - /nix/store/4m6453i7as66sdmkck8fiwk793lx1b29-python3.11-sniffio-1.3.0/lib/python3.11/site-packages
  - /nix/store/1c9w7989rjr417rs50bxsrinjf63mz8n-python3.11-h11-0.14.0/lib/python3.11/site-packages
  - /nix/store/q9c0733drxk1zv5j60hq0lz61g395dic-python3.11-rfc3986-2.0.0/lib/python3.11/site-packages
  - /nix/store/i24kszh5ajb03gwhjlmb1df6sm0s0x72-python3.11-zstandard-0.21.0/lib/python3.11/site-packages
  - /nix/store/608w6rc3xa1mj44ppara7pnmr6z6kbpn-python3.11-tkinter-3.11.6/lib/python3.11/site-packages
  - /nix/store/k2xqs2rp3kyb0pbq1sn9v0sh05q0cmvc-python3.11-lxml-4.9.3-3/lib/python3.11/site-packages
  - /nix/store/vp0ivbg65mnsjvfgyran0x1bzsf3x66l-python3.11-deprecated-1.2.14/lib/python3.11/site-packages
  - /nix/store/bj6g1njs2z0ivrwlpkls9qr1ar1psjaw-python3.11-wrapt-1.14.1/lib/python3.11/site-packages
  - /nix/store/cghmw4316dgv9f86y6prdybi50v1f8sn-python3.11-immutabledict-3.0.0/lib/python3.11/site-packages
  - /nix/store/s0xm1cjrghaa6g11b8nivv64ga6bpwwd-python3.11-libparse/lib/python3.11/site-packages
  - /nix/store/a9v9j1mwxp5ij7yijfvahm30bfadxfr3-python3.11-ioplace_parser/lib/python3.11/site-packages
  - /nix/store/1z7vs2crryc63akgcyzyq9is33p7y230-python3.11-antlr4-python3-runtime-4.10.1/lib/python3.11/site-packages
  - /nix/store/8wkz5a9z1ppwjvr54375k69avhkmz4nc-python3.11-psutil-5.9.6/lib/python3.11/site-packages
  - /nix/store/6bxns8xhf27bimgb5sfzxmk89pjfmycd-klayout-pymod/lib/python3.11/site-packages
tkinter: True
container_info: None
nix_info:
  version: nix-env (Nix) 2.18.1
  multi_user: True
  sandbox: True
  channels: None
  nixpkgs: /nix/var/nix/profiles/per-user/root/channels/nixos
```

### Reproduction material

499367593546b27c7583f6aaaf9bdfa4313ef56f is the commit id.
I have used nix environment and I am also on NixOS. I think it should be reproducable. Otherwise something is missing in flake.nix

### Relevant log output

```sh
[23:03:08] INFO     Loading the design configuration file has generated the following warnings:                                                    config.py:671
...
[23:03:08] WARNING  Key 'VERILOG_FILES' provided is unused by the current flow.                                                                    config.py:675
[23:03:08] INFO     Starting a new run of the 'Opening in OpenROAD' flow with the tag 'RUN_2024-01-29_23-03-08'.                                     flow.py:525
[23:03:08] INFO     Starting…                                                                                                                  sequential.py:222
───────────────────────────────────────────────────────────────────────── Open In GUI ──────────────────────────────────────────────────────────────────────────
[23:03:08] VERBOSE  Running 'OpenROAD.OpenGUI'… (Log: ./openlane/examples/preg/runs/RUN_2024-01-29_23-03-08/1-openroad-opengui/openroad-opengui.log) step.py:878
[23:03:08] ERROR    The flow has encountered an unexpected error: OpenGUI: missing required input 'ODB'                                          __main__.py:146
Traceback (most recent call last):
  File "/home/emre/Desktop/openlane2/openlane/flows/sequential.py", line 266, in run
    current_state = step.start(
                    ^^^^^^^^^^^
  File "/home/emre/Desktop/openlane2/openlane/steps/step.py", line 900, in start
    raise StepException(
openlane.steps.step.StepException: OpenGUI: missing required input 'ODB'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/emre/Desktop/openlane2/openlane/__main__.py", line 135, in run
    flow.start(
  File "/home/emre/Desktop/openlane2/openlane/flows/flow.py", line 555, in start
    final_state, step_objects = self.run(
                                ^^^^^^^^^
  File "/home/emre/Desktop/openlane2/openlane/flows/sequential.py", line 271, in run
    raise FlowException(str(e))
openlane.flows.flow.FlowException: OpenGUI: missing required input 'ODB'
```

kareefardi, 2024-01-30:
@hakan-demirli There should be a message similar to this:
```
15:19:14] VERBOSE  Using state at                                                                                 flow.py:526
                    './openlane/examples/spm/runs/v1/61-misc-reportmanufacturability/stat
                    e_out.json'.
```
Indicating that the flow accepted `--run-tag v1`. Are you sure you were passing this argument in the output above?

donn, 2024-01-30:
> I assume nix shell is reproducible. Otherwise something is missing in flake.nix

For future reference: that's different. For issues that involve bugs in OpenLane, we typically require the design files so we can isolate and fix the issue quickly. Nix eliminates variance in your environment, not in your design files. :)

---

As @kareefardi said, make sure you pass `--run-tag v1` to both the initial run and `OpenInOpenROAD`, as the log output here implies that the run tag v1 was not found.

hakan-demirli, 2024-01-30:
I have solved all superfluous errors. The real issue was Qt dynamic plugin loading.
```Cannot mix incompatible Qt library (5.15.12) with this library (5.15.11)```
I missed the error first time and performed other operations which evolved into the error given in the first message.

This was a nixpkgs bug which has fixed a long time ago with the introduction of [wrapQtAppsHook](https://github.com/NixOS/nixpkgs/pull/54525). And, you guys are using wrapQtAppsHook. So, no problem on the openlane2 side.

I assume this is a Nixos bug. There are other peoples on nixos forums complaining about the same issue on different packages. I have changed the nixpkgs of the flake to unstable which resolved the issue.
<details>
  <summary>flake.nix</summary>
...
    nixpkgs.url = "github:NixOS/nixpkgs/nixpkgs-unstable";
...
</details>
<details>
  <summary>flake.lock</summary>
    {
  "nodes": {
    "flake-compat": {
      "locked": {
        "narHash": "sha256-kvjfFW7WAETZlt09AgDn1MrtKzP7t90Vf7vypd3OL1U=",
        "rev": "0f9255e01c2351cc7d116c072cb317785dd33b33",
        "revCount": 57,
        "type": "tarball",
        "url": "https://api.flakehub.com/f/pinned/edolstra/flake-compat/1.0.1/018afb31-abd1-7bff-a5e4-cff7e18efb7a/source.tar.gz"
      },
      "original": {
        "type": "tarball",
        "url": "https://flakehub.com/f/edolstra/flake-compat/1.tar.gz"
      }
    },
    "nixpkgs": {
      "locked": {
        "lastModified": 1706487304,
        "narHash": "sha256-LE8lVX28MV2jWJsidW13D2qrHU/RUUONendL2Q/WlJg=",
        "owner": "NixOS",
        "repo": "nixpkgs",
        "rev": "90f456026d284c22b3e3497be980b2e47d0b28ac",
        "type": "github"
      },
      "original": {
        "owner": "NixOS",
        "ref": "nixpkgs-unstable",
        "repo": "nixpkgs",
        "type": "github"
      }
    },
    "root": {
      "inputs": {
        "flake-compat": "flake-compat",
        "nixpkgs": "nixpkgs"
      }
    }
  },
  "root": "root",
  "version": 7
}

</details>

[image]

hakan-demirli, 2024-02-14:
Related: https://github.com/NixOS/nixpkgs/issues/86369