chipfoundry/openlane2 #368: nix installation runtime (~4-5 hrs and counting)

cadnav, 2024-02-03:
### Description

This is not really a bug, but perhaps many people will have the same question.  What is the expected runtime for nix installation of openlane2? It's been running for ~4-5 hours and a lot of the time seems to be going in a loop.  Would be good to get a heads up in documentation so we know what to expect.  Other than that, thank you so much for the phenomenal work. I'm a huge fan and am sorry to file an issue. You guys are the sh***!

### Expected Behavior

shorter installation time / documented expectation

### Environment report

```sh
`
```

### Reproduction material

nix installation

### Relevant log output

```sh
too long to paste
```

donn, 2024-02-03:
Hi @cadnav, thank you for the kind words!

It really depends on your internet connection but also if you forget to configure Cachix (takes me about ~15 minutes.) If you forget to configure Cachix, it goes from a simple download to your computer *trying to build every single OpenLane utility*, which can take hours.

mole99, 2024-02-06:
Interestingly, I too followed the instructions [here](https://openlane2.readthedocs.io/en/latest/getting_started/common/nix_installation/installation_linux.html) (including setting up the binary cache) and it still builds all of the tools on my system, which takes a while. Maybe I am doing something wrong, but I completely removed and reinstalled Nix and still got the same behavior ^^ I'm using debian bookworm.

donn, 2024-02-11:
@mole99 That's really weird (and concerning.) Would you mind sharing the output of the following commands? It'd be a huge help:

* `uname -a`
* `nix-env -f "<nixpkgs>" -iA cachix`
* `cachix use openlane`
* `nix develop --accept-flake-config` (from the openlane directory-- just check if it pulls from the cache or also attempts to build tools, no need to wait for it to finish)
---

@cadnav Your outputs for those commands would be helpful as well.

mole99, 2024-02-11:
Hi @donn, sure!

FYI, in the meantime I did a fresh install of Debian Testing (Trixie). Same behavior here, the packages have to be built first, which takes a while.

```
$ uname -a
Linux debian-pc 6.6.13-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.6.13-1 (2024-01-20) x86_64 GNU/Linux
```

```
$ nix-env -f "<nixpkgs>" -iA cachix
replacing old 'cachix-1.7'
installing 'cachix-1.7'
```

I had to run cachix as sudo because of: `This user doesn't have permissions to configure binary caches.`.

```
$ sudo env PATH="$PATH" cachix use openlane
Configured https://openlane.cachix.org binary cache in /etc/nix/nix.conf
```

```
$nix develop --accept-flake-config
error: experimental Nix feature 'flakes' is disabled; add '--extra-experimental-features flakes' to enable it
```

And I had to pass some extra arguments:

```
$nix develop --extra-experimental-features nix-command --extra-experimental-features flakes --accept-flake-config
evaluating derivation 'git+file:///home/leo/Repositories/openlane2#devShells.x86_64-linux.default'
[nix-shell:~/Repositories/openlane2]$ 
```

This time `nix-shell` was there almost instant. Could it be because I build the tools already?

donn, 2024-02-11:
Yes… if you already built the tools every other invocation is instant. This is really weird. I'll have to spin up a VM and see what's going on.

donn, 2024-02-11:
Replicated. It appears that `nix-shell` attempts to build everything and `nix develop` just pulls things from the cache.

Fascinating. :upside_down_face: I need to either fix this or update the instructions to have people enable flakes. Urgh.

mole99, 2024-02-12:
I can confirm that `sudo pkill nix-daemon` seems to do the trick 🥳️ thanks!

First, I completely removed nix from my system and reinstalled it according to the documentation. This time I restarted the nix daemon before calling `nix-shell`. After a few minutes it completes without building anything!

donn, 2024-02-12:
Thank you so much for confirming, Leo! Time to add that to the documentation… 😅

closed 2024-02-20 by donn