chipfoundry/openlane2 #387: `Warning: * is used but has no driver.` better be recognized as a Warning in the pre-synthesis checks

mo-hosni, 2024-02-13:
### Description

Currently, those are recognized as errors in both pre-synthesis and post-synthesis checks.
I think it better be a Warning in pre-synthesis and an error in post-synthesis as the signals without drivers are usually tied low in synthesis.
```
Warning: Wire user_project_wrapper.\io_out [4] is used but has no driver.
Warning: Wire user_project_wrapper.\io_out [3] is used but has no driver.
Warning: Wire user_project_wrapper.\io_out [2] is used but has no driver.
```

donn, 2024-04-29:
With regards to this -- the problem is we can't tell what's intentionally left undriven and what's not would it not be more prudent to ask users to explicitly tie them low?

donn, 2024-05-02:
Resolved in #461.

PR #461 (2024-05-01): Ignore undriven checks when running `SYNTH_ELABORATE_ONLY`
## Steps 
* `Yosys.*Synthesis`
  * Syntheses with `SYNTH_ELABORATE_ONLY` no longer report undriven nets as a check error (frequently for some top-level integrations, output pins are left undriven entirely to save space.) 

---
Resolves #387 
Depends on https://github.com/efabless/openlane2-ci-designs/pull/28
commit a3015110 (2024-05-01): Ignore undriven checks when running `SYNTH_ELABORATE_ONLY` (#461)

## Steps 
* `Yosys.*Synthesis`
* Syntheses with `SYNTH_ELABORATE_ONLY` no longer report undriven nets
as a check error (frequently for some top-level integrations, output
pins are left undriven entirely to save space.)

---
Resolves #387 
Depends on https://github.com/efabless/openlane2-ci-designs/pull/28
closed 2024-05-02 by donn