chipfoundry/openlane2 #397: New macro integration method

mole99, 2024-02-25:
### Description

I would like to try another macro integration method that I read about here: https://github.com/The-OpenROAD-Project/OpenROAD/issues/215#issuecomment-617101546

If the macros and the top-level have the same PDN configuration and the macros are properly aligned, the vertical stripes could - in theorie - just be connected.

Therefore I have adjusted OpenLane2's PDN script to pass [`-extend_to_boundary`](https://github.com/The-OpenROAD-Project/OpenROAD/tree/master/src/pdn) to `add_pdn_stripe`.

The PDN stripes of the macros now extend all the way to the boundary and the macros are aligned with the top-level's PDN, but still they don't connect:

[image]

It seems the top-level PDN stripes don't want to touch the stripes inside the macros even tho they are the same net.

### Proposal

Is there a way in OpenLane 2 to make the stripes connect? Or is there a way to extend the macros' PDN stripes even further? (I am not sure if that would help since with the extended stripes, the top-level stripes just stop earlier.)

I can open an issue at OpenROAD's side, but wanted to ask first if anyone here has any ideas :)

If this method works, it could be added to: https://openlane2.readthedocs.io/en/latest/usage/pdn.html

mole99, 2024-02-27:
One solution is to place the macros without a gap. The pins and the PDN stripes are aligned and connected without any wiring. For this case it is mostly good:

[image]

But still it would be nice if the top-level's vertical PDN stripes were also connect :)

donn, 2024-02-29:
@mo-hosni Any ideas?