chipfoundry/openlane2 #401: Add different warning severity levels

donn, 2024-02-29:
### Description

Currently, all warnings are printed without mention of severity. The warnings all just blend in and are ignored as noise. We did try to mitigate that here: https://github.com/The-OpenROAD-Project/OpenLane/pull/2066 But it appears it wasn't enough.

### Proposal

We should:
* Have a list of warnings in the documentation @kareefardi 
* Categorize the warnings by severity (somehow) @mo-hosni

kareefardi, 2024-02-29:
If a warning is severe enough it should be an error. Otherwise categorizing the warnings is too subjective and severity vary depending on the design. Each tool used in the steps of OpenLane can spits numerous warnings and similarly OpenLane itself. In short I don't think it is a good idea to categorize the severity of warnings. Having said that, we can have different categories but not necessarily indicate severity.

donn, 2024-02-29:
Let's discuss on our Monday meeting - I'm not the one who prompted this 🙂