chipfoundry/openlane2 #405: Upstream antenna patch

mole99, 2024-03-04:
### Description

Are there any plans to upstream this patch here: [nix/patches/openroad/antenna.patch](https://github.com/efabless/openlane2/blob/main/nix/patches/openroad/antenna.patch)

Using a locally compiled OpenROAD without it leads to a failure in `Odb.CheckDesignAntennaProperties`:
<details>
  <summary>Error Message</summary>

```
AttributeError: 'dbMTerm' object has no attribute 'hasDiffArea'. Did you mean: 'getDiffArea'?                                                                                        
[09:11:34] ERROR      File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke                                                                           step.py:1170
                        return __callback(*args, **kwargs)                                                                                                                           
                      File "/usr/local/lib/python3.10/dist-packages/openlane/scripts/odbpy/reader.py", line 205, in wrapper                                                          
                        function(**kwargs)                                                                                                                                           
                      File "/usr/local/lib/python3.10/dist-packages/openlane/scripts/odbpy/check_antenna_properties.py", line 109, in check_antenna_properties                       
                        report = check_cells(cells)                                                                                                                                  
                      File "/usr/local/lib/python3.10/dist-packages/openlane/scripts/odbpy/check_antenna_properties.py", line 34, in check_cells                                     
                        has_diff_area = mterm.hasDiffArea()                                                                                                                          
                    AttributeError: 'dbMTerm' object has no attribute 'hasDiffArea'. Did you mean: 'getDiffArea'?
```
</details>

Thanks!

### Proposal

Upstream the patch to OpenROAD

donn, 2024-03-04:
@kareefardi I think we discussed upstreaming this but we stalled because we were busy with release…? What's the status on this?

kareefardi, 2024-03-04:
https://github.com/The-OpenROAD-Project/OpenROAD/pull/4733

kareefardi, 2024-03-27:
This will likely be address by https://github.com/efabless/openlane2/pull/436

donn, 2024-07-03:
Done in dev

PR #472 (2024-06-12): Update OpenROAD
## Steps

* `Odb.ReportDisconnectedPins`
  * Disconnected dummy instances created during CTS, prefixed `clkload`, are now ignored.
 
* `Odb.SetPowerConnections`
  * **Internal**: Restructure `power_utils.py` to provide better error messages and use dictionaries instead of oddball iterator-based filtering

## Tool Updates

* `openroad` -> `b16bda7`
  * Removed OpenLane-specific patch for querying existence of antenna information
  * `openroad-abc` -> `ef5389d`
    * `ABC_USE_NAMESPACE` now set, value also injected into header files
  * `opensta` -> `e01d3f1`

---

Resolves #405  [... cut]