chipfoundry/openlane2 #440: Failing flow on step "Check Antenna Properties of Pins in The Generated Design LEF view"

hpretl, 2024-03-28:
### Description

Running a simple Verilog example using OL2 v2.0.0b17.

### Expected Behavior

No fail.

### Environment report

```sh
foss/designs/example/dig > openlane.env_info || python3 ./openlane/env_info.py
Failed to get Docker info: [Errno 13] Permission denied: 'docker'
Failed to get Docker info: [Errno 2] No such file or directory: 'nix-shell'
kernel: Linux
kernel_version: 6.6.16-linuxkit
supported: True
distro: ubuntu
distro_version: 22.04
python_version: 3.10.12
python_path:
  - /usr/local/bin
  - /usr/lib/python310.zip
  - /usr/lib/python3.10
  - /usr/lib/python3.10/lib-dynload
  - /usr/local/lib/python3.10/dist-packages
  - /usr/lib/python3/dist-packages
  - /usr/lib/python3.10/dist-packages
  - /foss/tools/ngspyce/154a272/local/lib/python3.10/dist-packages
  - /foss/tools/pyopus/0.11/local/lib/python3.10/dist-packages
  - /foss/tools/yosys/share/yosys/python3
  - /foss/tools/klayout/v0.28.17/pymod
tkinter: True
container_info: None
nix_info: None
```

### Reproduction material

```
/foss/designs/example/dig > cat counter.json 
{
    "DESIGN_NAME": "counter",
    "VERILOG_FILES": "dir::counter.v",
    "CLOCK_PORT": "clk_i",
    "CLOCK_PERIOD": 20
}
```

### Relevant log output

```sh
[INFO ODB-0134] Finished DEF file: /tmp/tmpkj__5trb                                                                                                    
Traceback (most recent call last):                                                                                                                     
File "/usr/local/lib/python3.10/dist-packages/openlane/scripts/odbpy/check_antenna_properties.py", line 116, in <module>                               
check_antenna_properties()                                                                                                                             
File "/usr/lib/python3/dist-packages/click/core.py", line 1128, in __call__                                                                            
return self.main(*args, **kwargs)                                                                                                                      
File "/usr/lib/python3/dist-packages/click/core.py", line 1053, in main                                                                                
rv = self.invoke(ctx)                                                                                                                                  
File "/usr/lib/python3/dist-packages/click/core.py", line 1395, in invoke                                                                              
return ctx.invoke(self.callback, **ctx.params)                                                                                                         
File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke                                                                               
return __callback(*args, **kwargs)                                                                                                                     
File "/usr/local/lib/python3.10/dist-packages/openlane/scripts/odbpy/reader.py", line 205, in wrapper                                                  
function(**kwargs)                                                                                                                                     
File "/usr/local/lib/python3.10/dist-packages/openlane/scripts/odbpy/check_antenna_properties.py", line 109, in check_antenna_properties               
report = check_cells(cells)                                                                                                                            
File "/usr/local/lib/python3.10/dist-packages/openlane/scripts/odbpy/check_antenna_properties.py", line 34, in check_cells                             
has_diff_area = mterm.hasDiffArea()                                                                                                                    
AttributeError: 'dbMTerm' object has no attribute 'hasDiffArea'. Did you mean: 'getDiffArea'?                                                          
[14:45:05] ERROR      File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke                                             step.py:1170
                        return __callback(*args, **kwargs)                                                                                             
                      File "/usr/local/lib/python3.10/dist-packages/openlane/scripts/odbpy/reader.py", line 205, in wrapper                            
                        function(**kwargs)                                                                                                             
                      File "/usr/local/lib/python3.10/dist-packages/openlane/scripts/odbpy/check_antenna_properties.py", line 109, in                  
                    check_antenna_properties                                                                                                           
                        report = check_cells(cells)                                                                                                    
                      File "/usr/local/lib/python3.10/dist-packages/openlane/scripts/odbpy/check_antenna_properties.py", line 34, in                   
                    check_cells                                                                                                                        
                        has_diff_area = mterm.hasDiffArea()                                                                                            
                    AttributeError: 'dbMTerm' object has no attribute 'hasDiffArea'. Did you mean: 'getDiffArea'?                                      
                                                                                                                                                       
[14:45:05] ERROR    Log file: 'runs/RUN_2024-03-28_14-44-32/52-odb-checkdesignantennaproperties/odb-checkdesignantennaproperties.log'      step.py:1171
[14:45:05] ERROR    The following error was encountered while running the flow: Check Antenna Properties of Pins in The Generated       __main__.py:153
                    Design LEF view: subprocess (1, ['openroad', '-exit', '-no_splash', '-metrics',                                                    
                    '/foss/designs/example/dig/runs/RUN_2024-03-28_14-44-32/52-odb-checkdesignantennaproperties/or_metrics_out.json',                  
                    '-python', '/usr/local/lib/python3.10/dist-packages/openlane/scripts/odbpy/check_antenna_properties.py',                           
                    '--input-lef', '/foss/pdks/sky130A/libs.ref/sky130_fd_sc_hd/techlef/sky130_fd_sc_hd__nom.tlef', '--input-lef',                     
                    '/foss/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lef/sky130_fd_sc_hd.lef', '--input-lef',                                              
                    '/foss/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lef/sky130_ef_sc_hd.lef', '--design-lef',                                             
                    '/foss/designs/example/dig/runs/RUN_2024-03-28_14-44-32/51-magic-writelef/counter.lef', '--cell-name', 'counter',                  
                    '--report-file',                                                                                                                   
                    '/foss/designs/example/dig/runs/RUN_2024-03-28_14-44-32/52-odb-checkdesignantennaproperties/report.yaml',                          
                    '/foss/designs/example/dig/runs/RUN_2024-03-28_14-44-32/45-openroad-fillinsertion/counter.odb']) failed                            
[14:45:05] ERROR    OpenLane will now quit.                                                                                             __main__.py:154
Classic - Stage 57 - Check Antenna Properties of Pins in The Generated Design LEF view ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╸━━━━━━━━ 56/71 0:00:3
```

donn, 2024-03-28:
As mentioned on Slack, this is pending https://github.com/efabless/openlane2/issues/405, as we currently patch this functionality into OpenROAD.

We apologize for the inconvenience.

donn, 2024-04-01:
@hpretl `--skip Odb.CheckDesignAntennaProperties --skip Odb.CheckMacroAntennaProperties`

You could just sed the steps out in `openlane/flows/classic.py` or pull our OpenROAD patch, both are viable

donn, 2024-07-03:
Fixed in `dev`

PR #472 (2024-06-12): Update OpenROAD
## Steps

* `Odb.ReportDisconnectedPins`
  * Disconnected dummy instances created during CTS, prefixed `clkload`, are now ignored.
 
* `Odb.SetPowerConnections`
  * **Internal**: Restructure `power_utils.py` to provide better error messages and use dictionaries instead of oddball iterator-based filtering

## Tool Updates

* `openroad` -> `b16bda7`
  * Removed OpenLane-specific patch for querying existence of antenna information
  * `openroad-abc` -> `ef5389d`
    * `ABC_USE_NAMESPACE` now set, value also injected into header files
  * `opensta` -> `e01d3f1`

---

Resolves #405  [... cut]
closed 2024-07-03 by donn