chipfoundry/openlane2 #442: Report illegal overlap errors in klayout DRC marker format

urish, 2024-04-01:
### Description

Currently, the illegal overlap report (magic-spiceextraction/feedback.txt) is a list of magic box/feedback commands, e.g.:

```
box 222756 88994 223076 89176
feedback add "Illegal overlap between obsm4 and metal4 (types do not connect)" medium
box 222756 88758 222798 88994
feedback add "Illegal overlap between obsm4 and metal4 (types do not connect)" medium
box 222798 88758 223034 88994
feedback add "Illegal overlap between obsm4 and via4 (types do not connect)" medium
```

It would be useful to convert it into a KLayout marker database, to be able to quickly view the errors in KLayout.

Currently, I'm dividing all the coordinates by 200, then go into KLayout, and use the Goto Position (Ctrl+G) function to manually see where the errors are.

donn, 2024-05-12:
Do you have a full example of this file? I don't have any of hand (a reproducible would also help us add a test)

urish, 2024-05-12:
Here's a workflow run where this happened on Tiny Tapeout 6: https://github.com/TinyTapeout/tinytapeout-06/actions/runs/8504476855

The GDS artifact contains the full logs. Specifically, this is the complete `29-magic-spiceextraction/feedback.txt` from that artifact:

[feedback.txt](https://github.com/efabless/openlane2/files/15285856/feedback.txt)

donn, 2024-07-03:
Fixed in `dev`

PR #478 (2024-05-27): Parse SPICE Extraction feedback, Convert into KLayout Marker Database Format
## Steps

* `Magic.*`
  * All steps now use a new processor, `openlane.steps.magic.MagicOutputProcessor`, to capture and count errors

* `Magic.SpiceExtraction`
  * A `feedback.xml` is now created, with the contents being the SPICE Extraction feedback in the KLayout marker database format

## Misc Enhancements/Bugfixes
* `openlane.steps.DefaultOutputProcessor`
  * `%OL_METRICS_F` now uses Decimals instead of Floats
* `openlane.common`
   * `DRC`: Work around a weird macOS-only bug where boxes in exported KLayout marke [... cut]
closed 2024-07-03 by donn