chipfoundry/openlane2 #454: Volare version setup

EpSilicon, 2024-04-22:
Hello

Is there a way to define the Volare version to be used in openlane2 configuration?
Currently, it seems to fixed to the bdc9412b3e468c102d01b7cf6337be06ec6e9c9a version of the SKY130 PDK.

Thanks and regards,
Erwann

kareefardi, 2024-04-22:
I suppose you mean the version of sky130 PDK used by volare which is different than the version of the tool itslef (volare) that manages the PDK. You can pass `--manual-pdk` to use a custom version of the PDK. Basically, you have to do something similar to this:
```
volare enable <insert_custom_pdk_version>
openlane --manual-pdk <config.json>
```

EpSilicon, 2024-04-22:
You're right, that is exactly what I meant.
Thanks for the answer, it works perfectly.

closed 2024-04-22 by EpSilicon