chipfoundry/openlane2 #467: Synthesis Exploration dead lock

EpSilicon, 2024-05-08:
Hello
The Synthesis Exploration flow in Openlane v2.0.4 seems to be stuck at DELAY 3.

RTL is:
`module comp32(
    a, b, clk, z);

  input [31:0] a, b;  //define inputs and outputs
  input clk;
  output z;

  reg z;

  always @ (posedge clk)
  begin
     if (a > b)
     	z <= 1'b1;
     else 
     	z <= 1'b0;
  end
endmodule`

and config.json is:
`{
  "DESIGN_NAME": "comp32",
  "PDK": "sky130A", 
  "STD_CELL_LIBRARY": "sky130_fd_sc_hd",
  "VERILOG_FILES": "dir::src/*.v",
  "CLOCK_PORT": "clk",
  "CLOCK_PERIOD": 25
}
`
Is there something wrong here?

Thanks and regards,
Erwann

donn, 2024-05-12:
Hi @EpSilicon -- could just be a Yosys bug.

Could I ask you to use the issue template please? It helps us address these issues as we're lacking critical information (including your platform, your method of installation, etc.)

donn, 2024-07-03:
Continued in #501

closed 2024-07-03 by donn