chipfoundry/openlane2 #479: Crash on step 11 in smoke-test

hpretl, 2024-05-22:
### Description

Running `openlane --smoke-test` results in a crash in step 11 (Static Timing Analysis (Pre-PnR))

### Expected Behavior

No crash in smoke test.

### Environment report

```sh
Failed to get Docker info: [Errno 2] No such file or directory: 'docker'
Failed to get Nix info: [Errno 2] No such file or directory: 'nix'
kernel: Linux
kernel_version: 6.6.26-linuxkit
supported: True
distro: ubuntu
distro_version: 22.04
python_version: 3.10.12
python_path:
  - /usr/local/lib/python3.10/dist-packages/openlane
  - /usr/lib/python310.zip
  - /usr/lib/python3.10
  - /usr/lib/python3.10/lib-dynload
  - /usr/local/lib/python3.10/dist-packages
  - /usr/lib/python3/dist-packages
  - /usr/lib/python3.10/dist-packages
  - /foss/tools/ngspyce/154a272/local/lib/python3.10/dist-packages
  - /foss/tools/pyopus/0.11/local/lib/python3.10/dist-packages
  - /foss/tools/yosys/share/yosys/python3
  - /foss/tools/klayout/v0.29.1/pymod
tkinter: True
container_info: None
nix_info: None
```

### Reproduction material

`openlane --smoke-test`

### Relevant log output

```sh
Running 'OpenROAD.STAPrePNR'…                                                                                              step.py:1082
[16:20:44] INFO     Starting STA for the nom_tt_025C_1v80 timing corner…                                                                    openroad.py:560
[16:20:44] INFO     Starting STA for the nom_ss_100C_1v60 timing corner…                                                                    openroad.py:560
[16:20:44] INFO     Starting STA for the nom_ff_n40C_1v95 timing corner…                                                                    openroad.py:560
[16:20:44] INFO     Skipping corner min_tt_025C_1v80 for STA (identical to nom_tt_025C_1v80 at this stage)…                                 openroad.py:594
[16:20:44] VERBOSE  Logging subprocess to                                                                                                      step.py:1262
                    ../../tmp/tmp9rfairu4openlane2/smoke_test_design/runs/RUN_2024-05-22_16-20-42/11-openroad-staprepnr/nom_ss_100C_1v60/sta.l             
                    og…                                                                                                                                    
[16:20:44] INFO     Skipping corner min_ss_100C_1v60 for STA (identical to nom_ss_100C_1v60 at this stage)…                                 openroad.py:594
[16:20:44] INFO     Skipping corner min_ff_n40C_1v95 for STA (identical to nom_ff_n40C_1v95 at this stage)…                                 openroad.py:594
[16:20:44] INFO     Skipping corner max_tt_025C_1v80 for STA (identical to nom_tt_025C_1v80 at this stage)…                                 openroad.py:594
[16:20:44] INFO     Skipping corner max_ss_100C_1v60 for STA (identical to nom_ss_100C_1v60 at this stage)…                                 openroad.py:594
[16:20:44] INFO     Skipping corner max_ff_n40C_1v95 for STA (identical to nom_ff_n40C_1v95 at this stage)…                                 openroad.py:594
[16:20:44] VERBOSE  Logging subprocess to                                                                                                      step.py:1262
                    ../../tmp/tmp9rfairu4openlane2/smoke_test_design/runs/RUN_2024-05-22_16-20-42/11-openroad-staprepnr/nom_tt_025C_1v80/sta.l             
                    og…                                                                                                                                    
[16:20:44] VERBOSE  Logging subprocess to                                                                                                      step.py:1262
                    ../../tmp/tmp9rfairu4openlane2/smoke_test_design/runs/RUN_2024-05-22_16-20-42/11-openroad-staprepnr/nom_ff_n40C_1v95/sta.l             
                    og…                                                                                                                                    
[16:20:44] ERROR    Subprocess had a non-zero exit.                                                                                            step.py:1308
[16:20:44] ERROR    Last 10 line(s):                                                                                                           step.py:1313
                    %OL_METRIC_F power__leakage__total 2.268423941131914e-6                                                                                
                    %OL_METRIC_F power__total 0.0006134541472420096                                                                                        
                                                                                                                                                           
                    %OL_END_REPORT                                                                                                                         
                    %OL_CREATE_REPORT skew.min.rpt                                                                                                         
                                                                                                                                                           
[... 6 lines cut ...]
                    '../../tmp/tmp9rfairu4openlane2/smoke_test_design/runs/RUN_2024-05-22_16-20-42/11-openroad-staprepnr/nom_ss_100C_1v60/sta.             
                    log'                                                                                                                                   
[16:20:44] ERROR    Failed STA for the nom_ss_100C_1v60 timing corner:                                                                      openroad.py:577
[16:20:44] ERROR    Subprocess had a non-zero exit.                                                                                            step.py:1308
[16:20:44] ERROR    Last 10 line(s):                                                                                                           step.py:1313
                    %OL_METRIC_F power__leakage__total 1.5029198996430182e-9                                                                               
                    %OL_METRIC_F power__total 0.0007705658208578825                                                                                        
                                                                                                                                                           
                    %OL_END_REPORT                                                                                                                         
                    %OL_CREATE_REPORT skew.min.rpt                                                                                                         
                                                                                                                                                           
                    ===========================================================================                                                            
                    Clock Skew (Hold)                                                                                                                      
                    ============================================================================                                                           
                    Error: corner.tcl, 145 Wrong number of arguments :sta::worst_clk_skew_cmd setup_hold include_internal_latency  argument 2              
                                                                                                                                                           
[16:20:44] ERROR    Full log file:                                                                                                             step.py:1316
                    '../../tmp/tmp9rfairu4openlane2/smoke_test_design/runs/RUN_2024-05-22_16-20-42/11-openroad-staprepnr/nom_tt_025C_1v80/sta.             
                    log'                                                                                                                                   
[16:20:44] ERROR    Failed STA for the nom_tt_025C_1v80 timing corner:                                                                      openroad.py:577
[16:20:44] ERROR    The following error was encountered while running the flow:                                                             __main__.py:152
                    STA (Pre-PnR): subprocess (1, ['sta', '-no_splash', '-exit',                                                                           
                    '/usr/local/lib/python3.10/dist-packages/openlane/scripts/openroad/sta/corner.tcl']) failed                                            
[16:20:44] ERROR    OpenLane will now quit.
```

hpretl, 2024-05-22:
@donn Any idea?

donn, 2024-05-22:
What likely happened is that OpenSTA changed an internal API that we happened to be relying on to generate metrics (which for the record they're within their right to do)

When we update OpenSTA we'll likely fix the function call. For now, we'd advise downgrading OpenSTA if you're manually rolling your own tools.

hpretl, 2024-05-22:
We are on OpenROAD latest, where is the list of proposed versions so that we can follow your tags?

hpretl, 2024-05-22:
Ah I see, in the `.nix` files there are GH hashes. Is that the place to look, or is there a central location as well?

donn, 2024-05-22:
That is the place to look, yep

hpretl, 2024-05-23:
Fixed by dialing back OpenROAD version.

closed 2024-05-23 by hpretl