chipfoundry/openlane2 #480: Point to the report file of Checker.YosysSynthChecks

kareefardi, 2024-05-23:
### Description

`Checker.YosysSynthChecks` reports errors on `synthesis__check_error__count`. However, there is nothing pointing towards the errors themselves and where are they located. 

### Proposal

Point to the report file relating to the error.

donn, 2024-05-23:
I don't think this will work given the cardinal rule of OpenLane 2 which is "steps should not be looking in other step's folders on their own"

kareefardi, 2024-05-25:
It isn't a requirement that it point to the exact file. The current solution gives no indication to even where to look for checks errors except that both are called yosys. Some renames to the report file path would be a better start rather than what we have.

KelvinChung2000, 2024-06-04:
Such a change will be helpful. Also, the error message should also indicate a warning (e.g., a wire is used but has no driver), which is also classified as an error.

donn, 2024-10-07:
We should really find a way to do this that doesn't break OpenLane 2's architecture. Maybe there should be a "metrics references" that shows the source of the metric(s)…?

KelvinChung2000, 2024-10-07:
This should do, as long as it can point to where to find the information. It would be nicer if it could show the information directly, but it does not matter.

Nado15, 2024-10-10:
Hi everyone,

I'm currently working with OpenLane2 and have reached a point where my synthesis fails at the Checker.YosysSynthChecks step. I struggled to identify the cause, as the latest generated folder (in my case, 08-checker...) didn't provide any useful information.

I implemented a shared bus using tri-state buffers, and I suspect this might be the source of the issue. In another log file, I noticed that Yosys reported 8 problems, which aligns with my 8-bit bus design.

I wanted to share this experience and suggest that it would be beneficial to streamline the error reporting process to help us identify issues more quickly.

Besides that, thanks for your work!

IAmMarcelJung, 2025-02-24:
Just want to add that for me the checker failed because I had some undriven outputs in my design. Maybe this helps someone in the future.