chipfoundry/openlane2 #485: clarify the meaning of a views and DesignFormat in the architectural overview

maliberty, 2024-05-30:
### Description

It is confusing that in https://openlane2.readthedocs.io/en/latest/reference/architecture.html
```
States
A openlane.steps.State is a snapshot of paths to views at any point in time, in dictionary form.
Keys must be of the type openlane.steps.DesignFormat
```
mentions views without ever defining what a view is.  Likewise DesignFormat.

### Proposal

Define views and DesignFormat before using those terms.  Also those terms are not links in the docs while other terms are.

donn, 2024-05-30:
I *think* they used to be terms but broke when I created a dedicated submodule for state-- will have to track down why it didn't warn about that

donn, 2024-07-03:
Fixed in `dev`: https://openlane2.readthedocs.io/en/dev/reference/architecture.html

maliberty, 2024-07-03:
This is a circular definition:
```
class openlane.state.DesignFormat(value)
A list of design formats that may be kept in the state object.
```
I still don't know what it is.  The parameters have no doc string so they aren't clear either.

maliberty, 2024-07-03:
Its also confusing that it is described as a list but seems to be a singular object.

closed 2024-07-23 by donn