chipfoundry/openlane2 #493: Install nix on ubuntu 22.04 - running smoke test - missing libgit2.so.1.1

jorgenkraghjakobsen, 2024-06-22:
### Description

Did fresh install of nix on a linux ubuntu 22.04 - cloned the openlane2 repo - opened a nix-shell in that repo and ran openlane --smoke_test 

File log: - Looks like libgit2.so.1.1 is missing

File "/nix/store/jr2jxhcp4qybkyz8was3ym6m35cd869d-python3-3.11.6-env/lib/python3.11/site-packages/klayout/db/__init__.py", line 3, in <module>                                                                                 
import klayout.dbcore                                                                                                                                                                                                          
ImportError: libgit2.so.1.1: cannot open shared object file: No such file or directory                                                                                                                                         
[11:05:45] ERROR    Subprocess had a non-zero exit.                                                                                                                                                                step.py:1308
[11:05:45] ERROR    Last 9 line(s):                                                                                                                                                                                step.py:1313
                    Traceback (most recent call last):                                                                                                                                                                         
                      File "/nix/store/jr2jxhcp4qybkyz8was3ym6m35cd869d-python3-3.11.6-env/lib/python3.11/site-packages/openlane/scripts/klayout/stream_out.py", line 50, in <module>                                          
                        import pya                                                                                                                                                                                             
                      File "/nix/store/jr2jxhcp4qybkyz8was3ym6m35cd869d-python3-3.11.6-env/lib/python3.11/site-packages/pya/__init__.py", line 5, in <module>                                                                  
                        from klayout.db import *  # noqa                                                                                                                                                                       
                        ^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                                                                               
                      File "/nix/store/jr2jxhcp4qybkyz8was3ym6m35cd869d-python3-3.11.6-env/lib/python3.11/site-packages/klayout/db/__init__.py", line 3, in <module>                                                           
                        import klayout.dbcore                                                                                                                                                                                  
                    ImportError: libgit2.so.1.1: cannot open shared object file: No such file or directory                                                                                                                     
                                                                                                                                                                                                                               
[11:05:45] ERROR    Full log file: '../../../../tmp/tmppxwkart1openlane2/smoke_test_design/runs/RUN_2024-06-22_11-05-04/54-klayout-streamout/klayout-streamout.log'                                                step.py:1316
[11:05:45] WARNING  The following warnings were generated by the flow:                                                                                                                                              flow.py:671
[11:05:45] WARNING  [Odb.CustomIOPlacement] Overriding minimum distance 0.1 with 0.42 for pins on side N to avoid overlap.                                                                                          flow.py:673
[11:05:45] WARNING  [OpenROAD.DetailedRouting] [DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer mcon (and 9 similar warnings)                                                       flow.py:673
[11:05:45] WARNING  [Checker.WireLength] Threshold for Threshold-surpassing long wires is not set. The checker will be skipped.                                                                                     flow.py:673
[11:05:45] WARNING  [OpenROAD.IRDropReport] 'VSRC_LOC_FILES' was not given a value, which may make the results of IR drop analysis inaccurate. If you are not integrating a top-level chip for manufacture, you may flow.py:673
                    ignore this warning, otherwise, see the documentation for 'VSRC_LOC_FILES'.                                                                                                                                
[11:05:45] ERROR    The following error was encountered while running the flow:                                                                                                                                 __main__.py:152
                    GDSII Stream Out (KLayout): subprocess (1, ['/nix/store/jr2jxhcp4qybkyz8was3ym6m35cd869d-python3-3.11.6-env/bin/python3.11',                                                                               
                    '/nix/store/jr2jxhcp4qybkyz8was3ym6m35cd869d-python3-3.11.6-env/lib/python3.11/site-packages/openlane/scripts/klayout/stream_out.py',                                                                      
                    '/tmp/tmppxwkart1openlane2/smoke_test_design/runs/RUN_2024-06-22_11-05-04/49-openroad-fillinsertion/spm.def', '--output',                                                                                  
                    '/tmp/tmppxwkart1openlane2/smoke_test_design/runs/RUN_2024-06-22_11-05-04/54-klayout-streamout/spm.klayout.gds', '--top', 'spm', '--lyp',                                                                  
                    '/home/jakobsen/.volare/volare/sky130/versions/bdc9412b3e468c102d01b7cf6337be06ec6e9c9a/sky130A/libs.tech/klayout/tech/sky130A.lyp', '--lyt',                                                              
                    '/home/jakobsen/.volare/volare/sky130/versions/bdc9412b3e468c102d01b7cf6337be06ec6e9c9a/sky130A/libs.tech/klayout/tech/sky130A.lyt', '--lym',                                                              
                    '/home/jakobsen/.volare/volare/sky130/versions/bdc9412b3e468c102d01b7cf6337be06ec6e9c9a/sky130A/libs.tech/klayout/tech/sky130A.map', '--input-lef',                                                        
                    '/home/jakobsen/.volare/volare/sky130/versions/bdc9412b3e468c102d01b7cf6337be06ec6e9c9a/sky130A/libs.ref/sky130_fd_sc_hd/techlef/sky130_fd_sc_hd__nom.tlef', '--input-lef',                                
                    '/home/jakobsen/.volare/volare/sky130/versions/bdc9412b3e468c102d01b7cf6337be06ec6e9c9a/sky130A/libs.ref/sky130_fd_sc_hd/lef/sky130_fd_sc_hd.lef', '--input-lef',                                          
                    '/home/jakobsen/.volare/volare/sky130/versions/bdc9412b3e468c102d01b7cf6337be06ec6e9c9a/sky130A/libs.ref/sky130_fd_sc_hd/lef/sky130_ef_sc_hd.lef', '--with-gds-file',                                      
                    '/home/jakobsen/.volare/volare/sky130/versions/bdc9412b3e468c102d01b7cf6337be06ec6e9c9a/sky130A/libs.ref/sky130_fd_sc_hd/gds/sky130_fd_sc_hd.gds']) failed                                                 
[11:05:45] ERROR    OpenLane will now quit.                                                                                                                                                                     __main__.py:153
Classic - Stage 57 - GDSII Stream Out (KLayout) ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╸━━━━━━━━━━ 56/75 0:00:41
[nix-shell:~/opentools/openlane2]$ klayout 
klayout: error while loading shared libraries: libgit2.so.1.1: cannot open shared object file: No such file or directory

[nix-shell:~/opentools/openlane2]$ 

### Expected Behavior

Build some thing :-) 

### Environment report

```sh
kernel: Linux
kernel_version: 6.8.1-060801-generic
supported: True
[... cut ...]
  - /nix/store/8wkz5a9z1ppwjvr54375k69avhkmz4nc-python3.11-psutil-5.9.6/lib/python3.11/site-packages
  - /nix/store/kn3xxzgq9bx9av42h495bvxnslyg41v1-klayout-pymod/lib/python3.11/site-packages
tkinter: True
container_info: None
nix_info:
  version_string: nix (Nix) 2.23.0
  channels:
  nix_command: True
  flakes: True
```

### Reproduction material

did not work

### Relevant log output

```sh
[nix-shell:~/opentools/openlane2]$ klayout -version
klayout: error while loading shared libraries: libgit2.so.1.1: cannot open shared object file: No such file or directory
```

jorgenkraghjakobsen, 2024-06-22:
Found the error :-)  - And all good now 
 
* Antenna
Passed ✅

* LVS
Passed ✅

* DRC
Passed ✅

[11:30:31] INFO     Saving views to '/tmp/tmp4ehmkxuuopenlane2/smoke_test_design/runs/RUN_2024-06-22_11-29-39/final'…                                                                                              state.py:193
[11:30:31] INFO     Flow complete.                                                                                                                                                                            sequential.py:294
Classic - Stage 75 - Report Manufacturability ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 75/75 0:00:51
[11:30:31] WARNING  The following warnings were generated by the flow:                                                                                                                                              flow.py:671
[11:30:31] WARNING  [Odb.CustomIOPlacement] Overriding minimum distance 0.1 with 0.42 for pins on side N to avoid overlap.                                                                                          flow.py:673
[11:30:31] WARNING  [OpenROAD.DetailedRouting] [DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer mcon (and 9 similar warnings)                                                       flow.py:673
[11:30:31] WARNING  [Checker.WireLength] Threshold for Threshold-surpassing long wires is not set. The checker will be skipped.                                                                                     flow.py:673
[11:30:31] WARNING  [OpenROAD.IRDropReport] 'VSRC_LOC_FILES' was not given a value, which may make the results of IR drop analysis inaccurate. If you are not integrating a top-level chip for manufacture, you may flow.py:673
                    ignore this warning, otherwise, see the documentation for 'VSRC_LOC_FILES'.                                                                                                                                
[11:30:31] INFO     Smoke test passed.                                                                                                                                                                          __main__.py:267

[nix-shell:~/opentools/openlane2]$

jorgenkraghjakobsen, 2024-06-22:
On my machine it use a local klayout build and have set LD_LIBRARY_PATH 
``` jakobsen@t14:~$ type klayout
klayout is /home/jakobsen/opentools/klayout/bin-release/klayout
jakobsen@t14:~$ echo $LD_LIBRARY_PATH 
:/home/jakobsen/opentools/klayout/bin-release
jakobsen@t14:~$ klayout -v
KLayout 0.29.0
jakobsen@t14:~$ 
```
In my nix environment - the LIB_LIBRARY_PATH is still there an klayout misses out on libgit2.so.... 
```
akobsen@t14:~$ cd opentools/openlane2/
jakobsen@t14:~/opentools/openlane2$ ls
Authors.md    flake.lock  nix                Readme.md              setup.py
Changelog.md  flake.nix   notebook.ipynb     requirements_dev.txt   shell.nix
default.nix   License     openlane           requirements_docs.txt  test
docs          Makefile    openlane2_env.log  requirements.txt       type_stubs
jakobsen@t14:~/opentools/openlane2$ nix-shell 

[nix-shell:~/opentools/openlane2]$ type klayout 
klayout is /nix/store/gvq51plg4ydzfxma0m46w82jbvngvnch-klayout/bin/klayout

[nix-shell:~/opentools/openlane2]$ echo $LD_LIBRARY_PATH 
:/home/jakobsen/opentools/klayout/bin-release:/home/jakobsen/opentools/klayout/bin-release

[nix-shell:~/opentools/openlane2]$ klayout -v
klayout: error while loading shared libraries: libgit2.so.1.1: cannot open shared object file: No such file or directory

[nix-shell:~/opentools/openlane2]$ 
```
/J

donn, 2024-06-22:
I don't know how you're setting `LD_LIBRARY_PATH`, but you could set it as follows:

```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/jakobsen/opentools/klayout/bin-release/klayout
```

This will make sure the right libraries can be loaded first.

Thanks for giving OpenLane 2 a try, Jørgen!

jorgenkraghjakobsen, 2024-06-22:
Hi Donn 
That is how it all ready get set from my .bashrc file: 
```
export PATH=$PATH:/home/jakobsen/opentools/klayout/bin-release:/home/jakobsen/.cargo/bin 
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/jakobsen/opentools/klayout/bin-release
```
In nix-shell I just get the my original LD_LIBRARY_PATH added again.  
```
jakobsen@t14:~/opentools/openlane2$ nix-shell 

[nix-shell:~/opentools/openlane2]$ echo $LD_LIBRARY_PATH 
:/home/jakobsen/opentools/klayout/bin-release/klayout:/home/jakobsen/opentools/klayout/bin-release/klayout
```
I will try to see if I can find out why. 
Yes - Openlane2 looks very nice. I will try so run some of my design this way. 
/J

donn, 2024-10-07:
Sadly this appears to be inherent to Nix. You can add something like this to your rcfile:

```sh
if [ "$IN_NIX_SHELL" = "" ]; then
	export LD_LIBRARY_PATH=…
fi
```

closed 2024-10-07 by donn