chipfoundry/openlane2 #494: Please update `rich` dependency

hpretl, 2024-06-24:
### Description

Hi @donn could you please update the `rich` dependency to be compatible with `cace` from @mole99:

[image]

### Proposal

Upgrading to rich < 14 would be ideal.

donn, 2024-06-24:
Looking into it

hpretl, 2024-06-25:
BTW, when you are looking at this: Bumping KLayout to <=0.92.2 (or at least 0.91.1) would also be quite helpful! :-)

kareefardi, 2024-06-25:
Rich issue addressed by https://github.com/efabless/openlane2/pull/495

hpretl, 2024-06-26:
Thank you, the issue regarding `rich` looks solved, `openlane` and `cace` can now coexist, thanks!

Unfortunately, there is still an incompatibility with `kfactory` (an important piece of `gdsfactory`) on `klayout`, please see here:

```
33.63 ERROR: Cannot install kfactory[git,ipy]==0.17.4 and openlane==2.0.10 because these package versions have conflicting dependencies.
33.63
33.63 The conflict is caused by:
33.63     openlane 2.0.10 depends on klayout<0.29.0 and >=0.28.17.post1
33.63     kfactory[git,ipy] 0.17.4 depends on klayout>=0.29.1
```

hpretl, 2024-06-26:
@donn @kareefardi Would an update of the dependency to `klayout<=0.29.1` work?

hpretl, 2024-06-26:
I tested klayout 0.29.2, and it runs fine on the smoke test.

hpretl, 2024-06-26:
See https://github.com/efabless/openlane2/pull/496

donn, 2024-07-03:
Done in dev

closed 2024-07-03 by donn