chipfoundry/openlane2 #497: Two bullet points not rendered as bullet points in documentation

jeras, 2024-06-28:
Sorry for the trivial issue.

In the source code:
https://github.com/efabless/openlane2/blob/bc143861bdd1fe22169a73054bcb3207fc3ce3f0/openlane/flows/flow.py#L513

In documentation:
https://openlane2.readthedocs.io/en/latest/reference/api/flows/index.html#openlane.flows.Flow.start

jeras, 2024-06-28:
Another trivial one, a broken wikipedia link in the same source file:
https://github.com/efabless/openlane2/blob/bc143861bdd1fe22169a73054bcb3207fc3ce3f0/openlane/flows/flow.py#L939

In documentation the link is missing the closing parenthesis is missing from the link:
https://openlane2.readthedocs.io/en/latest/reference/api/flows/index.html#openlane.flows.Flow.FlowFactory

jeras, 2024-06-28:
Another bullet point issue:
https://github.com/efabless/openlane2/blob/bc143861bdd1fe22169a73054bcb3207fc3ce3f0/openlane/flows/cli.py#L198
https://openlane2.readthedocs.io/en/latest/reference/api/flows/index.html#openlane.flows.cloup_flow_opts

donn, 2024-06-28:
Never apologize for being helpful. Thank you. We'll be looking into these.