chipfoundry/openlane2 #501: spm synthesis exploration got stuck

vijayank88, 2024-07-02:
### Description

Synthesis exploration not working for spm design

### Expected Behavior

Complete spm synthesis exploration

### Environment report

```sh
bash: openlane.env_info: command not found
warning: unknown experimental feature '='
warning: unknown experimental feature 'flakes\nextra-substituters'
warning: unknown experimental feature 'https://openlane.cachix.org\nextra-trusted-public-keys'
warning: unknown experimental feature 'openlane.cachix.org-1:qqdwh+QMNGmZAuyeQJTH9ErW57OWSvdtuwfBKdS254E=\n'
warning: unknown experimental feature '='
warning: unknown experimental feature 'flakes\nextra-substituters'
warning: unknown experimental feature 'https://openlane.cachix.org\nextra-trusted-public-keys'
warning: unknown experimental feature 'openlane.cachix.org-1:qqdwh+QMNGmZAuyeQJTH9ErW57OWSvdtuwfBKdS254E=\n'
kernel: Linux
kernel_version: 5.15.0-113-generic
supported: True
distro: ubuntu
distro_version: 20.04
python_version: 3.8.10
python_path:
  - /home/vijayan/openlane2/openlane
  - /usr/lib/python38.zip
  - /usr/lib/python3.8
  - /usr/lib/python3.8/lib-dynload
  - /home/vijayan/.local/lib/python3.8/site-packages
  - /usr/local/lib/python3.8/dist-packages
  - /usr/lib/python3/dist-packages
tkinter: True
container_info:
  engine: docker
  version: 26.1.4
  conmon: False
  rootless: False
nix_info:
  version_string: nix (Nix) 2.20.4
  channels:
  nix_command: True
  flakes: False
```

### Reproduction material

Run following command for spm design
```
openlane ./config.json --flow SynthesisExpolaration
```

### Relevant log output

```sh
[18:06:56] INFO     Loading the design configuration file has generated the following warnings:                                                                                                       config.py:670
[18:06:56] WARNING  Key 'FP_CORE_UTIL' provided is unused by the current flow.                                                                                                                        config.py:674
[18:06:56] WARNING  Key 'FP_PDN_HOFFSET' provided is unused by the current flow.                                                                                                                      config.py:674
[18:06:56] WARNING  Key 'FP_PDN_HPITCH' provided is unused by the current flow.                                                                                                                       config.py:674
[18:06:56] WARNING  Key 'FP_PDN_HWIDTH' provided is unused by the current flow.                                                                                                                       config.py:674
[18:06:56] WARNING  Key 'FP_PDN_SKIPTRIM' provided is unused by the current flow.                                                                                                                     config.py:674
[18:06:56] WARNING  Key 'FP_PDN_VOFFSET' provided is unused by the current flow.                                                                                                                      config.py:674
[18:06:56] WARNING  Key 'FP_PDN_VPITCH' provided is unused by the current flow.                                                                                                                       config.py:674
[18:06:56] WARNING  Key 'FP_PDN_VWIDTH' provided is unused by the current flow.                                                                                                                       config.py:674
[18:06:56] WARNING  Key 'FP_PIN_ORDER_CFG' provided is unused by the current flow.                                                                                                                    config.py:674
[18:06:56] INFO     Starting a new run of the 'SynthesisExploration' flow with the tag 'RUN_2024-07-02_18-06-56'.                                                                                       flow.py:627
[18:06:57] [V] Running 'synthesis-AREA 0'…                                                                                                                                                             step.py:1082
[18:06:57] [V] Logging subprocess to runs/RUN_2024-07-02_18-06-56/1-synthesis-area-0/synthesis-area-0.log…                                                                                             step.py:1262
[18:07:03] [V] Parsing synthesis checks…                                                                                                                                                                yosys.py:55
[18:07:03] [V] Running 'sdc-AREA 0'…                                                                                                                                                                   step.py:1082
[18:07:03] [V] Running 'synthesis-AREA 1'…                                                                                                                                                             step.py:1082
[18:07:03] [V] Running 'sta-AREA 0'…                                                                                                                                                                   step.py:1082
[18:07:03] [I] Skipping corner min_tt_025C_1v80 for STA (identical to nom_tt_025C_1v80 at this stage)…                                                                                              openroad.py:594
[18:07:03] [I] Skipping corner min_ss_100C_1v60 for STA (identical to nom_ss_100C_1v60 at this stage)…                                                                                              openroad.py:594
[18:07:03] [I] Skipping corner min_ff_n40C_1v95 for STA (identical to nom_ff_n40C_1v95 at this stage)…                                                                                              openroad.py:594
[18:07:03] [I] Skipping corner max_tt_025C_1v80 for STA (identical to nom_tt_025C_1v80 at this stage)…                                                                                              openroad.py:594
[18:07:03] [I] Skipping corner max_ss_100C_1v60 for STA (identical to nom_ss_100C_1v60 at this stage)…                                                                                              openroad.py:594
[18:07:03] [I] Skipping corner max_ff_n40C_1v95 for STA (identical to nom_ff_n40C_1v95 at this stage)…                                                                                              openroad.py:594
[18:07:03] [V] Logging subprocess to runs/RUN_2024-07-02_18-06-56/1-synthesis-area-1/synthesis-area-1.log…                                                                                             step.py:1262
[18:07:05] [V] Parsing synthesis checks…                                                                                                                                                                yosys.py:55
[18:07:05] [V] Running 'sdc-AREA 1'…                                                                                                                                                                   step.py:1082
[18:07:05] [V] Running 'sta-AREA 1'…                                                                                                                                                                   step.py:1082
[18:07:05] [I] Skipping corner min_tt_025C_1v80 for STA (identical to nom_tt_025C_1v80 at this stage)…                                                                                              openroad.py:594
[18:07:05] [I] Skipping corner min_ss_100C_1v60 for STA (identical to nom_ss_100C_1v60 at this stage)…                                                                                              openroad.py:594
[18:07:05] [I] Skipping corner min_ff_n40C_1v95 for STA (identical to nom_ff_n40C_1v95 at this stage)…                                                                                              openroad.py:594
[18:07:05] [I] Skipping corner max_tt_025C_1v80 for STA (identical to nom_tt_025C_1v80 at this stage)…                                                                                              openroad.py:594
[18:07:05] [I] Skipping corner max_ss_100C_1v60 for STA (identical to nom_ss_100C_1v60 at this stage)…                                                                                              openroad.py:594
[18:07:05] [I] Skipping corner max_ff_n40C_1v95 for STA (identical to nom_ff_n40C_1v95 at this stage)…                                                                                              openroad.py:594
SynthesisExploration - Stage 1 - Synthesis Exploration ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 0/1 0:11:52
```

jworozco, 2024-07-02:
I am facing the same issue, i have tried with 2 different openlane2 versions and 2 different simple designs (simple fifo and simple adder), but the flow gets stuck at the same spot. Adding comments as I'd like to get updates on this, if a solution is found.

donn, 2024-07-03:
So, problem replicated, and it's a threading deadlock because I misunderstood how threadpools work in Python. 

Unfortunately, I'm still trying to wrap my head around a proper fix without rewriting this whole damn thing but a workaround is to pass the flag `-j10`.

I'm extremely sorry for the inconvenience.

PR #502 (2024-07-03): Fix Deadlock
## Misc Enhancements/Bugfixes

* Fixed a deadlock in some situations because of `OpenROAD.STAPrePNR` using the
  global thread-pool for OpenLane, which may be used to run the step itself.

---

Resolves #501
commit 7691a45f (2024-07-03): Fix Deadlock (#502)

## Misc Enhancements/Bugfixes

* Fixed a deadlock in some situations because of `OpenROAD.STAPrePNR`
using the
global thread-pool for OpenLane, which may be used to run the step
itself.

---

Resolves #501
closed 2024-07-03 by donn