chipfoundry/openlane2 #508: Padframe generation in OpenLane2

EpSilicon, 2024-07-11:
### Description

Hello,
What is the best way to generate a padframe in OpenLane2 ?
It's seems no more possible to use an interactive TCL script as in OpenLane:
https://openlane.readthedocs.io/en/2023.09.07/usage/chip_integration.html#hardening-the-full-chip
Thanks and regards,
Erwann

donn, 2024-07-23:
Padframe generation is planned, but as a low priority (as we as a company tend to reuse manually-made padframes :) )

You may want to try a custom step based on OpenROAD for padframe generation-- if you are willing to be more specific with your use-case we can help you out with that.

EpSilicon, 2024-07-24:
I've tried a custom step using YosisHQ's Padring script. Is this the way you have in mind?

donn, 2024-07-24:
That's one way, yeah. Another would be OpenROAD `pad`-based https://github.com/The-OpenROAD-Project/OpenROAD/tree/master/src/pad

As a side note, we'd be happy to add a padring step to the flow if you've already made one.