chipfoundry/openlane2 #509: PDN unconnected to the macros

FriedrichWu, 2024-07-11:
### Description

Hello, 

I'm trying to use the "ring method", which is described here[https://openlane2.readthedocs.io/en/latest/usage/pdn.html](url), to connect the power ring of several macros, but failed.
[image]
It seems that horizontal power line is also missing 
[image]

### Expected Behavior

The power grid is generated at met3(horizontal), met4(vertical), and be able to connect to the power ring around the macro (met3 horizontal, met4 vertical), and does not across the macro

### Environment report

```sh
NA
```

### Reproduction material

I'm current running 2.0.0rc2, and the I could provide my config.json and pdn.tcl
[issue_files.zip](https://github.com/user-attachments/files/16178794/issue_files.zip)

### Relevant log output

```sh
'PNR_SDC_FILE' is not defined. Using generic fallback SDC for OpenROAD PnR steps.
'SIGNOFF_SDC_FILE' is not defined. Using generic fallback SDC for OpenROAD PnR steps.
No libs found for macro sky130_sram_1rw_tiny_bank_0 at corner nom_tt_025C_1v80. The module will be black-boxed.
No libs found for macro sky130_sram_1rw_tiny_control_0 at corner nom_tt_025C_1v80. The module will be black-boxed.
No libs found for macro sky130_sram_1rw_tiny_data_dff_0 at corner nom_tt_025C_1v80. The module will be black-boxed.
No libs found for macro sky130_sram_1rw_tiny_spare_wen_dff_0 at corner nom_tt_025C_1v80. The module will be black-boxed.
No libs found for macro sky130_sram_1rw_tiny_row_addr_dff_0 at corner nom_tt_025C_1v80. The module will be black-boxed.
No libs found for macro sky130_sram_1rw_tiny_wmask_dff_0 at corner nom_tt_025C_1v80. The module will be black-boxed.
No libs found for macro sky130_sram_1rw_tiny_bank_0 at corner nom_tt_025C_1v80. The module will be black-boxed.
No libs found for macro sky130_sram_1rw_tiny_control_0 at corner nom_tt_025C_1v80. The module will be black-boxed.
No libs found for macro sky130_sram_1rw_tiny_data_dff_0 at corner nom_tt_025C_1v80. The module will be black-boxed.
No libs found for macro sky130_sram_1rw_tiny_spare_wen_dff_0 at corner nom_tt_025C_1v80. The module will be black-boxed.
No libs found for macro sky130_sram_1rw_tiny_row_addr_dff_0 at corner nom_tt_025C_1v80. The module will be black-boxed.
No libs found for macro sky130_sram_1rw_tiny_wmask_dff_0 at corner nom_tt_025C_1v80. The module will be black-boxed.
[ORD-2011] LEF master sky130_sram_1rw_tiny_bank_0 has no liberty cell.
[ORD-2011] LEF master sky130_sram_1rw_tiny_control_0 has no liberty cell.
[ORD-2011] LEF master sky130_sram_1rw_tiny_data_dff_0 has no liberty cell.
[ORD-2011] LEF master sky130_sram_1rw_tiny_row_addr_dff_0 has no liberty cell.
[ORD-2011] LEF master sky130_sram_1rw_tiny_spare_wen_dff_0 has no liberty cell.
[ORD-2011] LEF master sky130_sram_1rw_tiny_wmask_dff_0 has no liberty cell.
Cell 'sky130_sram_1rw_tiny_bank_0' has (23) input pin(s) without antenna gate information. They might not be connected to a gate.
Cell 'sky130_sram_1rw_tiny_bank_0' has (10) output pin(s) without antenna diffusion information. They might not be driven.
[PSM-0094] bank_ins is not connected to vccd1.
[PSM-0094] control_0_ins is not connected to vccd1.
[PSM-0094] data_dff_0_ins is not connected to vccd1.
[PSM-0094] row_addr_dff_0_ins is not connected to vccd1.
[PSM-0094] spare_wen_dff_0_ins is not connected to vccd1.
[PSM-0094] wmask_dff_0_ins is not connected to vccd1.
Grid check for vccd1 failed: PSM-0069
[PSM-0094] bank_ins is not connected to vssd1.
[PSM-0094] control_0_ins is not connected to vssd1.
[PSM-0094] data_dff_0_ins is not connected to vssd1.
[PSM-0094] row_addr_dff_0_ins is not connected to vssd1.
[PSM-0094] spare_wen_dff_0_ins is not connected to vssd1.
[PSM-0094] wmask_dff_0_ins is not connected to vssd1.
Grid check for vssd1 failed: PSM-0069
All instances are FIXED/FIRM.
No need to perform global placement.
Skipping…
All instances are FIXED/FIRM.
No need to perform global placement.
Skipping…
870 power grid violations (as reported by OpenROAD PSM- you may ignore these if LVS passes) found.
```

donn, 2024-07-11:
looking into it…

FriedrichWu, 2024-07-17:
Actually it is OK to do connection over the macro, but seems not possible to do "ring method" like in the documentation, or maybe I lost some special commands ( but in doc says "The core ring method does not actually require any special configuration for the top-level integration" )?
[image]