chipfoundry/openlane2 #510: Allow Docker drop-in alternatives, such as podman

maxrdz, 2024-07-12:
### Description

I have a Docker drop-in alternative installed on my system, podman, which is a very popular choice. It seems that when the `--dockerized` argument is passed, it searches explicitly for a `docker` binary. Could OpenLANE search for alternatives as well? Or maybe configurable?

Aliasing `docker` as `podman` doesn't seem to work, since I assume it looks for the binary in the filesystem, and then executes it, instead of going through the shell. So, currently the only work around on a Linux environment is to create a symlink to the `podman` binary with the name `docker` in the system bin.

donn, 2024-07-12:
`export OPENLANE_CONTAINER_ENGINE=podman` would do it. I use nerdctl personally, albeit with a symlink instead of an alias. 🙂

donn, 2024-07-14:
Did this work for you, Max?

maxrdz, 2024-07-15:
> Did this work for you, Max?

Yes, it did. Thank you for the prompt response, and sorry for my late response. 😛

closed 2024-07-15 by donn