chipfoundry/openlane2 #514: VHDL flow not working as `ghdl` not found inside `yosys`

hpretl, 2024-07-17:
### Description

A user tried to use the VHDL flow in OpenLane in our IIC-OSIC-TOOLS image, see https://github.com/iic-jku/IIC-OSIC-TOOLS/issues/43.

The error is that the command `ghdl` is not working inside Yosys. If I start Yosys just by
```
> yosys
```
then running `ghdl` inside, it throws the error `ERROR: No such command: ghdl (type 'help' for a command overview)`.

If I start Yosys with
```
> yosys -m ghdl
```
then running `ghdl` inside Yosys works fine!

Question: What is wrong with the OpenLane setup in our image?

### Expected Behavior

VHDLClassic flow should work.

### Environment report

```sh
Failed to get Docker info: [Errno 2] No such file or directory: 'docker'
Failed to get Nix info: [Errno 2] No such file or directory: 'nix'
kernel: Linux
kernel_version: 6.6.32-linuxkit
supported: True
distro: ubuntu
distro_version: 22.04
python_version: 3.10.12
python_path:
  - /usr/local/lib/python3.10/dist-packages/openlane
  - /headless/.local/lib/python3.10/site-packages
  - /usr/lib/python310.zip
  - /usr/lib/python3.10
  - /usr/lib/python3.10/lib-dynload
  - /usr/local/lib/python3.10/dist-packages
  - /usr/lib/python3/dist-packages
  - /usr/lib/python3.10/dist-packages
  - /foss/tools/ngspyce/154a272/local/lib/python3.10/dist-packages
  - /foss/tools/pyopus/0.11/local/lib/python3.10/dist-packages
  - /foss/tools/yosys/share/yosys/python3
  - /foss/tools/klayout/v0.29.4/pymod
tkinter: True
container_info: None
nix_info: None
```

### Reproduction material

None needed.

### Relevant log output

```sh
See above.
```

hpretl, 2024-07-17:
@donn Maybe the above issue just boils down to the question: How is OpenLane calling `yosys`, and how do you load modules like `ghdl`?

hpretl, 2024-07-17:
I don't understand nix, but it looks like there is some kind of wrapper involved?

hpretl, 2024-07-22:
We implemented now a wrapper similar to the above one in our IIC-OSIC-TOOLS. An alternative way, I think cleaner, is that OpenLane adapts the calling of yosys. Like using `yosys` for Verilog, or using `yosys -m gdhl` for VHDL, or `yosys -m systemverilog`.

Or just use `yosys -m ghdl -m systemverilog` in any case, if this modules are available.

@donn What do you think?

donn, 2024-07-22:
Doable- will think about it, I'm messing with Yosys stuff atm

hpretl, 2024-07-22:
Please notify me when you change things because then I will remove the Yosys wrapper again :-)

donn, 2024-09-29:
2.2.0 loads plugins explicitly, however, it introduces another issue because you need to pull in https://github.com/YosysHQ/yosys/pull/4553 until it gets merged

Anyways, should be in the next release. Thank you for your patience

closed 2024-09-29 by donn