chipfoundry/openlane2 #522: Cross-OS Determinism

donn, 2024-07-24:
### Description

OpenLane is fairly deterministic on either macOS or Linux, i.e., the same inputs/design will broadly run into the same errors.

However, a design failing on macOS may succeed on Linux, and vice versa.

### Proposal

The culprit appears to be Synthesis. We need to track down why the same file (heck, the same filepath even) may have two different results on two different operating systems.

donnie-j, 2024-07-25:
I can add we encounter this issue with yosys (outside of OpenLane's flow) between yosys builds, from the same repository revision, and on the same OS, but e.g. different OS versions.

We had assumed it was a random seed issue, or a parallel processing issue, but so far these appear to not be supported by experiments to track the cause down.

One more data point:  It appears that some yosys generated netlists for an FPGA target on MacOS will completely fail to place an route, even when moved to a Linux machine for P&R...

donn, 2024-07-25:
My money would be on ABC, specifically.

donn, 2024-07-25:
Well there might be more, but this definitely is A culprit.

[image]

kareefardi, 2024-07-28:
is the flow deterministic post synthesis?

donn, 2024-07-28:
As far as we can tell.