chipfoundry/openlane2 #541: `PL_MAX_DISPLACEMENT_{X,Y}` set atrociously high in both OpenLanes

donn, 2024-09-01:
### Description

The variable that governs how much an instance can be moved after global placement for legalization, `PL_MAX_DISPLACEMENT_{X,Y}`, is set atrociously high at `0.1mm` along the Y axis and `0.5mm`(!!) along the X axis.

This is the case for both OpenLanes.

### Proposal

The defaults should be changed.

kareefardi, 2024-10-21:
Turns out that this is the default value set by OR:
https://github.com/The-OpenROAD-Project/OpenROAD/blob/c12d979536d577c22b7559b2dc346a06d687ea80/src/dpl/src/Opendp.cpp#L129-L130

kareefardi, 2025-02-12:
Action: Remove overrides in default configuration in OpenLane

kareefardi, 2025-02-12:
The X and Y values are passed as as a list for a single option to the tcl API. Changing them both to optional variables will cause a migration headache which is not worth the trouble considering the default value have been there since forver