chipfoundry/openlane2 #543: Inject custom Verilog into the flow

mole99, 2024-09-04:
### Description

Hello! While writing a custom step I'm facing the challenge to inject custom Verilog into the flow.

By this I mean the ability to add new files to the `VERILOG_FILES` variable. Unfortunately, config variables are immutable and therefore it's not directly possible to add new files.

I think this ability would be generally quite useful as you could write custom steps to convert your favorite high-level HDL to Verilog.
For example, you could simply write a step to convert SystemVerilog to Verilog using [sv2v](https://github.com/zachjs/sv2v).

### Proposal

I would propose the following changes:

1. Add a new `DesignFormat` called e.g. `VERILOG`, which will be used to pass on Verilog code from step to step.
2. Update `Verilator.Lint`, `Yosys.JsonHeader`, `Yosys.Synthesis` to use the new `DesignFormat` to access the Verilog sources.
3. Add a new step at the beginning of the `Classic` flow to convert the `VERILOG_FILES` config variable to the `DesignFormat` `VERILOG`.

This way, all steps can be reused for other flows that do not require a `VERILOG_FILES` variable.

The only concern I have is how to deal with the `VHDLClassic` flow. We could:

1. Add a new `DesignFormat` called e.g. `VHDL` to pass on VHDL files.
2. Or, `Yosys.VHDLSynthesis` uses the `VHDL_FILES` variable as usual and writes to the `DesignFormat` `VERILOG`.

Since no other steps use the VHDL code, solution 2 would be simple and straightforward.

I would like to take a shot at implementing this, if you also think this is a workable solution.

Thanks!

donn, 2024-09-05:
For one thing, this would be a breaking change and can arrive in OpenLane 3 at the minimum.

---

The config is not actually immutable in custom flows, [only _steps_ are not allowed to mutate the config](https://openlane2.readthedocs.io/en/latest/reference/architecture.html#steps), and we do take advantage of this, for example, in the Synthesis Exploration flow: https://github.com/efabless/openlane2/blob/main/openlane/flows/synth_explore.py.

So you can write a custom flow that for example takes `VERILOG_FILES`, runs `sv2v`, then `config.copy(VERILOG_FILES=[…])`

That aside, I'm weary of the current way we're doing Verilog and VHDL altogether. I think I want people to be able to provide a heterogenous list of files (`.sv`, `.v`, `.vhd`, …) and `Yosys.Synthesis` should figure out and load them appropriately.

I can integrate `sv2v` into `Yosys.Synthesis` into OpenLane 2.2 if you'd like. That's a non-breaking change and would be fairly quick.

mole99, 2024-09-05:
You are right! Anyone not using one of the predefined flows would need to add this new Step to the very top. That's a bummer...

But great to hear that OL3 is on the horizon 😄

---

Thanks for pointing out that flows can mutate the config. That sounds like a good solution for me as I use a custom flow anyway!

Interesting idea, but I worry that you might loose some information this way that can't be deduced from the file extensions. For example `.sv` can be SV source files or packages that might need to be handled differently by some tools 🤔

Thanks for the offer, but `sv2v` was just an example what one could do! (Still, would be a nice feature to have.) My actual use case is to generate Verilog for a custom FPGA fabric based on the input config via FABulous.