chipfoundry/openlane2 #550: Warning in the PDK LEF files

kareefardi, 2024-09-18:
### Description

```
LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer mcon
```

### Expected Behavior

No warning

### Environment report

```sh
N/A
```

### Reproduction material

Run the smoke test example through the flow

### Relevant log output

```sh
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer mcon
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer mcon
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer via
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer via
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer via2
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer via2
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer via3
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer via3
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer via4
[WARNING DRT-0349] LEF58_ENCLOSURE with no CUTCLASS is not supported. Skipping for layer via4
```

donn, 2024-09-18:
Nothing's wrong with the PDK. This is a matter of OpenROAD not supporting this specifically.

We can suppress the warning in OpenLane, but we shouldn't modify the PDK for this.

kareefardi, 2024-09-18:
It could be a warning related to the LEF spec that is printed by OpenROAD

kareefardi, 2024-10-01:
Started a discussion in OpenROAD https://github.com/The-OpenROAD-Project/OpenROAD/discussions/5839