chipfoundry/openlane2 #555: Add an OpenROAD.WriteAbstractLEF step

smunaut, 2024-09-21:
### Description

Currently when building a macro with OL2, the only option to get a LEF out (which is required for integration of the macro later on) is to use Magic.

For iHP there is no Magic support yet, but OpenROAD can actually generate a LEF.

Currently in `write_views` if you have `SAVE_OPENROAD_LEF` it will save an LEF from OpenROAD however there are two issues with that :

* It's not recognized by OL2 as a real output because the steps won't have `DesignFormat.LEF` listed in their outputs
* You ideally want the `-bloat_occupied_layers` option specified ( equivalent of the `-hide` option in magic ) when writing a macro LEF.

### Proposal

So I think the best option is to have a separate step that one can use to achieve that in a custom flow :

```python
@Step.factory.register()
class WriteAbstractLEF(OpenROADStep):

    id = "OpenROAD.WriteAbstractLEF"
    name = "Write Abstract LEF"

    outputs = [
        DesignFormat.LEF
    ]

    def get_script_path(self):
        return os.path.join(get_script_dir(), "openroad", "write_abstract_lef.tcl")

```

And then `write_abstract_lef.tcl` something like :

```tcl
source $::env(SCRIPTS_DIR)/openroad/common/io.tcl
read_current_odb
write_abstract_lef -bloat_occupied_layers $::env(STEP_DIR)/$::env(DESIGN_NAME).lef
```

donn, 2024-09-22:
Should be fairly easy to get through.

donn, 2024-09-22:
So there's `OpenROAD.WriteViews` that does have `DesignFormat.OPENROAD_LEF` as a "true" output. All it's missing is `-bloat_occupied_layers`. Would that work for you?

smunaut, 2024-09-22:
The `-bloat_occupied_layers` is required for sure else during integration the top-level tries to route stuff over the macro which is always a mess, but if that can be added as an option, then I guess I could use `WriteViews`

donn, 2024-09-29:
Added to in 2.2.0 -- please wait for release

PR #557 (2024-09-26): Update `OpenROAD.WriteViews` to allow "bloating" occupied layers in output LEF
* `OpenROAD.WriteViews`:
  * Add `OPENROAD_LEF_BLOAT_OCCUPIED_LAYERS` with a default value of `true`

---
Fixes #555
closed 2024-09-29 by donn