chipfoundry/openlane2 #569: openlane2 non going through --smoke-test

ctalar, 2024-10-05:
### Description

On MacOS BigSur ver. 11.7.10 (with processor intel i7) the command:
`openlane --log-level ERROR --condensed --show-progress-bar --smoke-test`

exits with errors (the issue seems to be klayout related)

`klayout -v`
KLayout 0.29.1

### Expected Behavior

Complete the flow all the way to the generation of the gds

### Environment report

```sh
openlane.env_info || python3 ./openlane/env_info.py
kernel: Darwin
kernel_version: 20.6.0
supported: True
distro: macOS
distro_version: 10.16
python_version: 3.11.9
python_path:
  - /nix/store/7n90jnz7706vhbzxsmbsvkgkbp21j868-python3.11-openlane-2.1.9/bin
  - /nix/store/h6zbwshm93y4a3936fvh5grav1swkzvk-python3-3.11.9-env/lib/python3.11/site-packages
  - /nix/store/6jy28lfjr42adzi10p158aq6m1q9qgs0-python3-3.11.9/lib/python311.zip
  - /nix/store/6jy28lfjr42adzi10p158aq6m1q9qgs0-python3-3.11.9/lib/python3.11
  - /nix/store/6jy28lfjr42adzi10p158aq6m1q9qgs0-python3-3.11.9/lib/python3.11/lib-dynload
  - /nix/store/6jy28lfjr42adzi10p158aq6m1q9qgs0-python3-3.11.9/lib/python3.11/site-packages
  - /nix/store/7n90jnz7706vhbzxsmbsvkgkbp21j868-python3.11-openlane-2.1.9/lib/python3.11/site-packages
  - /nix/store/vw59dxsnalr63j7ggr31q407fwbgg7fh-python3.11-click-8.1.7/lib/python3.11/site-packages
  - /nix/store/9ni8a59d2qg5k4pdyr42s228h40ckrpv-python3.11-cloup-3.0.5/lib/python3.11/site-packages
  - /nix/store/7ff0b5rpacwwrkwf6a0gj29xz66l0njj-python3.11-pyyaml-6.0.1/lib/python3.11/site-packages
  - /nix/store/0f1qnjr36x9i76ckavh4q8c8sfm3ncil-python3.11-rich-13.7.1/lib/python3.11/site-packages
  - /nix/store/pzvkwr0w2lj96jlprr3i1c0vy2hbzqag-python3.11-markdown-it-py-3.0.0/lib/python3.11/site-packages
  - /nix/store/3q4g27d0876gwgc4g7jdk8l44b4lmk7w-python3.11-mdurl-0.1.2/lib/python3.11/site-packages
  - /nix/store/sghrwdn6lhpj569vghyl6czdciq8ym5p-python3.11-pygments-2.17.2/lib/python3.11/site-packages
  - /nix/store/xxss12nn5yp4hv23489inn3cfrymgy18-python3.11-requests-2.31.0/lib/python3.11/site-packages
  - /nix/store/jhadp06fvxgdqi5fvpwcr24j0f5zn4na-python3.11-brotlicffi-1.1.0.0/lib/python3.11/site-packages
  - /nix/store/3dw40f049qqqqpw4rhnzkx9lvz48fizp-python3.11-cffi-1.16.0/lib/python3.11/site-packages
  - /nix/store/m8b04i0g8nkm64ljfbd2jf0r678h56hq-python3.11-pycparser-2.22/lib/python3.11/site-packages
  - /nix/store/ql7xgha07jdqpr53gscm388rxmmn104h-python3.11-certifi-2024.02.02/lib/python3.11/site-packages
  - /nix/store/vb5bfaq3sizmc0fvczxpqzg8xrh82pix-python3.11-charset-normalizer-3.3.2/lib/python3.11/site-packages
  - /nix/store/8rlszfxx2q3mi8b24d9nmzf7iyfpab7r-python3.11-idna-3.7/lib/python3.11/site-packages
  - /nix/store/5bcvg026s5rx2qw4nvfx4ik9l8xry67h-python3.11-urllib3-2.2.1/lib/python3.11/site-packages
  - /nix/store/r1qm4ngrbx5xmxyhy05djdyykccdamd5-python3.11-pcpp-1.30/lib/python3.11/site-packages
  - /nix/store/c3d105x78d8i0kr2jqyaw4p4ib10kqm0-python3.11-volare/lib/python3.11/site-packages
  - /nix/store/a8x499l1lq631i8flfvly604awgmsa7q-python3.11-httpx-0.27.0/lib/python3.11/site-packages
  - /nix/store/9dgxmwjx5qakby6rw7h87kybzi46m28z-python3.11-anyio-4.3.0/lib/python3.11/site-packages
  - /nix/store/syclchnvxr1aba8db6mp8gwfl9di8dsj-python3.11-sniffio-1.3.1/lib/python3.11/site-packages
  - /nix/store/d7cf8g64v7nqakzlywv4ksyyrhksgaz8-python3.11-httpcore-1.0.5/lib/python3.11/site-packages
  - /nix/store/7b982kb0688kn2a3br061h16k440i2v3-python3.11-h11-0.14.0/lib/python3.11/site-packages
  - /nix/store/prgcw1i05l698d2ckg0w2sdl5frqbah0-python3.11-zstandard-0.22.0/lib/python3.11/site-packages
  - /nix/store/ydkd7rm5ap0g8l5ngk0a1h7rgxpws6pz-python3.11-truststore-0.8.0/lib/python3.11/site-packages
  - /nix/store/yglp8cxpn4zsw20q8f4az3h9cswhajl2-python3.11-aiohttp-3.9.5/lib/python3.11/site-packages
  - /nix/store/8wa4pzxx7q41qr1gsr962vr473m93mh8-python3.11-attrs-23.2.0/lib/python3.11/site-packages
  - /nix/store/znf4fggwcif3iyhgh32ckqc39rshzd7m-python3.11-multidict-6.0.5/lib/python3.11/site-packages
  - /nix/store/k92m9i0v1ca1kcy7aqmbwl5a0fa7aajc-python3.11-async-timeout-4.0.3/lib/python3.11/site-packages
  - /nix/store/lx7fwjh3acwbby1xknhvrj6zhwp51mnz-python3.11-typing-extensions-4.11.0/lib/python3.11/site-packages
  - /nix/store/d638h318y06dlpzkhmwac78mp5p2myz3-python3.11-yarl-1.9.4/lib/python3.11/site-packages
  - /nix/store/nb79gip3r2yrnicl28bbdhy9r8nshjbm-python3.11-frozenlist-1.4.1/lib/python3.11/site-packages
  - /nix/store/pm595hsja782fn9zhrrsq5dm8kir1dfg-python3.11-aiosignal-1.3.1/lib/python3.11/site-packages
  - /nix/store/yrw71drihc4d2q63flw4dw842s38nrds-python3.11-aiodns-3.2.0/lib/python3.11/site-packages
  - /nix/store/km1minc92ha6iqdkqv2p469szybcsk1h-python3.11-pycares-4.4.0/lib/python3.11/site-packages
  - /nix/store/803w1ji9hmx94xrjnjq25a1yipcyzpd9-python3.11-brotli-1.1.0/lib/python3.11/site-packages
  - /nix/store/nv7ki12457wi1grf7rqkcaca9vgy6w9p-python3.11-cryptography-42.0.5/lib/python3.11/site-packages
  - /nix/store/rl0by88vlhc56z25igqn2i3i0py7wa1r-python3.11-pyopenssl-24.1.0/lib/python3.11/site-packages
  - /nix/store/ymy61ac42vf3mx2a6g0bl23r52yrrr60-python3.11-trustme-1.1.0/lib/python3.11/site-packages
[... 152 lines cut ...]
               ERROR: Unable to load plugin: /nix/store/6mm1g37bl0lwf9gxbql2q06bphb3jz0b-klayout/lib/db_plugins/libpcb.0.dylib                                                   
               ERROR: Unable to load plugin: /nix/store/6mm1g37bl0lwf9gxbql2q06bphb3jz0b-klayout/lib/db_plugins/libpcb.dylib                                                     
               'LoadLayoutOptions' object has no attribute 'lefdef_config'                                                                                                       
                                                                                                                                                                                 
[10:41:49] [E] Full log file: '../../../tmp/tmpujok5rieopenlane2/smoke_test_design/runs/RUN_2024-10-05_10-40-45/55-klayout-streamout/klayout-streamout.log'          step.py:1322
Classic - Stage 58 - GDSII Stream Out (KLayout) ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╺━━━━━━━━━ 57/76 0:01:03
[10:41:49] [E] The following error was encountered while running the flow:                                                                                        __main__.py:160
               GDSII Stream Out (KLayout): subprocess (1, ['/nix/store/h6zbwshm93y4a3936fvh5grav1swkzvk-python3-3.11.9-env/bin/python3.11',                                      
               '/nix/store/h6zbwshm93y4a3936fvh5grav1swkzvk-python3-3.11.9-env/lib/python3.11/site-packages/openlane/scripts/klayout/stream_out.py',                             
               '/tmp/tmpujok5rieopenlane2/smoke_test_design/runs/RUN_2024-10-05_10-40-45/50-openroad-fillinsertion/spm.def', '--output',                                         
               '/tmp/tmpujok5rieopenlane2/smoke_test_design/runs/RUN_2024-10-05_10-40-45/55-klayout-streamout/spm.klayout.gds', '--top', 'spm', '--lyp',                         
               '/Users/claudio/share/pdks/volare/sky130/versions/bdc9412b3e468c102d01b7cf6337be06ec6e9c9a/sky130A/libs.tech/klayout/tech/sky130A.lyp', '--lyt',                  
               '/Users/claudio/share/pdks/volare/sky130/versions/bdc9412b3e468c102d01b7cf6337be06ec6e9c9a/sky130A/libs.tech/klayout/tech/sky130A.lyt', '--lym',                  
               '/Users/claudio/share/pdks/volare/sky130/versions/bdc9412b3e468c102d01b7cf6337be06ec6e9c9a/sky130A/libs.tech/klayout/tech/sky130A.map',                           
               '--input-lef',                                                                                                                                                    
               '/Users/claudio/share/pdks/volare/sky130/versions/bdc9412b3e468c102d01b7cf6337be06ec6e9c9a/sky130A/libs.ref/sky130_fd_sc_hd/techlef/sky130_fd_sc_h                
               d__nom.tlef', '--input-lef',                                                                                                                                      
               '/Users/claudio/share/pdks/volare/sky130/versions/bdc9412b3e468c102d01b7cf6337be06ec6e9c9a/sky130A/libs.ref/sky130_fd_sc_hd/lef/sky130_ef_sc_hd.le                
               f', '--input-lef',                                                                                                                                                
[... cut ...]
               f', '--with-gds-file',                                                                                                                                            
               '/Users/claudio/share/pdks/volare/sky130/versions/bdc9412b3e468c102d01b7cf6337be06ec6e9c9a/sky130A/libs.ref/sky130_fd_sc_hd/gds/sky130_fd_sc_hd.gd                
               s']) failed                                                                                                                                                       
[10:41:49] [E] OpenLane will now quit.
```

donn, 2024-10-05:
Hi Claudio, thanks for creating an issue- do you have KLayout installed externally at all? i.e. Can you run `klayout --version` without going into the Nix shell first?

ctalar, 2024-10-06:
Hi Mohammed,
Yes, I have also klayout installed externally.
(However, it is an older version since I am on macOS BigSur and I installed the most recent “packaged” version I found: it is 0.28.12)

Should I uninstall it, to have the klayout under openlane2 to work ?

Thanks,
Claudio

ctalar, 2024-10-06:
Hi Mohamed, first of all thank you for the help. By the way I joined the slack.
I removed the "external" klayout and made sure is not in the path.
`(base) Claudios-MacBook-Pro:~ claudio$ open -a klayout`
Unable to find application named 'klayout'
`(base) Claudios-MacBook-Pro:~ claudio$ echo $PATH`
/Applications/MATLAB_R2023b.app/bin:/Users/claudio/anaconda3/bin:/Users/claudio/anaconda3/condabin:/Users/claudio/scripts:/Users/claudio/opt/xschem/bin:/Users/claudio/opt/ngspice/bin:/Users/claudio/opt/magic/bin:/Users/claudio/opt/netgen/bin:/Users/claudio/.nix-profile/bin:/nix/var/nix/profiles/default/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/TeX/texbin:/opt/X11/bin:/Library/Apple/usr/bin:/opt/local/bin:/opt/local/sbin

Unfortunately the problem is still there.
`[nix-shell:~/openlane2]$ openlane --log-level ERROR --condensed --show-progress-bar --smoke-test
`
[10:29:19] [E] Subprocess had a non-zero exit.                                                                                                                                                                step.py:1314
[10:29:19] [E] Last 10 line(s):                                                                                                                                                                               step.py:1319
               ERROR: Unable to load plugin: /nix/store/6mm1g37bl0lwf9gxbql2q06bphb3jz0b-klayout/lib/db_plugins/libnet_tracer.dylib                                                                                       
               ERROR: Unable to load plugin: /nix/store/6mm1g37bl0lwf9gxbql2q06bphb3jz0b-klayout/lib/db_plugins/liboasis.0.29.1.dylib                                                                                     
               ERROR: Unable to load plugin: /nix/store/6mm1g37bl0lwf9gxbql2q06bphb3jz0b-klayout/lib/db_plugins/liboasis.0.29.dylib                                                                                       
               ERROR: Unable to load plugin: /nix/store/6mm1g37bl0lwf9gxbql2q06bphb3jz0b-klayout/lib/db_plugins/liboasis.0.dylib                                                                                          
               ERROR: Unable to load plugin: /nix/store/6mm1g37bl0lwf9gxbql2q06bphb3jz0b-klayout/lib/db_plugins/liboasis.dylib                                                                                            
               ERROR: Unable to load plugin: /nix/store/6mm1g37bl0lwf9gxbql2q06bphb3jz0b-klayout/lib/db_plugins/libpcb.0.29.1.dylib                                                                                       
               ERROR: Unable to load plugin: /nix/store/6mm1g37bl0lwf9gxbql2q06bphb3jz0b-klayout/lib/db_plugins/libpcb.0.29.dylib                                                                                         
               ERROR: Unable to load plugin: /nix/store/6mm1g37bl0lwf9gxbql2q06bphb3jz0b-klayout/lib/db_plugins/libpcb.0.dylib                                                                                            
               ERROR: Unable to load plugin: /nix/store/6mm1g37bl0lwf9gxbql2q06bphb3jz0b-klayout/lib/db_plugins/libpcb.dylib                                                                                              
               'LoadLayoutOptions' object has no attribute 'lefdef_config'                                                                                                                                                
                                                                                                                                                                                                                          
[10:29:19] [E] Full log file: '../../../tmp/tmpk2u6u_z9openlane2/smoke_test_design/runs/RUN_2024-10-06_10-28-16/55-klayout-streamout/klayout-streamout.log'                                                   step.py:1322
Classic - Stage 58 - GDSII Stream Out (KLayout) ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╺━━━━━━━━━ 57/76 0:01:03
[10:29:19] [E] The following error was encountered while running the flow:                                                                                                                                 __main__.py:160
               GDSII Stream Out (KLayout): subprocess (1, ['/nix/store/h6zbwshm93y4a3936fvh5grav1swkzvk-python3-3.11.9-env/bin/python3.11',                                                                               
               '/nix/store/h6zbwshm93y4a3936fvh5grav1swkzvk-python3-3.11.9-env/lib/python3.11/site-packages/openlane/scripts/klayout/stream_out.py',                                                                      
               '/tmp/tmpk2u6u_z9openlane2/smoke_test_design/runs/RUN_2024-10-06_10-28-16/50-openroad-fillinsertion/spm.def', '--output',                                                                                  
               '/tmp/tmpk2u6u_z9openlane2/smoke_test_design/runs/RUN_2024-10-06_10-28-16/55-klayout-streamout/spm.klayout.gds', '--top', 'spm', '--lyp',                                                                  
               '/Users/claudio/share/pdks/volare/sky130/versions/bdc9412b3e468c102d01b7cf6337be06ec6e9c9a/sky130A/libs.tech/klayout/tech/sky130A.lyp', '--lyt',                                                           
[... cut ...]
ERROR: Unable to load plugin: /nix/store/6mm1g37bl0lwf9gxbql2q06bphb3jz0b-klayout/lib/db_plugins/liboasis.dylib
ERROR: Unable to load plugin: /nix/store/6mm1g37bl0lwf9gxbql2q06bphb3jz0b-klayout/lib/db_plugins/libpcb.0.29.1.dylib
ERROR: Unable to load plugin: /nix/store/6mm1g37bl0lwf9gxbql2q06bphb3jz0b-klayout/lib/db_plugins/libpcb.0.29.dylib
ERROR: Unable to load plugin: /nix/store/6mm1g37bl0lwf9gxbql2q06bphb3jz0b-klayout/lib/db_plugins/libpcb.0.dylib
ERROR: Unable to load plugin: /nix/store/6mm1g37bl0lwf9gxbql2q06bphb3jz0b-klayout/lib/db_plugins/libpcb.dylib

donn, 2024-10-06:
We do claim to support macOS 11, so if it doesn't work on macOS 11 either the documentation or the code has to be updated.

I'll take a copy of Big Sur with me to the office tomorrow and see what's going on.

mguthaus, 2024-12-17:
Check if there's a copy of the klayout Python module installed externally as well. This caused an issue for me

ctalar, 2024-12-18:
@mguthaus thank you for the hint. 
I indeed had a klayout Python module so I removed it, and checked that there are no longer any external klayout Python, but unfortunately the issue is still there ?
`pip list | grep kla`

@Don does the installation on Big Sur work for you ?

donn, 2024-12-18:
Sorry, forgot to follow up - awkwardly, Big Sur is the one version for Apple Silicon macOS that does NOT run under [UTM](https://github.com/utmapp/UTM). So I was unable to verify that it is indeed a macOS Big Sur issue.