chipfoundry/openlane2 #570: Checking the results problem

ripudamank2, 2024-10-06:
### Description

I just followed the steps mention in below link 
https://openlane2.readthedocs.io/en/latest/getting_started/newcomers/index.html  and when i try to open klyaout its giving error.

[nix-shell:~/Documents]$ openlane --last-run --flow openinklayout my_designs/pm32/config.json 
[22:27:29] INFO     Loading the design configuration file has generated the following warnings:                                                                            config.py:669
[22:27:29] WARNING  Key 'VERILOG_FILES' provided is unused by the current flow.                                                                                            config.py:673
[22:27:29] INFO     Using existing run at 'RUN_2024-10-06_00-41-27' with the 'Opening in KLayout' flow.                                                                      flow.py:577
[22:27:29] VERBOSE  Using state at '/home/asic/Documents/my_designs/pm32/runs/RUN_2024-10-06_00-41-27/72-misc-reportmanufacturability/state_out.json'.                       flow.py:616
[22:27:29] INFO     Starting…                                                                                                                                          sequential.py:278
───────────────────────────────────────────────────────────────────────────────────── Open In GUI ──────────────────────────────────────────────────────────────────────────────────────
[22:27:29] VERBOSE  Running 'KLayout.OpenGUI'…                                                                                                                              step.py:1088
Opening in KLayout - Stage 1 - Open In GUI ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 0/1 0:00:00Authorization required, but no authorization protocol specified

**Info: Could not load the Qt platform plugin "xcb" in "" even though it was found.
Fatal: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.**

**Available platform plugins are: wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.**

Opening in KLayout - Stage 1 - Open In GUI ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 0/1 0:00:00
[22:27:29] ERROR    The flow has encountered an unexpected error:                                                                                                        __main__.py:156
                    Open In GUI: Interrupted (SIGABRT)                                                                                                                                  
[22:27:29] ERROR    OpenLane will now quit.                              

### Expected Behavior

openlane --last-run --flow openinklayout ~/my_designs/pm32/config.json

its should have open the gui of klayout 

### Environment report

```sh
Failed to get Docker info: [Errno 2] No such file or directory: 'docker'
Failed to get Nix info: [Errno 13] Permission denied: 'nix'
kernel: Linux
kernel_version: 6.8.0-45-generic
supported: True
distro: ubuntu
distro_version: 24.04
python_version: 3.11.9
python_path:
  - /nix/store/bjh9pl8hrw5bz2mayra4z6l6znl5ncyr-python3.11-openlane-2.1.9/bin
  - /nix/store/ly8g3mcmscy67krhznzmkhbdvmqi6y4d-python3-3.11.9-env/lib/python3.11/site-packages
  - /nix/store/7hnr99nxrd2aw6lghybqdmkckq60j6l9-python3-3.11.9/lib/python311.zip
  - /nix/store/7hnr99nxrd2aw6lghybqdmkckq60j6l9-python3-3.11.9/lib/python3.11
  - /nix/store/7hnr99nxrd2aw6lghybqdmkckq60j6l9-python3-3.11.9/lib/python3.11/lib-dynload
  - /nix/store/7hnr99nxrd2aw6lghybqdmkckq60j6l9-python3-3.11.9/lib/python3.11/site-packages
  - /nix/store/bjh9pl8hrw5bz2mayra4z6l6znl5ncyr-python3.11-openlane-2.1.9/lib/python3.11/site-packages
  - /nix/store/4flhri1nwh8n0d8a4bx2yjnyigpz00bg-python3.11-click-8.1.7/lib/python3.11/site-packages
  - /nix/store/ralzzlb48f9dn7dcd5hlbf80vi3vfpdm-python3.11-cloup-3.0.5/lib/python3.11/site-packages
  - /nix/store/v05ziwqwbfnrgdd7bgqaygj4wj2asfgs-python3.11-pyyaml-6.0.1/lib/python3.11/site-packages
  - /nix/store/gkfkq6d57dran0fkgz81738zc8wc3w13-python3.11-rich-13.7.1/lib/python3.11/site-packages
  - /nix/store/klj6938h32ra3mlwkavcr285d2387h38-python3.11-markdown-it-py-3.0.0/lib/python3.11/site-packages
  - /nix/store/kalmrzx2fllfk713lm0m1yb1p5rqw1ly-python3.11-mdurl-0.1.2/lib/python3.11/site-packages
  - /nix/store/cid5kxmrjdgikliwbhry3wq6j0117595-python3.11-pygments-2.17.2/lib/python3.11/site-packages
  - /nix/store/nicjg1xpimrn3zfbndwix25gphv88zlx-python3.11-requests-2.31.0/lib/python3.11/site-packages
  - /nix/store/cf9j4x1ka7s6aajgswvw23d36q4d2a2y-python3.11-brotlicffi-1.1.0.0/lib/python3.11/site-packages
  - /nix/store/n4dzhaf8ybxmmk4v47mx00mlllrmg0f0-python3.11-cffi-1.16.0/lib/python3.11/site-packages
  - /nix/store/i65kg4y70qxgw8sb1s35yg1sssc6qxa4-python3.11-pycparser-2.22/lib/python3.11/site-packages
  - /nix/store/qgglxpjjja3qpxi6mayabj417n16d3lh-python3.11-certifi-2024.02.02/lib/python3.11/site-packages
  - /nix/store/jrr6l56xssk4szz6xxk9mxhk8pxwghhg-python3.11-charset-normalizer-3.3.2/lib/python3.11/site-packages
  - /nix/store/y84qvvzrarmks4k7qb9ras6qfsxksnds-python3.11-idna-3.7/lib/python3.11/site-packages
  - /nix/store/2sd6mncv58k6065w8cf9b5pmagf2jc2f-python3.11-urllib3-2.2.1/lib/python3.11/site-packages
  - /nix/store/8xal91yxlsjs1128i5qzn0zs602x6a6d-python3.11-pcpp-1.30/lib/python3.11/site-packages
  - /nix/store/rk8pjysd593p6wcgw2nk8v4vxd4m7082-python3.11-volare/lib/python3.11/site-packages
  - /nix/store/3jpad8nm52qp0c9wa2hjn129jppksc53-python3.11-httpx-0.27.0/lib/python3.11/site-packages
  - /nix/store/zgxhw6fw9imv7csy78d3226zgf0bcv7a-python3.11-anyio-4.3.0/lib/python3.11/site-packages
  - /nix/store/bvvfsk50fxqrh84siy70i48ycmsprak1-python3.11-sniffio-1.3.1/lib/python3.11/site-packages
  - /nix/store/yi418b5lldai2c0adkczs8is60c5d8ns-python3.11-httpcore-1.0.5/lib/python3.11/site-packages
  - /nix/store/61nxkwrszkj7kq8d5fp4ch1q8206nlry-python3.11-h11-0.14.0/lib/python3.11/site-packages
  - /nix/store/462fi500v0yppj7fz5f46dpqg99i3g3z-python3.11-zstandard-0.22.0/lib/python3.11/site-packages
  - /nix/store/kzhc4zn56nhsbxh6bri4db7r2g56pkdq-python3.11-truststore-0.8.0/lib/python3.11/site-packages
  - /nix/store/5wbpm0svz289hyvrhbjws54bd6n55gy1-python3.11-aiohttp-3.9.5/lib/python3.11/site-packages
  - /nix/store/vhmaphzfp7gz0yshamq728f6z312gfx9-python3.11-attrs-23.2.0/lib/python3.11/site-packages
  - /nix/store/6hh90hjln36y3gxvdjdgsjvxzp45ii36-python3.11-multidict-6.0.5/lib/python3.11/site-packages
  - /nix/store/xlkim03yac6jyf93p3s531gq1cwark6y-python3.11-async-timeout-4.0.3/lib/python3.11/site-packages
  - /nix/store/k920s30a5dqzndfj4ic23p7rs0pc927k-python3.11-typing-extensions-4.11.0/lib/python3.11/site-packages
  - /nix/store/hrq35mrgv32iiakb09k2f3pvaic5pyps-python3.11-yarl-1.9.4/lib/python3.11/site-packages
  - /nix/store/vwvxnkxdjqnsy10qb010d6l0yj6rlham-python3.11-frozenlist-1.4.1/lib/python3.11/site-packages
  - /nix/store/1i2x9gv9svw9xdp5zzinnyd4gpjkbw9v-python3.11-aiosignal-1.3.1/lib/python3.11/site-packages
  - /nix/store/fglfx6csv8zpif1pfn95rwwb168lbila-python3.11-aiodns-3.2.0/lib/python3.11/site-packages
  - /nix/store/1y7aypwp6a0pjdkl2favnfqi30vylagv-python3.11-pycares-4.4.0/lib/python3.11/site-packages
  - /nix/store/mj6hbmhs6s4ki6bs8y05mdgnk9brlr89-python3.11-brotli-1.1.0/lib/python3.11/site-packages
  - /nix/store/zsq2bhixjp3y9qzkjkvndhn6n6s7ixx4-python3.11-cryptography-42.0.5/lib/python3.11/site-packages
  - /nix/store/kvslfql2hk3v2h7xnymjwbaq88mnch0c-python3.11-pyopenssl-24.1.0/lib/python3.11/site-packages
  - /nix/store/jwva5ipsh6gm9xfnhdl6y25dc13gq0n5-python3.11-trustme-1.1.0/lib/python3.11/site-packages
  - /nix/store/f89l49aw0ia57dx28h74hzppyjbn5ykn-python3.11-socksio-1.0.0/lib/python3.11/site-packages
  - /nix/store/pv5vvspngggbxqdndyhrs4dhrbr2bwg4-python3.11-tkinter-3.11.9/lib/python3.11/site-packages
  - /nix/store/915v7rm20wlanij3sax1p0kzadqg6cbv-python3.11-lxml-5.1.0/lib/python3.11/site-packages
  - /nix/store/dmdqig719d3i2r9qyl4bz13dl6c5rd7n-python3.11-deprecated-1.2.14/lib/python3.11/site-packages
  - /nix/store/n70y9hzjmlv9g5aj4sg7lnbivkhl3qy5-python3.11-wrapt-1.16.0/lib/python3.11/site-packages
  - /nix/store/frmrhcq8q5vvy63maj21nqjshlnvnl3k-python3.11-libparse/lib/python3.11/site-packages
  - /nix/store/4vd46gwk0iglcwr3rjpbn3yvnq99wshm-python3.11-psutil-5.9.8/lib/python3.11/site-packages
  - /nix/store/x23vx172vfjczrmj8db9nyw4vns6x9d8-klayout-pymod/lib/python3.11/site-packages
  - /nix/store/vmkhknivrpf85h7mlkmhqg7fazcaahdq-python3.11-rapidfuzz-3.9.1/lib/python3.11/site-packages
  - /nix/store/zalfkfjglwiclp72z3rq9nq9cj57a7z0-python3.11-ioplace_parser/lib/python3.11/site-packages
  - /nix/store/7zc3jpvv1cpbppac12jpwcbrpqqxq9zs-python3.11-antlr4-python3-runtime-4.10.1/lib/python3.11/site-packages
tkinter: True
container_info: None
nix_info: None
```

### Reproduction material

i dont have 

### Relevant log output

```sh
which files you want i can share with you . 
I already share the console output in above section
```

donn, 2024-10-06:
Are you on Wayland? You may need to switch to X11 until we figure out why this happens...

ripudamank2, 2024-10-08:
Thank you after switch to X11 its working . I am able to open klayout or openroad gui.

ripudamank2, 2024-10-08:
One General Question are there any projects which i can try to familiar with openlane. 
Can you please share some links .

donn, 2024-10-09:
The newcomers' guide is very good for this. https://openlane2.readthedocs.io/en/latest/getting_started/newcomers/index.html

ripudamank2, 2024-10-09:
This I already tried.  is there any more project links. Which i can try ?