chipfoundry/openlane2 #590: Synthesis no longer works via Google Colab

ajkourabi, 2024-10-21:
### Description

When running the standard [template colab](https://colab.research.google.com/github/efabless/openlane2/blob/main/notebook.ipynb), the Synthesis step no longer works. 

### Expected Behavior

The synthesis step to be completed. 

### Environment report

```sh
I am running the standard set up provided via the google colab template.
```

### Reproduction material

https://colab.research.google.com/github/efabless/openlane2/blob/main/notebook.ipynb

### Relevant log output

```sh
─────────────────────────────────────────────────── Synthesis ────────────────────────────────────────────────────
[17:20:10] VERBOSE  Running 'Yosys.Synthesis' at 'openlane_run/1-yosys-synthesis'…                     step.py:1122
[17:20:10] VERBOSE  Logging subprocess to 'openlane_run/1-yosys-synthesis/yosys-synthesis.log'…        step.py:1318
                                                                                                                   
/----------------------------------------------------------------------------\                                     
|  yosys -- Yosys Open SYnthesis Suite                                       |                                     
|  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |                                     
|  Distributed under an ISC-like license, type "license" to see terms        |                                     
\----------------------------------------------------------------------------/                                     
Yosys 0.46 (git sha1 e97731b9dda91fa5fa53ed87df7c34163ba59a41, clang++ 17.0.6 -fPIC -O3)                           
Loaded SDC plugin                                                                                                  
Traceback (most recent call last):                                                                                 
File "/content/openlane_ipynb/openlane/scripts/pyosys/synthesize.py", line 36, in <module>                         
import click                                                                                                       
File                                                                                                               
"/nix/store/h0n4i85hqpprsqs9j6yydhj5h6vlh0j7-yosys-with-plugins-python3-3.11.9-env/lib/yosys-with-plugins-python3.1
1/site-packages/click/__init__.py", line 7, in <module>                                                            
from .core import Argument as Argument                                                                             
File                                                                                                               
"/nix/store/h0n4i85hqpprsqs9j6yydhj5h6vlh0j7-yosys-with-plugins-python3-3.11.9-env/lib/yosys-with-plugins-python3.1
1/site-packages/click/core.py", line 16, in <module>                                                               
from . import types                                                                                                
File                                                                                                               
"/nix/store/h0n4i85hqpprsqs9j6yydhj5h6vlh0j7-yosys-with-plugins-python3-3.11.9-env/lib/yosys-with-plugins-python3.1
1/site-packages/click/types.py", line 5, in <module>                                                               
from datetime import datetime                                                                                      
File "/nix/store/7hnr99nxrd2aw6lghybqdmkckq60j6l9-python3-3.11.9/lib/python3.11/datetime.py", line 12, in <module> 
import math as _math                                                                                               
ModuleNotFoundError: No module named 'math'                                                                        
ERROR: Python interpreter encountered an exception.                                                                
[17:20:11] ERROR    Subprocess had a non-zero exit.                                                    step.py:1364
[17:20:11] ERROR    Last 10 line(s):                                                                   step.py:1369
                      File                                                                                         
                    "/nix/store/h0n4i85hqpprsqs9j6yydhj5h6vlh0j7-yosys-with-plugins-python3-3.11.9-env             
                    /lib/yosys-with-plugins-python3.11/site-packages/click/__init__.py", line 7, in                
                    <module>                                                                                       
                        from .core import Argument as Argument                                                     
                      File                                                                                         
                    "/nix/store/h0n4i85hqpprsqs9j6yydhj5h6vlh0j7-yosys-with-plugins-python3-3.11.9-env             
                    /lib/yosys-with-plugins-python3.11/site-packages/click/core.py", line 16, in                   
                    <module>                                                                                       
                        from . import types                                                                        
                      File                                                                                         
                    "/nix/store/h0n4i85hqpprsqs9j6yydhj5h6vlh0j7-yosys-with-plugins-python3-3.11.9-env             
                    /lib/yosys-with-plugins-python3.11/site-packages/click/types.py", line 5, in                   
                    <module>                                                                                       
                        from datetime import datetime                                                              
                      File                                                                                         
                    "/nix/store/7hnr99nxrd2aw6lghybqdmkckq60j6l9-python3-3.11.9/lib/python3.11/datetim             
                    e.py", line 12, in <module>                                                                    
                        import math as _math                                                                       
                    ModuleNotFoundError: No module named 'math'                                                    
                    ERROR: Python interpreter encountered an exception.                                            
[17:20:11] ERROR    Full log file: 'openlane_run/1-yosys-synthesis/yosys-synthesis.log'                step.py:1372
---------------------------------------------------------------------------
StepError                                 Traceback (most recent call last)
<ipython-input-8-7b5c2bc7fd80> in <cell line: 7>()
      5     state_in=State(),
      6 )
----> 7 synthesis.start()

/content/openlane_ipynb/openlane/steps/step.py in start(self, toolbox, step_dir, _no_rule, **kwargs)
   1155                 ) from None
   1156             else:
-> 1157                 raise StepError(
   1158                     f"{self.name}: subprocess {e.args} failed", underlying_error=e
   1159                 ) from None

StepError: Synthesis: subprocess (1, ['yosys', '-y', '/content/openlane_ipynb/openlane/scripts/pyosys/synthesize.py', '--', '--config-in', '/content/openlane_run/1-yosys-synthesis/config.json', '--extra-in', '/content/openlane_run/1-yosys-synthesis/extra.json', '--output', '/content/openlane_run/1-yosys-synthesis/spm.nl.v']) failed
```

donn, 2024-10-21:
Argh, victim of the new release. Sorry about that!

You can roll back to working version temporarily as follows:

[image]

donn, 2024-10-21:
We've gone ahead and made the Colab use Version 2.1 by default until we figure out what's wrong with 2.2