chipfoundry/openlane2 #619: SYNTH_NO_FLAT:  _Design_add_blackbox_models() missing 2 required keyword-only arguments: 'includes' and 'defines'

DevrathIyer, 2024-12-06:
### Description

The last call to run the `d_flat` strategy (run when SYNTH_NO_FLAT is true) does not have the 'include' and 'define' arguments set. This can be demonstrated by modifying the `spm` example to have SYNTH_NO_FLAT to be true. Note that I am running openlane2 in a Nix-shell, isolated in a Debian Docker container.

### Expected Behavior

Expected `d_flat` to return successfully.

### Environment report

```sh
Failed to get Docker info: [Errno 2] No such file or directory: 'docker'
kernel: Linux
kernel_version: 6.10.11-linuxkit
supported: True
distro: debian
distro_version: 12
python_version: 3.11.9
python_path:
  - /nix/store/k8j6d34a1bxiq0b3j9vvl7s3x8i65d89-python3.11-openlane-2.2.7/bin
  - /nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/lib/python311.zip
  - /nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/lib/python3.11
  - /nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/lib/python3.11/lib-dynload
  - /nix/store/czlizmr5dmpxi7ysqknxys82776hldgk-python3-3.11.9/lib/python3.11/site-packages
  - /nix/store/xf77xcd9iw7bxqh2f2dwkp7zdxynphvg-python3-3.11.9-env/lib/python3.11/site-packages
  - /nix/store/k8j6d34a1bxiq0b3j9vvl7s3x8i65d89-python3.11-openlane-2.2.7/lib/python3.11/site-packages
  - /nix/store/m9c1yfjf098j1kigdwkgf6zs8zick9gq-python3.11-click-8.1.7/lib/python3.11/site-packages
  - /nix/store/kqvy9qmwns1ddwz2fbivhy7gsgqk48mk-python3.11-cloup-3.0.5/lib/python3.11/site-packages
  - /nix/store/01bc3isbfhf3kzp5k2alg1y86pcr2m3x-python3.11-pyyaml-6.0.1/lib/python3.11/site-packages
  - /nix/store/1k1nhsp8d42g9c8mmxi2i2wk6c0lv9vp-python3.11-yamlcore-0.0.2/lib/python3.11/site-packages
  - /nix/store/w6xq1prz9154hch0ibjh3hmmxsc2k78s-python3.11-rich-13.7.1/lib/python3.11/site-packages
  - /nix/store/0g98mc67vyrvkmnhjdcmiwq0zcilj4f5-python3.11-markdown-it-py-3.0.0/lib/python3.11/site-packages
  - /nix/store/mjnxvcich55hi1xwizpmsvyqwmzcinm9-python3.11-mdurl-0.1.2/lib/python3.11/site-packages
  - /nix/store/6f6v9dfyh0ci1dfznk0wrn56zj94pzlx-python3.11-pygments-2.17.2/lib/python3.11/site-packages
  - /nix/store/w4gg0xjr588nvn9ik8lplwxfhaf8mvcn-python3.11-requests-2.31.0/lib/python3.11/site-packages
  - /nix/store/kvgyp8l78ikc7wgmngk3pkdnhvl8l81q-python3.11-brotlicffi-1.1.0.0/lib/python3.11/site-packages
  - /nix/store/r7xfrdagikkhkrakfypdi8qz93is63wn-python3.11-cffi-1.16.0/lib/python3.11/site-packages
  - /nix/store/i95n4ijlpnakh48v7p0a53b54scacnj7-python3.11-pycparser-2.22/lib/python3.11/site-packages
  - /nix/store/j1c0zyz298y96pxx6jf8lsqp5xrlmk72-python3.11-certifi-2024.02.02/lib/python3.11/site-packages
  - /nix/store/d50zma88zgc4aqkabzia7150blbm3v39-python3.11-charset-normalizer-3.3.2/lib/python3.11/site-packages
  - /nix/store/9gnyfw31lkvsadfr1dwn5wmh93j74pxb-python3.11-idna-3.7/lib/python3.11/site-packages
  - /nix/store/g5lxmn85z7mvcb0wmqqsizbbyxnhlnyc-python3.11-urllib3-2.2.1/lib/python3.11/site-packages
  - /nix/store/gbhx29h42b41dis5lsvs789frcvyljdh-python3.11-pcpp-1.30/lib/python3.11/site-packages
  - /nix/store/bk3svsmd0n96n29x5rkwk4az7nyk0w3i-python3.11-volare/lib/python3.11/site-packages
  - /nix/store/lfngwmhl0vydan2x750b6dkpc0qjsd7g-python3.11-httpx-0.27.0/lib/python3.11/site-packages
  - /nix/store/jaaxs1aags7m73l2xsms79z6axvwc4yf-python3.11-anyio-4.3.0/lib/python3.11/site-packages
  - /nix/store/37rqx791p2b0yjvli5jfmx7xnrgc9g74-python3.11-sniffio-1.3.1/lib/python3.11/site-packages
  - /nix/store/2nh729n5wsr6gqsa20wlcmixdl5qshfp-python3.11-httpcore-1.0.5/lib/python3.11/site-packages
  - /nix/store/b0rk4ih8jrx0ijk0hvbypjqw09z4ajl1-python3.11-h11-0.14.0/lib/python3.11/site-packages
  - /nix/store/z7mj67hg6mhn8w1fm9lfa9l8a5diw414-python3.11-zstandard-0.22.0/lib/python3.11/site-packages
  - /nix/store/jpmv7rjza1fv862sqqg1flxhivp2hk13-python3.11-truststore-0.8.0/lib/python3.11/site-packages
  - /nix/store/6hj2gl8pwks57qib36f543sr1dcvk045-python3.11-aiohttp-3.9.5/lib/python3.11/site-packages
  - /nix/store/9p1gxzqcmvfqsmh8cyrwn295cy1j92wi-python3.11-attrs-23.2.0/lib/python3.11/site-packages
  - /nix/store/y6zifl6q0dwfh4pnkh49d5ln8p382g86-python3.11-multidict-6.0.5/lib/python3.11/site-packages
  - /nix/store/fxag2vqyvva3r2ph75hj5b5w909vqhrj-python3.11-async-timeout-4.0.3/lib/python3.11/site-packages
  - /nix/store/w6s7z70wx8fc951v0z3ddqmy94bwqg4q-python3.11-typing-extensions-4.11.0/lib/python3.11/site-packages
  - /nix/store/ld31bcfdkvimwb8fsflny2445g652lv1-python3.11-yarl-1.9.4/lib/python3.11/site-packages
  - /nix/store/3i99zvvs3qb5g4934qs6aph2z1cncw9w-python3.11-frozenlist-1.4.1/lib/python3.11/site-packages
  - /nix/store/bxalm8mw3y917mx9x68i6a2zgrb39n1j-python3.11-aiosignal-1.3.1/lib/python3.11/site-packages
  - /nix/store/y1j9sjam2qp7z6mfda6xfxaanq4jaihk-python3.11-aiodns-3.2.0/lib/python3.11/site-packages
  - /nix/store/z38mvg3az2434sf2i18g2k5akdly9hmn-python3.11-pycares-4.4.0/lib/python3.11/site-packages
  - /nix/store/gr65m29rdvrjclcz9nb00kisplhs4bv5-python3.11-brotli-1.1.0/lib/python3.11/site-packages
  - /nix/store/slw88ws3sn8gwffdjlnh71j49dgqn3qz-python3.11-cryptography-42.0.5/lib/python3.11/site-packages
  - /nix/store/byk4rl0fss43rmac5b2k6g0ywnlbvg1b-python3.11-pyopenssl-24.1.0/lib/python3.11/site-packages
  - /nix/store/kzl4wr5w5gllxjnfa9pqiidw5q0w6r8b-python3.11-trustme-1.1.0/lib/python3.11/site-packages
  - /nix/store/jg3x9qglnfh58y4qzhwnd9akpnwjhz4x-python3.11-socksio-1.0.0/lib/python3.11/site-packages
  - /nix/store/qz4ii6l2fjm62v1i82bqyxazspabafqq-python3.11-tkinter-3.11.9/lib/python3.11/site-packages
  - /nix/store/246xb0jbbwhcfpzmmlf1c0l0ab1dpyzq-python3.11-lxml-5.1.0/lib/python3.11/site-packages
  - /nix/store/4y0ab7wfjggb66h8znr5801llpan22fq-python3.11-deprecated-1.2.14/lib/python3.11/site-packages
  - /nix/store/sglj8kl5aqbkib7iyw8y0p2d29b92s5r-python3.11-wrapt-1.16.0/lib/python3.11/site-packages
[... 40 lines cut ...]
ERROR: Python interpreter encountered an exception.                                                                                                                                                 
[19:57:06] ERROR    Subprocess had a non-zero exit.                                                                                                                                     step.py:1364
[19:57:06] ERROR    Last 10 line(s):                                                                                                                                                    step.py:1369
                      File "/nix/store/8dgw9hbjqba485czzfxw970a9zn300sn-yosys-with-plugins-python3-3.11.9-env/lib/yosys-with-plugins-python3.11/site-packages/click/core.py", line                  
                    1434, in invoke                                                                                                                                                                 
                        return ctx.invoke(self.callback, **ctx.params)                                                                                                                              
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                              
                      File "/nix/store/8dgw9hbjqba485czzfxw970a9zn300sn-yosys-with-plugins-python3-3.11.9-env/lib/yosys-with-plugins-python3.11/site-packages/click/core.py", line 783,             
                    in invoke                                                                                                                                                                       
                        return __callback(*args, **kwargs)                                                                                                                                          
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                                          
                      File "/nix/store/xf77xcd9iw7bxqh2f2dwkp7zdxynphvg-python3-3.11.9-env/lib/python3.11/site-packages/openlane/scripts/pyosys/synthesize.py", line 355, in synthesize             
                        d_flat.add_blackbox_models(blackbox_models)                                                                                                                                 
[... cut ...]
                    '/fuse/openlane2/openlane/examples/spm/runs/RUN_2024-12-06_19-57-04/06-yosys-synthesis/config.json', '--extra-in',                                                              
                    '/fuse/openlane2/openlane/examples/spm/runs/RUN_2024-12-06_19-57-04/06-yosys-synthesis/extra.json', '--output',                                                                 
                    '/fuse/openlane2/openlane/examples/spm/runs/RUN_2024-12-06_19-57-04/06-yosys-synthesis/spm.nl.v']) failed                                                                       
[19:57:06] ERROR    OpenLane will now quit.
```

donn, 2024-12-07:
D'oh. Thanks for catching that. @wrs225 Can you open a PR with your fix? I can get that merged ASAP.

donn, 2024-12-07:
Wow, you already did. Fantastic

PR #620 (2024-12-07): bugfix: crash when `SYNTH_NO_FLAT` is set to `true`
* `Yosys.*Synthesis`
  * Fixed crash when `SYNTH_NO_FLAT` is set to `true`.

---

Resolves #619
closed 2024-12-07 by donn